espressif/esp32-camera #2: M5STACK ESP32CAM Example?

dharders, 2018-11-21:
Hi, 

Are you able to provide an example of how to use this library with the ESP32CAM M5STACK module?

I was able to get the module up and running with the demo at [https://github.com/igrr/esp32-cam-demo](https://github.com/igrr/esp32-cam-demo), however, this demo only enabled use of the OV2640 module at a max of **800x600** resolution as it didn't make use of the external PRAM.

That's why I'm so interested in your demo, as a quick look through your source code showed that you have provided support for all resolutions up to the max UXGA resolution of the OV2640.

So is it possible for you to provide a simple example project or update the instructions on how to integrate with the ESP32CAM M5STACK module? I'm new to the ESP world / tooling and need a little hand holding at the start 😄.

I was able to get the [https://github.com/igrr/esp32-cam-demo](https://github.com/igrr/esp32-cam-demo) project happening from just installing the dependencies and typing `make menuconfig` then `make flash`. I'd love for the instructions here to be as clear, even if we have to do a little extra work, but just a hint in the right direction!

I'm sure there are a lot of excited ESP32CAM M5STACK owners who have been waiting for a project like this as the demo shipped with the module was somewhat disappointing with the shipped low resolution enabled.

Thanks for reading. This project looks very promising!

dharders, 2018-11-21:
On further investigation, I don't think this module has any external RAM :(

I assumed it had 4MB of RAM, but in fact it's only 4MB of flash. It only has a small amount of internal RAM, which is probably why their demo only uses 800x600, which is very misleading as they advertise the module with capability of 1600x1200 😡 

I probably have to manually hack/attach a RAM chip to get the advertised resolution, but completely goes against the reason I bought the standalone module in the first place!

If what I'm saying is true, please confirm/correct me, then I guess you can close this issue. But I would love to hear your thoughts first before I hack or return this module.

Thanks!

igrr, 2018-11-21:
I suppose 1600x1200 might fit into internal RAM, provided that camera sends a JPEG encoded frame? Depending on compression ratio...

me-no-dev, 2018-11-21:
If you have the non-psram model, then the max you can stream is 800x600, 1 frame buffer and quality around 10 or more. With PSRAM you can do full res, full quality and 2 or more frame buffers.

closed 2018-11-21 by me-no-dev

dharders, 2018-11-22:
Thanks for the clarification! 

I think I'll return the module and look for one with psram. Cheers.