espressif/esp32-camera #5: Ai-Thinker CAM board and OV2640

B0RJA, 2018-12-08:
I am testing the code with the Ai-Thinker CAM board using an OV2640 sensor.

The initial camera probe fails with 0x20001 error (ESP_ERR_CAMERA_NOT_DETECTED).

I have used the same hardware and PIN configuration with the @igrr [example](https://github.com/igrr/esp32-cam-demo/) with no problems.

Have you tested the code using the OV2640 sensor?

This is the PIN config:

```
//Ai-Thinker CAM board PIN Map
#define CAM_PIN_PWDN    -1 //power down is not used
#define CAM_PIN_RESET   -1 //software reset will be performed
#define CAM_PIN_XCLK     0
#define CAM_PIN_SIOD    26
#define CAM_PIN_SIOC    27

#define CAM_PIN_D7      35
#define CAM_PIN_D6      34
#define CAM_PIN_D5      39
#define CAM_PIN_D4      36
#define CAM_PIN_D3      21
#define CAM_PIN_D2      19
#define CAM_PIN_D1      18
#define CAM_PIN_D0       5
#define CAM_PIN_VSYNC   25
#define CAM_PIN_HREF    23
#define CAM_PIN_PCLK    22
```
And the error:

```
E (397) camera: Enabling XCLK output                                            
E (397) camera: Initializing SSCB                                               
E (397) camera: Software resetting camera                                       
SCCB_Write [ff]=01 failed                                                       
SCCB_Write [12]=80 failed                                                       
E (407) camera: Searching for camera address                                    
E (427) camera: Camera probe failed with error 0x20001                          
E (427) wifi station: Camera Init Failed                                        
I (427) wifi station: ESP_WIFI_MODE_ST
```

me-no-dev, 2018-12-10:
I looked around for schematic of that board... not found :)
Can you please try changing the following lines https://github.com/espressif/esp32-camera/blob/master/driver/twi.c#L257-L258

to
```cpp
    pinMode(twi_sda, OPEN_DRAIN | PULLUP | INPUT | OUTPUT);
    pinMode(twi_scl, OPEN_DRAIN | PULLUP | INPUT | OUTPUT);
```

I also need to adjust timing, though that should not be an issue. I think that AI-Thinker forgot to add pull-ups to the camera i2c

B0RJA, 2018-12-10:
Thank you for your quick response.

I will test the changes this afternoon and report the result.

The board schematic can be found on the following link:
https://github.com/SeeedDocument/forum_doc/raw/master/reg/ESP32_CAM_V1.6.pdf
and more details on the product page:
https://www.seeedstudio.com/esp32-cam-p-3153.html

me-no-dev, 2018-12-10:
thanks for schematic :) what I see is that there are pull-ups on the I2C lines. But I also see that cam power is defined to pin 32. Can you add code to make sure that io32 is output and HIGH before you attempt to start the cam?

me-no-dev, 2018-12-10:
to be frank, I have tested the code in this repo with a few different camera boards without a single fail. I do not have this AI-Thinker though... I hope that it's this power thing

B0RJA, 2018-12-10:
the problem was solved setting CAM_PIN_RESET to 32.
Thank you!

closed 2018-12-10 by B0RJA

B0RJA, 2019-01-04:
@aguaviva Try this [demo](https://github.com/espressif/esp-who/tree/master/examples/single_chip/camera_web_server) placing the camera [driver](https://github.com/espressif/esp32-camera/) in the components folder of the demo. In the first make just configure the esp32-cam board.

aguaviva, 2019-01-04:
@B0RJA you made my day, it works like a charm!!!

lvturner, 2019-02-04:
Not sure if I should open a new issue on this or just comment here... for now here will suffice.

I'm have a similar issue, but different error messages - my setup is the same as the parents, I've corrected the reset pin, but I still get an error message (noted below, slightly different from others)

```
I (359) AuntyCam: Booting...
D (369) camera: Enabling XCLK output
D (369) ledc: LEDC_PWM CHANNEL 0|GPIO 00|Duty 0002|Time 0
D (379) camera: Initializing SSCB
D (379) camera: Resetting camera
I (379) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
D (409) camera: Searching for camera address
E (419) camera: Camera probe failed with error 0x20001
E (419) AuntyCam: Camera Init Failed
```
I tried the methods suggested in this thread in varying combinations, to no avail.

I've also attempted cloning the esp-who demo as suggested by @B0RJA but it also fails.

The only thing that does seem to work, is the following repository: https://github.com/donny681/ESP32_CAMERA_QR which will detect and run the camera on this board without (much) issue.

I've noticed there seem to be at least two different variants of the AI-Thinker ESP32-CAM board, one with a reset switch on the back, and one without - for what it's worth mine has the reset switch on the back.

If anyone has any ideas or suggestions, I'd be really keen to hear them - right now I'm at a total loss and quite confused!

B0RJA, 2019-02-04:
@lvturner Can you post the pinout of your board, or the page where you bought it, to check if is the same as mine?

lvturner, 2019-02-04:
@B0RJA Lightning fast with your response! Here's an update:

I've just modified `camera.c` line 870 to remove the else statement found there to force a reset on the camera, I now get the following in the logs

```
I (145) camera: Resetting OV2640
SCCB_Write [ff]=01 failed
SCCB_Write [12]=80 failed
E (165) camera: Camera probe failed with error 0x20001
E (165) AuntyCam: Camera Init Failed
```

Which is at least consistent with your original error!

Purchased from here: https://item.taobao.com/item.htm?spm=a1z09.2.0.0.fc652e8dP485Zp&id=577628508892&_u=43nqq4so7e01

I've also attached a couple of photos of the exact board I have, in case it helps any
 
[image]
[image]

lvturner, 2019-02-04:
Replaced the camera on the board with another, working properly now - suspect the original camera is just very temperamental :|

videobuff, 2019-05-04:
Is it possible to switch of the FLASH LED ?

B0RJA, 2019-05-04:
In my board (Ai-Thinker CAM) the flash led is connected to GPIO 4 with is also serves as Data 1 to the microSD card. If you do not use the microSD, it is possible to configure the pin as output with the gpio_set_direction and to low level with the gpio_set_level function. This turns off the light but unfortunately the pin is configured during boot as input and the light is visible for a moment after power on. I ended up removing physically the led from the board.

buitre-buitaker, 2019-05-04:
> Is it possible to switch of the FLASH LED ?

Yes, just set it to LOW.

int lampledPin = 4;

setup()

  pinMode(lampledPin, OUTPUT);

loop()

  digitalWrite(lampledPin, LOW);

maciokeks, 2019-10-20:
Hi @B0RJA. What I must do when I see this error?  https://pastebin.com/nyJXVChg

xenpac, 2020-03-14:
resolder pins
https://community.home-assistant.io/t/for-those-struggling-to-get-esp32-cam-clone-ai-thinker-module-working/127517

chandana9123, 2020-03-21:
how to take photo using esp32 cam when interrupt is called and send that photo to server without storing into sd card??

rtarta, 2020-04-09:
@chandana9123  My friend fallwo this url https://www.instagram.com/arduinonet/ where is a sample and show case will be in 4 days.

KeenStratium, 2020-04-18:
> @chandana9123 My friend fallwo this url https://www.instagram.com/arduinonet/ there is a sample and show case will be in 4 days.

Got any public repo code for that demo? thanks

rtarta, 2020-04-18:
@KeenStratium  You can check this tutorial.
https://rahmitugrulaltin.blogspot.com/2020/04/esp32-camera-taking-picture-and-send-to.html

KeenStratium, 2020-04-20:
> @KeenStratium You can check this tutorial.
> https://rahmitugrulaltin.blogspot.com/2020/04/esp32-camera-taking-picture-and-send-to.html

Thanks got it working! I've been struggling to have both WiFi and BLE enabled while sending a photo to a web server, this code helped.

pylover, 2022-03-09:
This repo is not working with OV2640 at all. I can confirm this repo is working well https://github.com/donny681/ESP32_CAMERA_QR.

Thanks to [lvturner](https://github.com/lvturner).