espressif/esp32-camera #7: Consider Arduino support?

cnidus, 2018-12-14:
I've got an esp32 project that's already built in arduino that I'm trying to add in OV2640 camera support. 

This is the simplest library I've found so far, but porting my existing project back to esp-idf is busy-work I'd like to avoid. 

Would you consider instructions / changes to use in arduino?

me-no-dev, 2018-12-14:
yes, I am thinking also of the best way to add it. I will be updating the IDF libs in Arduino any day now, and I think best is just to build the driver and add it that way. It's folder structure is otherwise incompatible with Arduino and can not be just included. Making a copy with the proper folder structure is also out of the question, since it will require to support the same code in two places. Keep an eye out over the weekend for update to Arduino (1.0.1-rc2)

me-no-dev, 2018-12-16:
esp32-camera is included in the latest Arduino git and in 1.0.1-rc2 from the dev package manager link :)

closed 2018-12-16 by me-no-dev

cnidus, 2018-12-18:
Nice! Thanks so much. That saves me a bunch of time!

5shekel, 2018-12-19:
Any pointers or lib out there for using this code? 
TTHO t-journal has working arduino port in their github for now. It works

me-no-dev, 2018-12-19:
the lib is already in Arduino, so all you need to do is include the headers and you are good to go :)
Here is an example using the lib with esp-idf: https://github.com/espressif/esp-who/tree/master/examples/single_chip/camera_web_server code can be copied and used in Arduino as is.

me-no-dev, 2018-12-19:
@5shekel I do not recommend that lib  if it is based on Ivan's earlier lib. This lib uses improved communication and registers with OV2640 and will perform much better than the old lib.

5shekel, 2018-12-19:
agree the lilygo fork is a bit slow , but it has a working exmaple. i was fighting with camera support for a few days. with esp-idf and arduino. its the only one that cam ready to go. i will dig deeper

here is my testing > wiki.idiot.io/esp32-cam

me-no-dev, 2018-12-19:
I linked to you to a working example :) includes face detection and recognition also :) And all kinds of controls for the camera.
[image]

me-no-dev, 2018-12-19:
If you need directions on how to run this code in Arduino let me know :) it's really easy :) you copy 4 files to the same folder as your sketch, two includes and a call to two functions and you have it running in Arduino :)

5shekel, 2018-12-19:
got it working !
using the camera_web_server example you linked above and an esp32-cam ai-thinker module (thanks to the gods, i have like 5 of this)

[image]

5shekel, 2018-12-19:
would appreciate an arduino example, just for completeness of this successful night.
 was about to give up on the whole esp32-cam adventure.

me-no-dev, 2018-12-19:
give me a few :) I'll post you the sketch

me-no-dev, 2018-12-19:
[esp32-camera.zip](https://github.com/espressif/esp32-camera/files/2696731/esp32-camera.zip)
 try this please and if all is good, I will add the example to Arduino

me-no-dev, 2018-12-19:
to get your other cameras to work, you need to set the proper pins and that is it :)

me-no-dev, 2018-12-19:
esp-who will soon be supported also in Arduino :)

me-no-dev, 2018-12-19:
oh! don't forget to select either WROVER or ESP32-Dev as board (and enable PSRAM), else it will not work

5shekel, 2018-12-19:
working on arduino with that ai-thinker board, using 
`Development release link: https://dl.espressif.com/dl/package_esp32_dev_index.json
`
i suspect its getting hot and stutter because of that after a few mintues. 

on first try ::weepii::

me-no-dev, 2018-12-19:
nice! I will add it to the repository :)

me-no-dev, 2018-12-19:
it is possible to have issues because of many things. I have 7 different sensors here and they all behave and look different... really depends on the quality of the product also.

s60sc, 2018-12-21:
Used the Arduino variant esp32-camera.zip with my ai-thinker cam board. Very impressed - fastest and most stable of the different libraries I've tried.  
Minor cosmetic - add \n to the Serial.printf statements.

me-no-dev, 2018-12-23:
ahh yes... I translated them from ESP_LOGX which adds it on it's own :) Example is already in github master of Arduino.

Re performance and stability: I spent months debugging and was even able to get some extra info from the manufacturer. The register definitions that float around the internet are not optimal for this sensor. Also it has implemented continuous mode for acquisition which allows for that high speed that you see :) glad you like it.

istvank, 2019-01-26:
Thank you so much, worked out of the box here as well. I tried it on an ESP32-CAM with a wide angle lense I got from a TTGO T-Journal order.

The only issue was connecting the ESP32-Cam to my USB to TTL: First connect VCC and Gnd, plug in USB, then connect RX/TX, finally ground IO0 and press reset. To start, disconnect IO0 from Gnd and press Reset.

Petrovjan, 2019-02-19:
Hello, thanks a lot for the arduino implementation, I was easily able to make it work! May I ask if you considered transferring also the other examples of esp-who to arduino? I'd especially love to use the detection_with_command_line, as in my project I only need to work with the detection results while keeping the video stream hidden (due to GDPR). In the current web server example, I unfortunately don't know how to hide the video stream without breaking the rest of the program.

TarjeiD, 2019-03-17:
How do I edit the menu items in the webpage? Where do I find the source?

ESP32DE, 2019-03-19:
> 
> 
> How do I edit the menu items in the webpage? Where do I find the source?
@TarjeiD 
step1: simple download @me-no-dev  [esp32-camera.zip](https://github.com/espressif/esp32-camera/files/2696731/esp32-camera.zip)

step2: unzip it, you will find camera_index.h

[image]

step3: open camera_index.h and copy the data from variable const uint8_t index_html_gz[] 

[image]

to an editor, example UEDIT32

[image]

step4: now replace all 0x to nothing
step5: now replace all ,[space] to nothing
step6: now replace all [space] to nothing
you get then a clear HEX STRING code
 
[image]

step8: select all and copy the data

[image]

step8: open [HxD](https://mh-nexus.de/en/downloads.php?product=HxD20) and create a new file
[image]

step9: paste the data and save the file to **index.html.gz**
[image]

step10: unpack the index.html.gz 

you find now your HTML file that you can edit.

[image]

If you want do then back, follow the steps back
and add at last 0x before a data with 2digits and so on 
like the data format in the header file.

all fine?
happy.

best wishes
rudi ;-)

TarjeiD, 2019-03-20:
Thanks for the detailed description @ESP32DE.
However, this is for me the hard way. The simple way is to choose "view source" in my web browser when connected to the ESP Cam, which give me the same content.
I was looking for the source with the related build script.
I've used ESPurna (https://github.com/xoseperez/espurna) which also compress the index file. The source and the build script is all there.

me-no-dev, 2019-03-21:
There is no build script :) the file is gzipped and turned into bytes. That is all. I wrote this very basic program to turn files into byte array: https://gist.github.com/me-no-dev/f137a950ce6dedb641d427d8db6355d2

`gzip index.html && filetoarray index.html.gz > camera_index.h`

ESP32DE, 2019-03-21:
> choose "view source" in my web browser ... which give me the same content...

..  be sure there is no server script inside - you see the src code then not same like it is in the server ( esp32 ) only the result .... 

example:
```
<html>
 <head>
  <title>PHP-Test</title>
 </head>
 <body>
 <?php echo '<p>Hello World</p>'; ?>
 </body>
</html>

```

let you see  as "view source"

```
<html>
 <head>
  <title>PHP-Test</title>
 </head>
 <body>
 Hello World
 </body>
</html>

```

best wishes
rudi ;-)

me-no-dev, 2019-03-21:
but there is no php or any other kind of backend language. plain html+css+js

TarjeiD, 2019-03-21:
Thank you @me-no-dev for the code and the clarification. I agree with @ESP32DE as a general precaution but I see from the code they are the same.

gustavoser78, 2020-09-08:
Hi! I´m new in Arduino... I started with ESP32 CAM and my intenttion is to use it during de night. So I read about de night mode. 
I must set a few registers but I can't use the SSCB_Write or SCCB_read, with the standard setting I don´t know how to modify them... I added the include but the Arduino IDE can´t find it.
I installed the package and run the web server without problems.

Thanks

Schaggo, 2020-09-08:
> Hi! I´m new in Arduino... I started with ESP32 CAM and my intenttion is to use it during de night. So I read about de night mode.
> I must set a few registers but I can't use the SSCB_Write or [...]

Hi! 
to be able to take images at very low light, you need a camera module without an IR filter.
The standard cam comes with a filter on the sensor side of the lens.

gustavoser78, 2020-09-08:
> Hi!
> to be able to take images at very low light, you need a camera module without an IR filter.
> The standard cam comes with a filter on the sensor side of the lens.

I added 4 IR leds and modify the exposure time, but it is not enough.  I read in the datasheet that i can improve it modifing the XVCLK  to increase the exposure or add dummy lines etc etc.... but i need to modify registers and I don´t know how.  I remove th IR filter just for testing, but i must use the camera during the day too.

Schaggo, 2020-09-08:
> s and modify the exposure time, but it is not enough. I read in the datasheet that I can improve it modifying the XVCLK to increase the exposure or add dummy lines etc etc.... but I need to modify registers and I don´t know-how. I remove the IR filter just for testing, but I must use the camera

Do you have a filter on your lens? It's a reddish purple coat on the sensor side of the tiny lens. With this filter attached you will not be able to take "night mode images" or utilize the IR LEDs.
The auto-exposure gets good results, so you don't have to mess with registers.

gustavoser78, 2020-09-08:
> Do you have a filter on your lens? It's a reddish purple coat on the sensor side of the tiny lens. With this filter attached you will not be able to take "night mode images" or utilize the IR LEDs. [...]

Yes, I have a double filter, I´m not completely sure of it. I must read more about it.
 I´m testing in a  complete dark room.  Without IR leds  and manual exposure control , the image is completely dark. If i turn on IR led I can see something but not clearly, but when i set the exposure time manually it improves  a lot.  So I want to try increasing the exposure time with a few recommended methods.

This is the best image i could get : 

[image]