espressif/esp32-camera #8: Streaming video

basfroman, 2018-12-20:
Hello
We are interested in the ability to transfer RTSP video using ESP32. Is it
possible Is it possible to transfer video from ESP32 to a server on the
Internet? What protocol can this be implemented in? We did a very big search
on the Internet and on your site, but we didn’t find some information.
We want to develop a device which will be connected to a local wifi network.
A camera and a relay will be connected to it. You need to manage this device
from anywhere in the world with access to the Internet. Tell me, is it
possible to implement using ESP32?

me-no-dev, 2018-12-20:
I am not sure about RTSP, but streaming to a server is possible in MJPG. Accessing the device from internet is possible in many ways.

basfroman, 2018-12-20:
> I am not sure about RTSP, but streaming to a server is possible in MJPG. Accessing the device from internet is possible in many ways.

How to make the device itself send an MJPG stream to the server?

me-no-dev, 2018-12-20:
In the readme there is an example on hot to stream to a browser. It's the same thing but to a server instead. https://github.com/espressif/esp32-camera#jpeg-http-stream

me-no-dev, 2018-12-20:
Closing this since it's not an actual issue with the library. We can continue communication regardless.

closed 2018-12-20 by me-no-dev

MauroMombelli, 2019-10-17:
hello, sorry for intrusion but i cant find if the streaming over wifi has some limitation. I see from #15 that we gan get quite decent framerate, but i wonder if those can be all transmit over wifi or we incur in other limitation. And if there are, if there is space for code improvement with some low hanging fruit or is going to be grinding for little improvement.