espressif/esp32-camera #10: Using the window function of the ov2640

buckybots, 2018-12-28:
I would like to use the windowing functionality, but I see that it is currently commented out in the sensor code.  Does this code work?, or were there problems with the implementation?  If problems, what needs to be worked out, or is this underway, and you have some time when you think it will be included.

Thanks!  This new functionality is great, thanks for taking the next step on making the camera code usable. 

from sensors/ov2640, line 136
> //Functions are not needed currently
> #if 0
> //Set the sensor output window
> int set_output_window(sensor_t *sensor, uint16_t x, uint16_t y, uint16_t width, uint16_t height)

me-no-dev, 2018-12-29:
functions are prepared, but I have not found a use for them yet, so I left them commented if someone like you pops around and wants to give them a go :) What would you use them for? They are generally called when resolution is changed, but those calls are optimised currently.

buckybots, 2018-12-29:
Basically, digital zoom.  I want to go high res on a portion of the scene.

me-no-dev, 2018-12-30:
Please give the functions a go :) If you find them working as expected, please provide some sort of simple example and we can uncomment and add them to the API :) Digital zoom could prove useful in some cases even for me :)

jjbubka, 2019-01-30:
I have used set_window_size, and it is working good. For example, I have an under-cif framesize of 64x64, and i can have a zoomed and centered window after initializing the sensor:
esp_err_t err = esp_camera_init(&camera_config);
esp_camera_sensor_get()->set_window_size(esp_camera_sensor_get(), 1600 / 8 - resolution[CAM_FRAMESIZE][0] / 2, 1200 / 8 - resolution[CAM_FRAMESIZE][1] / 2, resolution[CAM_FRAMESIZE][0], resolution[CAM_FRAMESIZE][1]);

Maybe it is a better option to have the functionality integrated in set_framesize, and have an input parameter about the zooming...

Best regards

cribskip, 2019-06-30:
As far as I understand, this code is provided with the Arduino IDE in compiled library and header form.

It would be great to have that ``#if 0`` removed so Arduino IDE users (like me) may use the advanced functions too:
- Set the sensor output window
- Set the image output size (final output resolution)
- Set the image window size >= output size
- Set the sensor resolution (UXGA, SVGA, CIF)

Is this possible?

Best regards

nigelorr, 2020-04-30:
It would be great to have access to this in Arduino, is there any progress?

me-no-dev, 2020-05-01:
all cameras now have their own function to manage the window. Will be updated in Arduino soon as well.

dougmsbbs, 2020-05-01:
Any hint as to how soon 'soon' is? I bought the camera assuming this was already in and working in the Arduino code. My bad, I should have verified it first. Sure would like to move the project forward, though. I know we're asking a lot of people already busy turning out some great stuff, but this really is  useful feature to have.

nigelorr, 2020-05-04:
> all cameras now have their own function to manage the window. Will be updated in Arduino soon as well.

That's great news. How long is it likely to take to push to Arduino? A week? A month? Looking forward to this, hope it is soon.

ntasher, 2020-07-19:
Any news? I also want to use this in Arduino and I can't figure out if there is a manual way to do it.
Thanks.

gcalhoun1949, 2021-02-26:
Is the windowing function available in Arduino IDE yet? I also have a need for setting ROI.
Thanks.

CraigBarton9, 2021-04-16:
I want to use the camera at its highest resolution to monitor size changes of a target over time. I only need to sample strips from the image relatively infrequently (every 5 mins). I am using the Arduino IDE and an ESP32Cam with 4Mb so pretty tight on memory to capture and process an image at high resolution (trying to use GRAYSCALE). So the ability to capture different parts of the scene and process them would be useful. Trying to understand all the inner workings of the program is stretching me to my limits. I see 
int set_window(sensor_t *sensor, ov2640_sensor_mode_t mode, int offset_x, int offset_y, int max_x, int max_y, int w, int h)  
in the ov2640.c file but not in the header and so am unsure what the parameters refer to.
Any advice/help would be gratefully received.

dougmsbbs, 2021-04-17:
Well, I'm going to guess they don't care about this. It's been about a year just since I asked, and still no response. Guess I'll give up on using their stuff and move on to some other platform. Too bad. I think it would have worked great.

gcalhoun1949, 2021-04-19:
Sorry it took so long to respond. I haven't checked this email in a few days.You want to do something like this :sensor_t * s = esp_camera_sensor_get();int unused = 0;s->set_res_raw(s, OV2640_MODE_SVGA, unused, unused, unused, offset_x, offset_y, total_x, total_y, width, height, unused, unused);If I remember correctly MODE sets the sensor resolution (3 options available), the offsets set the start of the ROI and totals set it's size. width and height set the output window size.I have some notes on how to map your variables somewhere. I will try to find them and send them to you.Regards,glenSent from my Galaxy Tab® A

  (#10) 

"target": "https://github.com/espressif/esp32-camera/issues/10#issuecomment-821902088",
"url": "https://github.com/espressif/esp32-camera/issues/10#issuecomment-821902088",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

nigelorr, 2021-09-12:
Has this made it into Arduino yet? Would be very useful but no further news since last April

closed 2021-09-24 by me-no-dev

garvitkhera, 2024-04-12:
@me-no-dev 
I'm using an OV5640 cam and want to have a cropped image from a scene:
Using the set_res_raw function is a bit tricky, but below is an experiment I did:

`s->set_res_raw(s, 500, 300, 900, 500, 0, 0, 1280, 720, 400, 200, false, false);`

Function definition:

`int  (*set_res_raw)         (sensor_t *sensor, int startX, int startY, int endX, int endY, int offsetX, int offsetY, int totalX, int totalY, int outputX, int outputY, bool scale, bool binning);`

I could get a cropped image (400x200) but the image is not useful and have muti-colored streaks. Example image (screenshot) attached:

[image]

Do you have some insights on what might be the issue? Or what the best way to use the set_res_raw function is. There is a lot of confusion regarding this function on multiple forums.

This is what my camera configuration looks like:

`// Camera configuration
    camera_config_t config;
    config.ledc_channel = LEDC_CHANNEL_0;
    config.ledc_timer = LEDC_TIMER_0;
    config.pin_d0 = Y2_GPIO_NUM;
    config.pin_d1 = Y3_GPIO_NUM;
    config.pin_d2 = Y4_GPIO_NUM;
    config.pin_d3 = Y5_GPIO_NUM;
    config.pin_d4 = Y6_GPIO_NUM;
    config.pin_d5 = Y7_GPIO_NUM;
    config.pin_d6 = Y8_GPIO_NUM;
    config.pin_d7 = Y9_GPIO_NUM;
    config.pin_xclk = XCLK_GPIO_NUM;
    config.pin_pclk = PCLK_GPIO_NUM;
    config.pin_vsync = VSYNC_GPIO_NUM;
    config.pin_href = HREF_GPIO_NUM;
    config.pin_sccb_sda = SIOD_GPIO_NUM;
    config.pin_sccb_scl = SIOC_GPIO_NUM;
    config.pin_pwdn = PWDN_GPIO_NUM;
    config.pin_reset = RESET_GPIO_NUM;
    config.xclk_freq_hz = 10000000;
    config.grab_mode = CAMERA_GRAB_LATEST;             
    config.pixel_format = PIXFORMAT_JPEG;                     // tried with RAW as well, same result
    config.fb_count = 2;                 
    config.frame_size = FRAMESIZE_HD;            
    config.jpeg_quality = 40;         // 0-63: lower means higher quality
    config.fb_location = CAMERA_FB_IN_PSRAM;     `
    
   Thanks.

gcalhoun1949, 2024-04-12:
The best I can recall is that: I have a large image from which I display a
smaller image.
The zero routine uses a x/y starting point to move to a new location within
the large image.
You can probably get this to work if you adjust code and hardware to match.
Hope this helps.

regards,
Glen