espressif/esp32-camera #15: Further increasing frame rate?

Knedox, 2019-01-17:
The frame rate is already pretty good, dont get me wrong :D
I am wondering whats currently limiting the frame rate in higher resolution modes like VGA and higher.
Seems like its independent from the data rate? Would it be possible to implement dynamic frame rate based on jpeg bandwidth?

me-no-dev, 2019-01-20:
the frame rate is limited by the sensor itself. in JPEG we have the bandwidth to read frames at max res at 25fps, but the camera itself is lowering it.

Knedox, 2019-01-20:
just FYI

i have disabled pixel clock divider for CIF, disabled clock prescaler and enabled clock doubler 

`REG32_CIF = 0x09`
`{CLKRC, 0x00 | CLKRC_2X}`

then i have set _config.xclk_freq_hz_ to 12mhz

now i get 60 FPS in CIF
and 30 FPS in VGA/SVGA

but SXGA and up is broken.
Setting _config.xclk_freq_hz_ to 10Mhz results in 15FPS for SXGA

might be good if you would include the changes in your code to get more FPS 👍

me-no-dev, 2019-01-20:
i tested different prescalers but they did not result in stable operation. on top of that the network and all got so bogged down that I was not able to do any meaningful processing and the quality had to be lowered as well. will do more testing when necessary, but for now I think this is quite safe :)

me-no-dev, 2019-01-20:
to note, not equal division from 80MHz is not a good idea on esp32 (and going above 20 is not even possible) because of how the peripheral works.

me-no-dev, 2019-01-26:
Your suggestions have been added [here](https://github.com/espressif/esp32-camera/commit/10b3d3c2a94ab44c92c199946190bbb4bbab5a95) :) Just set XCLK to 10MHz and double speed will be enabled

closed 2019-01-26 by me-no-dev

Knedox, 2019-04-25:
another update: 
seems like the HDF3M camera board that is supplied with the M5-Stack camera is very temperature dependent at high FPS. It has to reach temperature to get glitch free. The one supplied with esp32 cam board works at any temperature

rochus-keller, 2019-05-24:
@me-no-dev 
> the frame rate is limited by the sensor itself. in JPEG we have the bandwidth to read frames at max res at 25fps, but the camera itself is lowering it.

Interesting. Where do you have this information from? I didn't find any confirmation in the OV2640 data sheet; instead the sensor is supposed to work at 30 fps in SVGA and even 60 fpw in CIF. Could you please provide more information about your assumptions?

rochus-keller, 2019-05-24:
@Knedox
> seems like the HDF3M camera board that is supplied with the M5-Stack camera is very temperature dependent at high FPS. It has to reach temperature to get glitch free. The one supplied with esp32 cam board works at any temperature

Thanks, great work! Can I conclude from your findings that with an ordinary ESP32-CAM board available for ~6$ (e.g. like this one https://www.aliexpress.com/item/ESP32-Cam-ESP32-Cam-WiFi-Bluetooth-ESP32-Camera-Module-Development-Board-with-OV2640-Camera-Module/33025675750.html) you are able to stream VGA (640x480) at 30 fps over Wifi without any crashes with your fix? So you cannot confirm the findings by @me-no-dev that the operation is unstable?

Celppu, 2019-09-16:
[image]
I'm trying to achieve better frame rates but library provided in arduino ide is strange. In picture left and red is size in bytes and blue and right is time in ms. 

Is library in arduino ide unoptimized or am I doing something wrong? I set 10mhz setting and 12 jpeg compression.

me-no-dev, 2019-09-17:
there is nothing strange... how do you get those numbers? do you measure time between a frame is available or time between you getting and transfering it somewhere? You are trowing words but I do not get what you want to achieve? what is your target FPS at what resolution?

Celppu, 2019-09-17:
I measured only the time taken by esp_camera_fb_get(). I didn't undertand those examples and my code might be quite wrong... 
`
long startters = millis();
fb = esp_camera_fb_get(); 
if(!fb) {
    Serial.println("Camera capture failed");
    return;
  }
  Serial.println(millis() - startters);
  Serial.println(fb->len);

  esp_camera_fb_return(fb); 
`
Is there any example code how to achieve 30-60 frames second for vga and svga?

me-no-dev, 2019-09-18:
you can't. you can do 25 :)
frame rates depend on many things but in essence the pixel clock needs to be compatible with what the ESP can handle. On top of that you have to get all the bytes from DMA and put them into PSRAM fast enough. That also needs time :) Here is a sheet of what is currently possible:
[image]

fdoljack, 2019-10-22:
I would like to SLOW down the frame rate.  How can one do this in the code for the esp-who example for webcamserver??

imShara, 2020-03-07:
> you can't. you can do 25 :)
> frame rates depend on many things but in essence the pixel clock needs to be compatible with what the ESP can handle. On top of that you have to get all the bytes from [...]

Why 20MHz less framerate than 10MHz ?

me-no-dev, 2020-03-09:
Because allows for another internal multiplier to be turned on. I am hunting clock settings on the supported sensors even now. I am setting up OV5640 for example to work at "full" FPS at 20MHz XCLK and lower the FPS (and increase exposure time as result) with lower frequencies (tried 5-20 and it's awesome). OV2640 is quite different sensor and clock structure, so not sure how good I can do there.

xenpac, 2020-03-11:
> I am setting up OV5640 for example to work at "full" FPS at 20MHz XCLK 
did you insert the ov5640 in the same socket on the esp32-cam board?

tennisparty, 2021-01-28:
> you can't. you can do 25 :)
> frame rates depend on many things but in essence the pixel clock needs to be compatible with what the ESP can handle. On top of that you have to get all the bytes from [...]

Is there anyway to improve the framerate by changing the clock settings on the ESP itself?

me-no-dev, 2021-01-28:
this table is old. XCLK is driven by the ESP so in reality we are already changing the settings on the ESP

tennisparty, 2021-01-28:
Great thanks, so you have an updated table? Im looking to maximise the frame rate on an ov7725 sensor and wondering what the limit is

billy8119, 2021-02-05:
I have JPG frames being encoded, but even at VGA frames, and 10 MHz clock.. the video gets glitchy after about ~ 4 minutes.  20 seconds or so later, the camera finally crashes, and I can't reset it (reset line seems disconnected) it works when I hard reset via killing and restoring power. I have a heat sink on the back of the camera, but it's not helping.
Is this normal? Does someone else have 20+ fps VGA frames, with JPG compression? Is the 5640 more resilient than my ov2640? IE: should I switch? Ultimately, I'm not super interested in lots of frames per second, but I need low latency. Other solutions?

sohtamei, 2021-11-09:
Just FYI
[While outputting 5/10/20MHz master clock from LEDC or I2S to peripheral device, the Wifi throughput gets worse.](https://github.com/espressif/arduino-esp32/issues/5834)

thet0ast3r, 2022-02-02:
HI, this is mainly targeted at the author @Knedox . Struggle to change the clock prescaler settings in the esp32-camera driver. I am using the arduino ide, is there any way to do this in a somewhat simple way?

markjlorenz, 2022-05-19:
Since this thread is the top search result on how to increase the speed of the ov2640 camera, just a note for posterity...  I found that the current Arduino sample code (May-2022) doesn't benefit from setting XCLK to 10MHz.  What made by far the biggest difference for me was using an [IPEX](https://randomnerdtutorials.com/esp32-cam-connect-external-antenna/) connector and attaching an external antenna.  This allowed the board to keep a consistent frame rate without bogging down intermittently.

[image]
[image]
[image]
[image]

Without the antenna the SVGA was probably 0.1 FPS.   Frame rate was all over the place even at 96x96.

I'm not an expert at all, first time working with these cameras.  Please tell me if I'm doing something wrong!

B-l-u-e-f-i-r-e, 2024-09-29:
> just FYI
>
> i have disabled pixel clock divider for CIF, disabled clock prescaler and enabled clock doubler
>
> `REG32_CIF = 0x09` `{CLKRC, 0x00 | CLKRC_2X}`
>
> then i have set _config.xclk_freq_hz_ to [...]

How do you archive  those 60 FPS?
If I understand correctly those ( `REG32_CIF = 0x09` `{CLKRC, 0x00 | CLKRC_2X}`) are some things to send over to the cam via i2c to archive this framerate.
How would this look in code?

B-l-u-e-f-i-r-e, 2024-09-29:
This was the closest I could get:  
`//Initialize Camera first
 sensor_t * s = esp_camera_sensor_get();
    
  // Apply the optimizations mentioned in the GitHub issue
  s->set_reg(s, 0x111, 0x09, 0xFF);  // REG32_CIF = 0x09
  s->set_reg(s, 0x11, 0x00 | 0x80, 0xFF);  // CLKRC, 0x00 | CLKRC_2X`
  Sadly the speed decreased and not increased though.

B-l-u-e-f-i-r-e, 2024-09-29:
@Knedox can you share your code?