espressif/esp32-camera #17: any way to sent the picture via bluetooth?

simabuaboy, 2019-01-24:
the thing i had wondering was, how can it capture image and sent it via bluetooth, 
for example capturing an image with it and sending it to android via bluetooth and saving it to the cloud through android internet.

innergap, 2019-03-19:
Same question here. I need to capture the images via bluetooth since my users will use the external camera in remote places with no WiFi connectivity.

me-no-dev, 2019-03-20:
such examples are beyond this repository, but the basic flow is:
- init bluetooth
- init camera
- wait for bluetooth connection
- grab a frame
- send the bytes over bluetooth

that "over bluetooth part" really depends on what you want to implement.

yogendrakahane, 2020-07-14:
Same here I want to take picture with external interrupt and send the image from inbuiled esp32 cam bluetooth

closed 2021-09-19 by github-actions[bot]