espressif/esp32-camera #18: Any support for ov5640 possible?

searchingforcode, 2019-01-27:
Hi is it possible to integrate ov5640 driver? I found drivers of ov5640 for linux. Can esp32 process ov5640 images?

Niek, 2019-04-02:
+1 for this

I know there is OV3660 support in the corresponding branch (https://github.com/espressif/esp32-camera/tree/ov3660-support), but the OV5640 is more widely available on Taobao/AliExpress and supports 30fps @ 1080p resolution.

me-no-dev, 2019-04-02:
don't believe frame per second numbers like this :) while the camera might be able to do it, it does not mean that the ESP can handle it :) this is mainly where we hit a wall.
Also Adding a new driver takes time and effort that should have some sort of positive outcome in therms of business and OV5640 does not look like a prime time candidate. If any of you is willing to put some time into it, I can guide you through getting the driver to work.

Plus, I think we will have enough ov3660 cameras on the market soon :P

Niek, 2019-04-02:
OK, that makes sense - looking forward to see those OV3660 sensors/updated ESP-EYE board 👍

searchingforcode, 2019-04-02:
Hi great. Waiting for the ov3660 modules. And more thing, there is an micro-rtsp implementation for esp32 on platformio. Can it be ported/implemented to esp-idf? Thanks a lot :)

Niek, 2019-04-02:
@searchingforcode ADF will (very?) soon have SIP/VoIP added: https://github.com/espressif/esp-adf/issues/106 - I guess it will be possible to add a MJPEG video stream in there as well, which will be more powerful than RTSP.

me-no-dev, 2019-04-02:
MJPEG we already do :) example is in ESP-WHO repository.

Niek, 2019-04-02:
I am aware of that, I meant MJPEG + (microphone) audio combined over SIP so it's possible to create a true "webcam" video+audio stream.

me-no-dev, 2019-04-02:
Out SIP engineer has left the company, but I will poke arround and see if I can finish it. That said, SIP uses h263 and h264 for video transmission. Voice options are a couple, but all feasible. h263/h264 on ESP32 .... not a thing yet and maybe never a thing :)

searchingforcode, 2019-04-02:
I dont think h263/h264 is not feasible on esp32. Any idea about streaming to a cloud of this mjpeg streaming?

Niek, 2019-04-02:
> Out SIP engineer has left the company, but I will poke arround and see if I can finish it. That said, SIP uses h263 and h264 for video transmission. Voice options are a couple, but all feasible. h263/h264 on ESP32 .... not a thing yet and maybe never a thing :)

I thought SIP allowed different video codecs such as MJPEG as well, but I might be mistaken. Anyway, having the SIP support in ADF would be awesome.

me-no-dev, 2019-04-02:
for actual streaming to a server, MJPEG should be encapsulated in some sort of video container in order to keep the timing between frames correct, else you will watch a video with speed comparable to the speed of the connection to your server :)

h263/h264 and other good video compressions work by having multiple frames at the same time and transmitting the differences (simplified). The ESP32 does not have enough memory or speed to acquire and keep a few frames in RGB format and then calculate and transmit their differences. Maybe some really low res thing can be done... but not 100% sure.

me-no-dev, 2019-04-02:
@Niek honestly, I think that SIP should be it's own component. We will not use the compressions used by ADF, because 90% of SIPs out there use G711 audio compression (100% doable on ESP32).

asmalldev, 2019-05-21:
Hi @me-no-dev are there any higher resolution cameras that may be better candidates than OV5640? I am looking to add more resolution and may have time to work on a driver :)

me-no-dev, 2019-05-22:
@asmalldev honestly I am not sure if it makes sense to look at higher res cameras than 5MP. We do not have enough RAM available :) The sensor must support JPEG compression or there is no chance at all even for 3MP sensors :)

asmalldev, 2019-05-22:
@me-no-dev thanks for the input :) If the limiting factor is the RAM space, would it be possible to interface with higher resolution cameras by adding more SPI/PSRAM to the hardware? I think this could be interesting for projects that need high resolution + low power but don't have frame rate requirements/ only need to take still images.

me-no-dev, 2019-05-23:
You are limited to 4MB PSRAM access at a time, so that is the maximum.

asmalldev, 2019-05-23:
Ah right, bank switching stuff is needed to access the higher memory. Thanks for the info, @me-no-dev
I think I will try writing a driver for OV5640 to try to get 5MP w/ JPEG compression. If you know of any other 5MP sensors that would be good candidates, I'd love to hear about them :)

xxxajk, 2019-05-26:
I would be happier with any camera doing autofocus myself. 
I only need VGA resolution anyway.
I have a bunch OV5648 I pulled from dead tablets, but it has a different connector. 
I hope to be able to make my own board for it soon.
Another thought on using higher-res cameras, why not send the raw data and use javascript on the PC side to process it? :-)
Also, it seems that if the ESP is sending data there is corruption, and I get streaks in the image.
If I delay 500ms before each frame, these don't happen. 2FPS is probabbly OK for my project too.

me-no-dev, 2019-05-27:
Autofocus is a feature of the camera itself. If it can do it and provides control, it can be added.
You can use "raw" data if you want (check the BMP sending in the README) but that means many times more data that needs to go through the network. Therefore it's not good for streaming.
Please show the corruption in an image and post your camera config.

xxxajk, 2019-05-27:
My setup is:
CameraWebServer Arduino example.
ESP32-CAM board with OV2640 fixed focus camera.

I hope that the bare-bones SDK will work better, as it just could be an outdated version in Arduino-ESP32. I've recently downloaded the SDK, etc and will try the bare metal example as this issue is possibly already solved.

Project goal is to use a couple of these for vision. It will be nice to have them wireless v.s. adding more cabling.

The main processing and computation will be using OpenCV on an Android tablet or Linux PC. I haven't decided which yet.

xxxajk, 2019-05-27:
Adding example screen captures from tablet.
[image]
[image]
[image]
[image]

xxxajk, 2019-05-28:
Happens on the bare-metal variant as well, providing I turn off the debugging. Has to be a DMA conflict.

me-no-dev, 2019-05-28:
What XCLK frequency are you setting?

xxxajk, 2019-05-28:
20MHz.
Also tried 10, 12, and a few others. Result is the same.
From what I can tell, somehow, there is a race condition or DMA conflict.
I wonder if it might be a good idea to pause I2S between getting the FB and returning it back to the pool? According to the ESP32 RM, you shouldn't try to DMA and operate on a bank at the same time if you want 100% speed. Since I tried much lower speeds, that shouldn't be happening either way. 
Checked the power supply and voltages on my 'scope, and they are solid. No power problems that could perturb the camera module, but...
Seems to happen only if there is WiFi traffic. I am about to try tests using high-speed serial. If that works, then I could use an ESP8266 to do the actual transmitting. Would be awesome if it worked out-of the box though.

me-no-dev, 2019-05-28:
I have not seen any issues with 20MHz XCLK and OV2640. Have tested quite a few samples. Some were bad, but presented in another way. You can set the frame buffers to 1, which will start I2C and DMA when you request a frame and then stop it once done. That will limit the FPS by half though.

xxxajk, 2019-05-28:
Not worried about FPS at this point. 
I'll try that and let you know how it works out in a few minutes.

xxxajk, 2019-05-28:
Still happens with a singular fb. :-( Any other good ideas?

me-no-dev, 2019-05-28:
try another sensor?

xxxajk, 2019-05-28:
Have 4 of them, tried 2, they both the same thing. :-/
I'll try the other two.

xxxajk, 2019-05-28:
Tried 4 cams, on 4 boards, all showing same issue.

Niek, 2019-05-28:
@xxxajk I had similar issues on the ESP32-CAM boards with some power input sources. If you're using that board, make sure to use a 5V 2A input to avoid image distortions.

me-no-dev, 2019-05-28:
Since you say that WiFi is causing it, maybe it is power...

xxxajk, 2019-05-28:
I've got a 2A supply.... I'll try it.

xxxajk, 2019-05-28:
No difference using a different 5V supply. I'll rig up a 3.3 supply, if you think that would help.

jjbubka, 2019-05-28:
I am aware that you need that the only task that must be running in the core 1 is the dma_filter_task of camera.c, every other task related to wifi must be in core 0. That way you will prevent of skipping some lines of the dma being processed. Maybe that is your issue. Best Regards

xxxajk, 2019-05-28:
~/.arduino15/packages/esp32/hardware/esp32/1.0.2/tools/sdk
at sdkconfig | grep CORE | grep '=y'
CONFIG_ARDUINO_RUN_CORE1=y
CONFIG_ARDUINO_EVENT_RUN_CORE1=y
CONFIG_ARDUINO_UDP_RUN_CORE1=y
CONFIG_BTDM_CONTROLLER_PINNED_TO_CORE_0=y
CONFIG_BLUEDROID_PINNED_TO_CORE_0=y
CONFIG_ESP32_ENABLE_COREDUMP_TO_NONE=y
CONFIG_ESP32_WIFI_TASK_PINNED_TO_CORE_0=y
CONFIG_CAMERA_CORE1=y
CONFIG_FREERTOS_CORETIMER_0=y

xxxajk, 2019-05-28:
Possibly crap cam module?
[image]

me-no-dev, 2019-05-28:
are they all the same batch? I've stumbled on bad batches before (and we returned them to the seller).
In any way... OV2640 is EOL sensor and no longer supported/manufactured by OmniVision. Look for OV3660 (which we support and is not EOL). There should be good ones out there as opposed to "whatever is left" in the case of 2640.

xxxajk, 2019-05-28:
Been digging everywhere, and apparently either I'm stupid, or google is stupid, or both. Got a source for these? Adjustable focus, auto focus preferred. Fixed focus is useless because I need to focus from about 3cm-12cm distance. Most auto focus units can actually handle the close-up. The ones I have can autofocus @ 6.5, which is perfect for most of what I need to do.

me-no-dev, 2019-05-28:
ov2640 auto focus? what is that? None of those have it... the most basic explanation is that focus is done through optics and not through the sensor ;) You need a sensor that supports such optics and can control them.

xxxajk, 2019-05-28:
I mean for OV3660.

xxxajk, 2019-05-29:
Ok, well I am seeing some ov2640, but so far, only fisheye... way too much for them too.
I've ran it in gray scale mode, and it isn't missing anything, or if it is corrupting a pixel, I'm not seeing it.
Going to try single buffer + color and see what happens. If that works, even if it is slower, I'm all good with that.

xxxajk, 2019-05-29:
More testing shows that it happens, but that it is a rare occasion. Funny how the esp32 can't keep up with a 25Hz freq with a 4.64ms low pulse... Should be able to do that with ease.

xxxajk, 2019-05-29:
One more thing, bit busy here, but what I should do is output a pulse on an unused pin to see if we really are missing the vertical pulse. Not sure how to do that for the arduino side, but...

xxxajk, 2019-06-02:
Seems to be missing the horizontal pulses. 
They are visible on the scope, so it has to be something else going on.
I'm not entirely sure how DMA works on ESP32, such as arbitration, etc, or if something else is getting in the way.
One thing I can validate is that it isn't the camera module. I only say this because if I delay a long while between frames, the pictures stream fine. If I ensure that there is no WiFi activity, it is fine too.
I still need to have my scope do missing pulse detection, to be totally sure. Meanwhile I'm going to go another step farther and use the runt feature on the 'scope and see if the signal is indeed clean.

xxxajk, 2019-06-03:
Got in some new modules form a totally different vendor (Arducam), same problem.
So, something isn't right here. It is either missing the IRQ, or there is DMA contention.

xxxajk, 2019-06-06:
Good news, found the problem! Seems the ESP32-CAM is *VERY* sensitive to any minor voltage droop.
My fix, Add 3 1000uF tantalum caps on the 5v rail. Problem solved! Woo hoo!

Err-Rahul, 2019-08-13:
Is there anyway to implement h264 encoding on esp32 board?

mitchins, 2019-09-02:
@err-Rahul it simply can’t handle it. MJPG is the best you can reasonably hope for. Transcode it externally if you need.

msaufyrohmad, 2019-11-02:
is there anyone still working on using ov5640 on esp32? i need help to make it happened

sabas1080, 2020-01-02:
I am working in the driver for ov5640, coming soon the pullrequest

You can see my advance in 

https://github.com/ElectronicCats/esp32-camera/tree/support-ov5640

your help is welcome ;)

Niek, 2020-01-02:
Awesome work @sabas1080 🎉

me-no-dev, 2020-01-02:
I also have the driver done. waiting for a board to tune the clocks, because I am not very happy with the frame rates. Other than that, the driver is coming along with other changes. @sabas1080 you might want to throttle down a bit :)

sabas1080, 2020-01-02:
@me-no-dev 

that's great, can you share your code? I can help test I have the following hardware

https://www.arducam.com/product/5-mp-ov5640d-cmos-af-camera-module-1-4-inch-module/

kotran88, 2020-01-06:
[E][camera.c:1049] camera_probe(): Detected camera not supported.
[E][camera.c:1249] esp_camera_init(): Camera probe failed with error 0x20004

> I am working in the driver for ov5640, coming soon the pullrequest
> 
> You can see my advance in
> 
> https://github.com/ElectronicCats/esp32-camera/tree/support-ov5640
> 
> your help is welcome ;)

I cloned it and extract it to root directory of my project. 
but skiped Enable PSRAM in menuconfig process because I don't find any article about it...

and compiled Esp32 - Camera - CameraWebServer example. 
but have below error. 

[E][camera.c:1049] camera_probe(): Detected camera not supported.
[E][camera.c:1249] esp_camera_init(): Camera probe failed with error 0x20004

when I deep into camera.c 's 1049 , 
it's #if CONFIG_OV5640_SUPPORT

kotran88, 2020-01-06:
> I am working in the driver for ov5640, coming soon the pullrequest
> 
> You can see my advance in
> 
> https://github.com/ElectronicCats/esp32-camera/tree/support-ov5640
> 
> your help is welcome ;)

I have 
below error everytime...even though I paste your code on my root directory of my folder
funny thing is that even though I deleted 1049 line of camera.c /driver/include/camera.c,
it cause same error....

E][camera.c:1049] camera_probe(): Detected camera not supported.
[E][camera.c:1249] esp_camera_init(): Camera probe failed with error 0x20004

sabas1080, 2020-01-13:
@me-no-dev something new? can i help trying

@kotran88 my driver my driver is not finished and I have stopped the development waiting for the work of @me-no-dev

misghna, 2020-01-18:
@me-no-dev it looks like a lot of people are waiting your driver :) anyluck with the board?

me-no-dev, 2020-01-19:
I'm hoping to get the test board in a week or so. The current code IMO is not releasable. I am not happy with the clocks, info is limited and just generally hacks to get it going. Please do understand that I can't just post test code in Espressif repos, but do rest assured that the sensor is working and you will soon have support for it wether I get the results I hope for or not.

pushtoopen, 2020-02-02:
Just throwing another name in the mix. I'm highly interested in a ov5640 driver as well! The auto-focus is imperative for my use. Using OV2640 in the mean time, but i will be very interested in OV5640!

[51 more replies]

closed 2020-03-11 by me-no-dev