espressif/esp32-camera #21: Rotation of image

Luc3as, 2019-02-14:
Hello, could someone point me to right direction how could I rotate the image for let we say 90 degrees ? I can flip it horizontally or vertically but as the image from camera is landscape when the module is horizontal. I cannot find the right piece of code. I know there could be problem with resolution, but  square image would be acceptable in this situation. 
thanks

me-no-dev, 2019-02-14:
unfortunately I am not aware of such setting in the sensor.

solidminds, 2019-04-30:
bump! I would also like to know if this is possible. You found a rotation solution yet?

me-no-dev, 2019-05-01:
there is no such option in the sensor settings. Everything that is available is generally exposed through the API. H-Mirror and V-Flip is what is supported. Other rotation should be done in software

sriranjanr, 2019-12-13:
Was anyone able to rotate the image by 90 deg using software?

fgerard, 2020-08-20:
I just received my second esp32 cam, I noticed that I get the second one with the image wuth 90 totation, in de belt connection to the lense there is a diferent number, the one with: 8225N V2.0 171026 gives the image in the same direction al de text the chip "ESP32-CAM" the one with: TY-OV2 6400V2.0" gives the image rotated 90 degrees, at first start this second one with the sofrware that came the images where right, but when I loaded my code then it was rotated, I think ther is a cofigutation change, here is my configuration:
#define PWDN_GPIO_NUM     32
#define RESET_GPIO_NUM    -1
#define XCLK_GPIO_NUM      0
#define SIOD_GPIO_NUM     26
#define SIOC_GPIO_NUM     27

#define Y9_GPIO_NUM       35
#define Y8_GPIO_NUM       34
#define Y7_GPIO_NUM       39
#define Y6_GPIO_NUM       36
#define Y5_GPIO_NUM       21
#define Y4_GPIO_NUM       19
#define Y3_GPIO_NUM       18
#define Y2_GPIO_NUM        5
#define VSYNC_GPIO_NUM    25
#define HREF_GPIO_NUM     23
#define PCLK_GPIO_NUM     22

Is there a posibility to change some of this to get the image rotated?

closed 2021-09-19 by github-actions[bot]

Palingenesis, 2022-01-11:
You have to rotate the camera manually 
https://timsblogplace.blogspot.com/2022/01/esp32-cam-rotating-camera.html

themanyone, 2023-03-12:
The image rotates 90° after removing the camera and putting it back. No clue how this happened. Searching around for something that could cause this. Because now we have one esp32cam that is 90° different from the others with the same software upload!

Palingenesis, 2023-03-21:
If you know which orientation you want, check the manufacture number of the camera unit.
There re many variations.