espressif/esp32-camera #24: streaming to browser

fenyvesi, 2019-03-08:
The Arduino example program prints at start:
...
Starting stream server on port: '81'
...
After starting capturing I  tried to get the stream on port 81 in a browser. It waits for data, but can't see anything. I tried Firefox, Chrome, Edge. VLC breaks down immediately.

Did I misunderstand something? Am I able to get the stream only in the web server window?

Thank you.

fenyvesi, 2019-03-13:
Anybody?

me-no-dev, 2019-03-13:
did you try `http://[ip-of-esp]:81/stream` ?

fenyvesi, 2019-03-14:
Yes.
the IP is: 192.168.0.206
Firefox, Chrome, Edge:
http://192.168.0.206:81/stream

Result: Waiting for the server, nothing happens.

jardous, 2019-03-19:
try to save the stream and analyze it offline

$ curl -o stream.raw http://192.168.0.206:81/stream
$ file stream.raw

and try to convert it to other format

caustinbakker, 2019-04-27:
I am having the same problem. Sometimes it works and sometimes it doesn't

czosnekltd, 2019-04-29:
I am having the same problem

rwagnervm, 2019-05-23:
I am having the same problema. Waiting for the server end get only black screen.

Shellfishgene, 2019-05-26:
I have the same problem, everything seems to work, the menu is there, but the browser shows the "broken image" symbol instead of picture. 
However it may be a hardware problem: I ordered two modules, from the same vendor. One works, one does not. It seems to be a camera problem: If I switch the OV2460 camera from one to the other, it starts to work.

xxxajk, 2019-05-26:
Same issue in chrome on android, but the plain ASOP browser works. 
This is most likely a bug in the CSS/javascript/whatever.

sdey76, 2019-06-12:
Hii
Able to get still images but same issue..no stream
Need help

xxxajk, 2019-06-12:
Try a different browser?

sdey76, 2019-06-12:
Hii i updated google chrome and it works! But not great luck on android phone on chrom! Thanx a lot

megla2, 2019-08-08:
Only worked for me if I disconnected from port 80.

abergmann, 2020-02-14:
I had the same problem with `VLC` and `ffplay`. No connection was working.

I've looked at the MJPG Stream and it turned out that the content and boundary delimiter are not in the right order. I've send a PR to [`espressif/arduino-esp32`](https://github.com/espressif/arduino-esp32/pull/3720). Still waiting for acceptance.

abergmann, 2020-02-14:
I also created #120 against `espressif/esp32-camera`.

arris138, 2020-02-27:
> Only worked for me if I disconnected from port 80.
This solved my issue, although that seems to be a bug.

mic191, 2020-03-05:
Same issue for me : no video streaming; everything seems to work, the menu is there, but the browser shows the "broken image" symbol instead of picture. I can just make screenshot of picture,  but no video. In local all is ok.

rsimiciuc, 2020-03-27:
I noticed that the stream is broken if is more than one connection to it, which means only one client can connect at a time.

sdey76, 2020-03-30:
Hi All,
Is there any way to transmit esp32 cam still image over LoRa chip using FSK or any other mode

LoRa SX1278,  or SX1268 

Thanks 
Soumitra Dey

arkhipenko, 2020-05-17:
Guys, I have been working on this issue and was able to successfully stream to VLC and browsers.
Also implemented multi-client streaming. 
Please check these repos:
https://github.com/arkhipenko/esp32-cam-mjpeg (single client)
https://github.com/arkhipenko/esp32-cam-mjpeg-multiclient (up to 10 clients)
https://github.com/arkhipenko/esp32-mjpeg-multiclient-espcam-drivers (up to 10 clients using latest ESP32-cam drivers from espressif).

sundarsg85, 2021-02-14:
Hi, I can able to code and get the streaming IP address in serial monitor page but unable to open the URL in any browser. showing fail to load after few mins.

WiFi connected
Starting web server on port: '80'
Starting stream server on port: '81'
Camera Ready! Use 'http://192.168.43.158' to connect

I am new to coding but its my hobby on doing these projects.
Kindly help me to advice on how to stream my ESP32 cam.

sundarsg85, 2021-02-14:
I tried mobile app, firefix, chrome but none able to stream the address shown on the serial monitor. Please anyone assist me

closed 2021-09-13 by fenyvesi