espressif/esp32-camera #28: Esp_http_server.h cannot find file

wdouglas1999, 2019-04-05:
Hi all, 

I am using esp-Idf v3.1. I am trying to compile the example but it cannot find Esp_http_server.h.
Also, is this example for Arduino IDE or for IDF?
Thank you in advance.

me-no-dev, 2019-04-05:
example is for IDF and requires IDF 3.2 for the web server. Arduino already has all of it integrated and available as example (not when used as IDF component).

wdouglas1999, 2019-04-05:
Wow, thank you for the lighting fast response. Will upgrade to idf 3.2 and retest.

wdouglas1999, 2019-04-05:
Hi, I just upgraded to idf 3.2 and still the same issue. Any ideas?

me-no-dev, 2019-04-05:
I am at a loss. Are you sure you have the correct IDF path? server files are there: https://github.com/espressif/esp-idf/tree/master/components/esp_http_server/include

wdouglas1999, 2019-04-07:
Hi, you are 100% correct. After the update to idf v3.3 I see the library and includes are working. 
I know this is out of the scope of the camera, but since I upgraded to idf v3.3, the WiFi does not connect anymore. It was working 100% with idf v2. Are you maybe aware of this issue? 
Thank you again for all you assistance. It is much appreciated.

me-no-dev, 2019-04-08:
Ahh... no clue! Maybe try erasing flash? I am running master all of the time and have generally no issues with wifi or bluetooth.

wdouglas1999, 2019-04-11:
Hi, just an update. To clarify there was no problem with the WiFi after upgrading from esp-idf 2.1 to 3.3. I had some issues with the memory and enabling the psram on the wrover I. All working like a charm and I must say that the camera code supplied is awesome. Thank you again for your assistance and your prompt responses.

me-no-dev, 2019-04-11:
Glad to hear all went well :)

closed 2019-04-11 by me-no-dev