espressif/esp32-camera #32: Question about compression bound value

wdouglas1999, 2019-04-26:
Good day

I would kindly like to ask how the values were selected in the following:
Are these strict values or could there be jpeg images that will require a larger frame buffer than these values allow?

int compression_ratio_bound = 1;
        if (qp > 10) {
            compression_ratio_bound = 16;
        } else if (qp > 5) {
            compression_ratio_bound = 10;
        } else {
            compression_ratio_bound = 4;
        }

me-no-dev, 2019-04-26:
Those are values gathered experimentally. By specs, the JPEG output can even exceed the equivalent RAW bytes, so exact formula is not available. Do you have issues? Maybe you are setting the quality too high?

wdouglas1999, 2019-04-26:
Hi, thank you for the reply.
Yes I am experiencing a problem. 
It seems to occur when the images I am capturing are more colorful and detailed.
A bright green band occurs at the bottom of the images.
I am not sure what is causing it and hence I am trying to do some fault finding. My current settings( with issues ) are:
Image size = VGA
Compression = 15
I have changed the compression to 20 to see if it helps.
I will test it and give feedback. If you have any other suggestions I would be happy to try them. Thx

me-no-dev, 2019-04-26:
I have seen the green band on some of the sensors. It is usually caused by flipped sensor (and when we flip is back through the software, the green band is seen). Not sure which example you are running, but in esp-who and in Arduino, there is a web server example that allows you to see what is on and play with the settings :)
Try flipping the image and see if that helps.
In any case, this is an issue internal to the sensor. It can only be fixed (hidden) with crop

wdouglas1999, 2019-04-26:
Hi,
I think you could be correct about the flipped image theory. I do flip the image in my application. However it only happens in sometimes like I mentioned earlier, when the image scene is more colorful and complex.
I will try to avoid the issue of flipping by physically flipping the camera. 
Are there any example of how to crop? I am using esp-idf.
Thank you so much for your help

me-no-dev, 2019-04-26:
There are no examples currently on crop, but I have some functions commented that do that. I have not tested them yet and they are raw, but you can give them a shot. I think that that green line is somehow an artefact from the JPEG encoder itself. JPEG has some tables for the colors and probably has some limitations.
Commented functions are here: https://github.com/espressif/esp32-camera/blob/master/sensors/ov2640.c#L136-L223

wdouglas1999, 2019-04-26:
Hi,
I noticed the flip command from you are using is 
#define REG04_VFLIP_IMG     0x40 
I have tried this before and not had great success.
It’s a long time ago but I believe it is an error.
What I use is:

SCCB_Write(0x30, 0x04, 0xF8)

wdouglas1999, 2019-04-27:
Hi, I have done some more testing.
I tried not flipping the camera and the problem persists.
I increased the compression value to 20 from 15, the problem still persists.
Now I am looking at the size of the image using FFmpeg and I see the image is 38,400kB.
This seems to be the maximum size of the buffer that the code allows using compression bound ratio of 16.
If I am calculating correctly:
(640 x 480 x 2) / 16 is about 38kB.
I haven’t had time to change the value and test as it’s weekend and family time, but I will update.

wdouglas1999, 2019-04-29:
To conclude the findings in my case:
The issue was caused by the jpeg image size being acquired by the ov2640 sensor being larger than the frame buffer size set by the ESP32. This caused artifacts and sometimes a green band to appear at the bottom of the image.
To solve this I changed the compression bound value from 16 to 12 in the Camera.c file. This increased the maximum frame buffer size from 38kB to 50kB.

My settings are:
Resolution: VGA 640 x 480
Compression value: 15
Compression: jpeg

I have no issues when flipping the sensor.

MrMabulous, 2019-10-04:
I also had to double framebuffer size for the kind of images I'm capturing to fit into the allocated buffer

closed 2021-09-19 by github-actions[bot]