espressif/esp32-camera #34: Can I use python to read the stream?

crouse12, 2019-05-03:
I find that the stream of ESP32-CAM is http://192.168.0.206:81/stream.
Can I use python opencv to read this stream?

me-no-dev, 2019-05-03:
I don't know. It's MJEPG and can surely be read in many ways, but I can not say what libraries support it out of the box.

nightgryphon, 2019-05-06:
Here is the example of catch stream with python and opencv
https://github.com/brunohorta82/ESP32_CAR_CAM

vikivivi, 2020-09-05:
Personally, I have tested ESP32-CAM using conda virtual environment with python 3.8.5 and opencv 4.4.0 package from conda-forge channel. I am using https://github.com/espressif/esp-who/tree/master/examples/single_chip/camera_web_server for ESP32-CAM.

Whereas for opencv code, both of these are tested to be working at my end. 
* https://gist.github.com/youjunjer/79e5dad5f47ee5757fcb9d401a95e76b
* https://stackoverflow.com/questions/60029614/esp32-cam-stream-in-opencv-python

closed 2021-09-19 by github-actions[bot]