espressif/esp32-camera #39: Clarification of XCLK value and number of fb.

wdouglas1999, 2019-05-16:
I would kindly like to know: 
I have the camera set to vga and jpeg.
Which XCLK setting achieves double the frame rate?
Can double the frame rate only be achieved at 10MHz XCLK or also at 20Mhz? 
Also, is there a maximum number of framebuffers that can be used and is there an advantage as to using more than 2 framebuffers?
Thank you in advance.

me-no-dev, 2019-05-16:
for OV2640 you need to set XCLK to 10MHz, not 20. Does not work on all resolutions. Beware the image is not 100% stable.
Usually more than 2 frame buffers should not be needed, though 3 might help in some cases. The maximum number generally is limited by the size of RAM/PSRAM

closed 2019-05-16 by me-no-dev

wdouglas1999, 2019-05-16:
Thank you for the quick reply as always.
When you say, double frame rate does not work on all resolutions.
Can you tell me on which resolutions it does work?