espressif/esp32-camera #41: Text overlay on image possible

eben80, 2019-06-09:
Hi!
Thanks for this awesome code. 
Currently I'm using it to serve a raw mjpeg stream to which I can connect to via browser. It works great. 

Is there any way to overlay text on the video with this as they are doing with the facial recognition code? 
I have been trying to figure it out but I'm stumped. 
I'm using an Ai Thinker esp32-cam

Thanks a lot.

me-no-dev, 2019-06-10:
You can to some limitations. With face recognition we use small frame sizes, we turn the image into RGB888 so the recognition code can run on it, then we can draw on that RGB buffer and turn it back to JPEG to stream to the browser. That turning images to and from JPEG takes quite some time, so streaming will be slow and resolution will be limited.

eben80, 2019-06-10:
Great!
I suppose the recognition takes up most of the resources in the case you mentioned above.
In case I just want to display text which shows for example temperature which is updated regularly, that would not affect frame rate that much right?

Could you point me to some example code that would help me to figure it out myself please?
I couldn't work out how to extract it from the face recognition code.

me-no-dev, 2019-06-10:
for 320x240 image, it takes about 300ms to convert to RGB888, draw text and convert back to JPEG. This means drop from 30 to 3 FPS for 320x240. Is that what you want? Higher resolutions take even more time. In the face-detect code, if face is not detected, the original JPEG is sent in order to save the time of converting back to JPEG. You could do things in the browser though... look into canvas :)

eben80, 2019-06-10:
> You could do things in the browser though... look into canvas :)
Ooooh I haven't thought of that. That would mean that the image is left alone and the overlay happens in html. Good idea I will try it out.
Thank you.

closed 2019-06-10 by eben80

flaviopuhl, 2022-02-18:
> Hi! Thanks for this awesome code. Currently I'm using it to serve a raw mjpeg stream to which I can connect to via browser. It works great.
>
> Is there any way to overlay text on the video with this [...]

Hey, I had the same problem and I coded an example. It works fine
https://github.com/flaviopuhl/ESP32-Cam-Text-overlay

eben80, 2022-02-19:
Great thank you. I will give it a go sometime.