espressif/esp32-camera #45: esp32-cam compile error

lekha48, 2019-06-25:


lekha48, 2019-06-25:
Tried to combine video streaming, wifi as access point and wifi UDP

code:
#define UDP_TX_PACKET_MAX_SIZE 15

#include "esp_camera.h"
#include <WiFi.h>
#include "esp_timer.h"
#include "img_converters.h"
#include "Arduino.h"
#include "fb_gfx.h"
#include "soc/soc.h" //disable brownout problems
#include "soc/rtc_cntl_reg.h"  //disable brownout problems
#include "dl_lib.h"
#include "esp_http_server.h"
#include <WiFiClient.h>
#include <WebServer.h>
#include <EEPROM.h>
#include <WiFiUDP.h>

int minUs = 1000;
int maxUs = 2000;
int x;
int y;
int z;
int m;
// the number of the LED pin
const int ledPin = 16;  // 16 corresponds to GPIO16
// setting PWM properties
const double freq = 5000;
const uint8_t ledChannel = 0;
const uint8_t resolution_bits = 8;

WebServer server(8888);
    unsigned int localPort = 8888;
    WiFiUDP UDP;
    boolean udpConnected = false;
    char packetBuffer[UDP_TX_PACKET_MAX_SIZE]; //buffer to hold incoming packet,
 
    char ReplyBuffer[] = "5"; // a string to send back
 boolean connectUDP();

// Replace with your network credentials
const char* ssid     = "ESP32-Access-Point";
const char* password = "123456789";

void handleRoot() {
  server.send(200, "text/html", "<h1>You are connected</h1>");
}  

#define PART_BOUNDARY "123456789000000000000987654321"

// This project was tested with the AI Thinker Model, M5STACK PSRAM Model and M5STACK WITHOUT PSRAM
#define CAMERA_MODEL_AI_THINKER

#if defined(CAMERA_MODEL_AI_THINKER)
#define PWDN_GPIO_NUM     32
#define RESET_GPIO_NUM    -1
#define XCLK_GPIO_NUM      0
#define SIOD_GPIO_NUM     26
#define SIOC_GPIO_NUM     27

#define Y9_GPIO_NUM       35
#define Y8_GPIO_NUM       34
#define Y7_GPIO_NUM       39
#define Y6_GPIO_NUM       36
#define Y5_GPIO_NUM       21
#define Y4_GPIO_NUM       19
#define Y3_GPIO_NUM       18
#define Y2_GPIO_NUM        5
[... 576 lines cut ...]

C:\Users\Lekha\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/nghttp/http_parser.h:138:31: note: previous declaration 'http_method HTTP_OPTIONS'

 #define XX(num, name, string) HTTP_##name = num,

                               ^

C:\Users\Lekha\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/nghttp/http_parser.h:102:3: note: in expansion of macro 'XX'

   XX(6,  OPTIONS,     OPTIONS)      \

   ^

C:\Users\Lekha\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/nghttp/http_parser.h:139:3: note: in expansion of macro 'HTTP_METHOD_MAP'

   HTTP_METHOD_MAP(XX)

   ^

exit status 1
Error compiling for board AI Thinker ESP32-CAM.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

me-no-dev, 2019-06-25:
So :)
1. This is not an issue with this library. Why posting here?
2. You use 2 different web servers and one for UDP. A mix of IDF and Arduino. In the same file. Bound to get issues :)
3. Your issue clearly shows that you have double declarations of the HTTP method names (that is quite fine. HTTP is the same all over the world, but you should have them in separate files so they do not overlap).

That should be enough to get you started :) Ya know... Arduino has it's own camera example...

closed 2019-06-25 by me-no-dev