espressif/esp32-camera #47: Failed to initialize after reset

Bozzibo, 2019-07-06:
Hi, 

I uploaded the code without any issue through FTDI programmer.
After reset I always get following error:

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:10088
load:0x40080400,len:6380
entry 0x400806a4

**SCCB_Write [ff]=01 failed**

Any idea how i can fix this or what is the cause? Correct model is defined.

Thanks!

me-no-dev, 2019-07-07:
what is the board? which sensor? which example? did you select software or hardware I2C(SCCB)?

Bozzibo, 2019-07-13:
TZT's ESP32-CAM board, with OV2640 cam module. Camerawebserver example.
But I found the problem, usb port didn't had enough power. With external powersource I got it working!

Thx!

closed 2019-07-13 by Bozzibo