espressif/esp32-camera #48: OV2640 initializing issue

heinerlohmann, 2019-07-15:
Hello,
If I set SCCB_HARDWARE_I2C to true, the SCCB_Probe() returns the address 0x3c although it should be 0x30 for the OV2640. Ultimately this results in a camera-not-supported error. If I disable SCCB_HARDWARE_I2C it fails to detect the camera. HOWEVER if I switch the pin_reset and pin_pwdn everything works.
Supposedly just because the reset is switched from 0->1 to 1->0 instead in the camera driver (lines 946-970):
```
    if(config->pin_pwdn >= 0) {
        ESP_LOGD(TAG, "Resetting camera by power down line");
        gpio_config_t conf = { 0 };
        conf.pin_bit_mask = 1LL << config->pin_pwdn;
        conf.mode = GPIO_MODE_OUTPUT;
        gpio_config(&conf);

        // carefull, logic is inverted compared to reset pin
        gpio_set_level(config->pin_pwdn, 1);
        vTaskDelay(10 / portTICK_PERIOD_MS);
        gpio_set_level(config->pin_pwdn, 0);
        vTaskDelay(10 / portTICK_PERIOD_MS);
    }

    if(config->pin_reset >= 0) {
        ESP_LOGD(TAG, "Resetting camera");
        gpio_config_t conf = { 0 };
        conf.pin_bit_mask = 1LL << config->pin_reset;
        conf.mode = GPIO_MODE_OUTPUT;
        gpio_config(&conf);

        gpio_set_level(config->pin_reset, 0);
        vTaskDelay(10 / portTICK_PERIOD_MS);
        gpio_set_level(config->pin_reset, 1);
        vTaskDelay(10 / portTICK_PERIOD_MS);
```
Is this an issue with the driver or with the ESP32-CAM board (A.I. Thinker i guess) I am using?

Mnark, 2019-07-15:
I'm having similar issues, but this from [the wiki](https://wiki.ai-thinker.com/esp32-cam) clarifies a little. 

What pin number are you defining as the reset pin? 

```
...
2. The ESP32 GPIO32 pin controls the camera power. When the camera is working, please lower the GPIO32.

3. Since IO0 is connected to the camera XCLK, please leave IO0 floating when using it, please do not connect high and low level.
...
```

heinerlohmann, 2019-07-15:
Sadly I don't speak (or read) Chinese.
With .pin_reset set to -1 and .pin_pwdn set to 32 everything is working fine so far.

Mnark, 2019-07-15:
Google translate :) 

That's what I have too. (-1 disables the use of the pin).

Mine was working until I started adding code, so I'm pretty sure the problems are in the code. 

I am NOT using hardware i2c.

heinerlohmann, 2019-07-15:
You're right, although I don't tend to trust automatic translation in technical matters. 
So far I have no problems with taking pictures, saving them to SD or sending them as http. 
Sometimes on boot the SCCB fails, but it succeeds on second try, so I have not looked further into the issue yet.

me-no-dev, 2019-07-16:
I've been reading about this but have not been able to reproduce it till recently. I got a hols of one board that has almost all pins mapped exactly as Wrover-Kit with the exception of power. The other issue with this is that 0x3C is actually the valid address for OV3660, so we can not treat it as error.
Overall the best action is to setup the pins properly for the given board. Maybe some sort of note needs to be left... or maybe I could just error out if 0x3C is found but SCCB fails... not sure what is the best approach to notify the user for this issue.

heinerlohmann, 2019-07-16:
Yes it's a bit tricky, but some kind of note would be nice to save other people some time, since the workaround is pretty easy, once you know what's up. Thank you for looking into it. :)

Mnark, 2019-07-16:
Here is my understanding of what should happen. This becomes a lot clearer when you don't look at the existing code....

The 2640 had 2 "modes"

-  programming mode (pin 32 low) : This is the initial state of the board, ready for configuration .

- run mode (pin 32 high) : The needs to be set after configuration for the camera to work.

To change the camera settings...

- Turn the board into programming mode (pin 32 low). (A slight delay is needed to ensure that the camera 'sees' the signal as in run mode pin 32 is being used for data transfer, so is turning on and off very quickly).
 - Adjust the setting.
 - Turn the camera on (pin 32 high).

To reset the board at run time, you need to...

- Turn the board into programming mode (pin 32 low).
- Send a 'reset' message to the sensor.. this clears the camera configuration.
- Reinitialise the sensor 
- Turn the camera on (pin 32 High)

I can see a few flaws in the existing logic, but I have another separate issue with my specific board (which turns out to actually be a 2642 not a 2640, so need to fix that first.

me-no-dev, 2019-07-17:
can you point me to that information in the datasheet? I'm not sure I understand what you mean by pin32. Which IO on the camera is this connected to?

heinerlohmann, 2019-07-23:
Thanks for your contribution to the topic, Mnark!

I know this is off topic but there is something else about the esp32-cam board I had to figure out the hard way. Apparently gpio 16 is somehow internally connected. (I am new to the whole topic of microcontrollers, but I guess it has to do with the SDRAM and core selection or something like that) So if you set it to input only, e.g. to connect a PIR, it messes with your tasks and causes crashes. So there are not a lot of pins one could use for getting an input. I turned to pin 12, which would be used by the sd-card, but since one has to use the sd-card in single line mode anyway to use the flash on pin4, pin 12 should become available.

Mnark, 2019-07-23:
It does appear to be connected internally, its labelled U2RXD on the datasheet, so would imply its some sort of external interrupt. 

This isn't my field of expertise, but I notice that the sensor module can be left running while the ESP is in sleep mode (_this functionality doesn't exist in this library_), I suspect this pin should be used to wake up the ESP from deep sleep.

(Probably what you are trying to do with a PIR... I'm intending using an RF motion detector for this task, but haven't got to it yet)

Mnark, 2019-07-24:
@me-no-dev  The mystery of the 0x03c is at the top of the program, it is the default value returned when SCCB fails... it should be 0x00.

static uint8_t ESP_SLAVE_ADDR   = 0x3c;

bartlomiejcieszkowski, 2019-12-29:
Stumbled on this error also, with the I2C hardware mode the passed slave_addr was not used,
so for the w/a where there was SCCB_Write(0x30, (...)) it wrote to the ESP_SLAVE_ADDR

Uploaded pull request for this issue: #106

xenpac, 2020-03-24:
I am suggesting this alternate code for SCCB_Probe function: (to remove this 0x3C issue)
tested.
`uint8_t SCCB_Probe()
{
    uint8_t slave_addr = 0;

#ifdef CONFIG_SCCB_HARDWARE_I2C
    while(slave_addr < 0x7f)
    {
        i2c_cmd_handle_t cmd = i2c_cmd_link_create();
        i2c_master_start(cmd);
        i2c_master_write_byte(cmd, ( slave_addr << 1 ) | WRITE_BIT, ACK_CHECK_EN);
        i2c_master_stop(cmd);
        esp_err_t ret = i2c_master_cmd_begin(SCCB_I2C_PORT, cmd, 1000 / portTICK_RATE_MS);
        i2c_cmd_link_delete(cmd);
        if( ret == ESP_OK)
        {
           return slave_addr; // got response = valid slv_addr
        }
        slave_addr++;
    }
#else
    for (uint8_t i = 0; i < 0x7F; i++)
    {
        if (twi_writeTo(i, &slave_addr, 1, true) == 0)
        {
            break;
        }

        if (i!=0x7E)
        {
            vTaskDelay(10 / portTICK_PERIOD_MS); // Necessary for OV7725 camera (not for OV2640).
        }
    }

#endif
if (slave_addr==0x7F) slave_addr=0; // indicate not found
    return slave_addr;
}
`

xenpac, 2020-03-24:
sorry formating got garbeled....

closed 2021-09-19 by github-actions[bot]