espressif/esp32-camera #51: esp_camera_fb_get hangs on semaphoreTake

uname2219, 2019-07-22:
Hello,

I am using the project with an OV2640 sensor. Occasionally I noticed that my camera task would hang because esp_camera_fb_get was blocked trying to take the frame_ready semaphore. I have a heavily loaded ESP32 and am using one frame buffer in JPEG mode.

frame_Ready is set in camera_fb_done which is called in dma_finish_frame, which is called in dma_filter_task (with buf_idx = SIZE_MAX). That in turn is called in i2s_stop which (in JPEG mode) appears to only be called in vsync_isr.

By enabling some ets_printf statements and using a logic analyzer, I see that under heavy LwIP/TLS usage (on the other core), I sometimes get a bad header or the data_ready queue fills before filtering a line - setting the bad flag. This can happen at the beginning of the image in which case the dma_filtered_count remains at 0. Then since I am using 1 frame buffer, the conditional:

`if(s_state->dma_filtered_count > 1 || s_state->config.fb_count > 1) {
                i2s_stop(&need_yield);
            }`

at line 573 will never call i2s_stop. The bad flag is reset in dma_finish_frame which can only be called from i2s_stop so the original esp_camera_fb_get call never returns.

I'm afraid I don't understand the conditionals within vsync_isr but suspect a change is needed, at least in 1 frame buffer mode? When I naively removed the conditional so i2s_stop is always called, I no longer see a hang - even under artificially high CPU loads. I'm still not sure if the i2s_stop call should be under the dma_receieved_count check, since it would seem to me i2s_stop should be called anytime VSYNC goes low (signaling the end of an image).

Thank you for your help and your work on this library.

vitaminwater, 2019-08-09:
I think I have the same problem, it just keeps calling `vsync_isr` perpetually, and `ets_printf("end_vsync\n");` is printed, but it keeps looping there without returning the frame_buffer.

Another detail: it seems to only happen with resolutions above `FRAMESIZE_SVGA`.

me-no-dev, 2019-08-29:
please try the latest commit :)

mirronelli, 2019-08-29:
Great. Just out of curiosity. Is the value of 4000ms for the timeout always safe? Even when taking pictures in very low light conditions?

me-no-dev, 2019-08-30:
yes, 4 seconds is plenty even to have a few bad frames in between. I will move it to a config setting, but I picked this as being below the default WDT timeout (if you monitor a thread).

uname2219, 2019-09-03:
The fix appears to work and the code keeps running even after a bad frame is thrown away. Thank you for your help!

closed 2019-09-03 by uname2219