espressif/esp32-camera #54: esp_camera_deinit() not working?

playGit888, 2019-07-29:
Is it possible to deinit camera completely? or put the camera in power saving mode? 
When I init it again, I get some error about GPIO and i2s. 

If I keep OV2640 or OV3660 running, the camera module will "eat" about 100ma current.
While at the same time, the ESP32 is busy and also "eat" about 100ma current.
These will create a lot of heat.
Maybe I can try to put ESP32 into light sleep mode, but I am not sure the camera can be brought back to life with ESP32 at the same time. 

I saw this for esp_camera_init():
 * This function detects and configures camera over I2C interface,
 * allocates framebuffer and DMA buffers,
 * initializes parallel I2S input, and sets up DMA descriptors.
 *
 * Currently this function can only be called once and there is
 * no way to de-initialize this module.

I prefer to use Anduino IDE, but it seems that esp_camera_deinit() is not implement-ed, because I got more error when I tried to init it again..

Any suggestion? 

Sorry for my poor English.
Thank you in advance.

me-no-dev, 2019-07-29:
I will test a few methods to halt/power-down the camera and will implement something to allow for both. You could try to stop XCLK and see if the consumption will drop. To do that, you can forward declare the following C methods in your code and call them to stop/start XCLK https://github.com/espressif/esp32-camera/blob/master/driver/private_include/xclk.h

eried, 2019-08-07:
> B) After the camera was deinit, I set the camera power pin (and let the reset pin as it was) to high to shut it down.

Hey, how do you deinit the camera in arduino? or use the new driver? I am using the ESP32 CAM shield that does not has the RST connected, but setting PWR as HIGH still consumes 0.03A in sleep :/ I removed the 5V regulator for the board and just using 3.3 directly.

playGit888, 2019-08-14:
hi, eried, I tried to use arduino library before. I found that the deinit () not working. Then I turned to IDF SDK version: v4.0-dev-1346-g1b6010b. 
Somehow, deinit() is working in IDF SDK, but not perfect.
Maybe we have to wait for an upgrade of arduino library.
0.03A is too much. It should be nearly zero. 
OV2640_data_sheet.pdf says something about the power consumtion in shutdown mode. You can try to get mroe information from that pdf.
Or maybe you try to disconnect the camera module manually to confirm. I guess it will not burn the module or the board, because I did it before.

eried, 2019-08-14:
I think is the regulators, the 1.8V and 2.xV because if I run a blinker sketch and sleep it consume the same (without even initializing the OV2640)

dimhoff, 2019-08-15:
@eried To disable the voltage regulators in deep sleep you set the CAM_PWR to HIGH and call rtc_gpio_hold_en() and gpio_deep_sleep_hold_en().

eried, 2019-08-16:
Thanks, I'll try tonight. Something like this should work right?:

```
digitalWrite(PWDN_GPIO_NUM, HIGH);
gpio_deep_sleep_hold_en();
gpio_hold_en(/*gpio_num_t*/ PWDN_GPIO_NUM);
esp_sleep_enable_timer_wakeup(mins * 60 * uS_TO_S_FACTOR); // it is actually using 4.9 mA at 3.3V
Serial.flush();

// Going to sleep
esp_deep_sleep_start();
Serial.println("This will never be printed");
```

eried, 2019-08-16:
> @eried To disable the voltage regulators in deep sleep you set the CAM_PWR to HIGH and call rtc_gpio_hold_en() and gpio_deep_sleep_hold_en().

It actually uses more doing this :/ 0.006A, without that code: 0.004A. With that pin LOW 0.126A

pierreh, 2020-03-13:
I think I have a similar problem, when I restart the board in software (after a OTA update) the camera won't init again and I get a 0x20004 error. I've tried esp_camera_deinit, I've tried to pull PWDN_GPIO_NUM to HIGH for 30s and then LOW again to try and reset. but no luck. The only way to get the camera to work again is press the reset button. What can I do? I need the camera to be able to recover without having to go there to manually reset it.
I have a ESP32-cam board.

closed 2021-09-19 by github-actions[bot]