espressif/esp32-camera #55: Adjust brightness while taking picture

eried, 2019-08-08:
Is there is any way to auto-adjust the brightest on a picture? I am using arduino and my outside photos are all white :/

me-no-dev, 2019-08-08:
which board? which sensor? which example? there are some "auto" things enabled, but sometimes you need to switch things on and off yourself. All relevant methods are exposed.

eried, 2019-08-08:
Hey, thanks. But I am wondering if those "auto" things are trigger automatically on a still? or do I have to "open the shutter" wait and then something?

My board is the ebay cheapest ESP32 CAM (AI THINKER with OV2640)  https://www.ebay.com/itm/UK-ESP32-CAM-ESP32-WiFi-Bluetooth-Development-Board-With-Camera-Module-OV2640/372731369427?hash=item56c88557d3:g:vRcAAOSwEC5dQqwx

me-no-dev, 2019-08-09:
all auto settings run auto if enabled. XCLK is running the camera and in the matter of a few frame it should adjust to the ambient conditions.

eried, 2019-08-10:
:/ it does not, for example:

[image]

some images are super overexposed. I am taking 10 photos and discarding the first 9 and saving the last, but it seems a bad approach. I guess I could decrease brightness manually by reading some pixels maybe? any suggestion?

me-no-dev, 2019-08-10:
I will suggest that you run the web server example and see what settings will work the best for you. It has all the controls available and you can toggle things. Generally it's a bit odd that you get such burn... not my experience.

cgmckeever, 2020-07-31:
@eried Any luck? When I take a single shot outdoor (sunny) I get this burn as well. If I stream video, it takes about 2-3 seconds to come in clean. Not sure how to compensate for that in a still, unless I take a few pictures before keeping the last ?

eried, 2020-08-01:
Nah :/ I did not find any solution

BBwally, 2020-11-01:
Same problem ... in a sketch that deep sleeps between being awoken by either PIR or timer (for time lapse). Pictures out the window are all way overexposed, and the camera does not seem to auto-adjust over time. Manually setting the exposure to -2 also seemed to have no effect.

cgmckeever, 2020-11-01:
I have had mixed results with taking a second picture moments after the first

BBwally, 2020-11-01:
> I have had mixed results with taking a second picture moments after the first

Thanks for the tip ... I’ll give that a try.

BBwally, 2020-11-05:
I now seem to be getting quite acceptable outdoor results (see sample) by injecting a delay after the initial camera configuration. Here's the code sequence (where cdelay is a value in seconds uploaded from a config.txt file) ...

    esp_err_t err = esp_camera_init(&config);
    if (err != ESP_OK) {
      bailout ("Camera init failed with error " + String (err));
    }
    camera_fb_t * fb = NULL;
    if (cdelay>0) {
      delay (cdelay*1000);
    }
    fb = esp_camera_fb_get ();

[image]

closed 2021-09-19 by github-actions[bot]

kjm1102, 2022-06-13:
How long did you to pause after the config BB? I'm delaying 5s & the outside pics are as over exposed as ever!