espressif/esp32-camera #58: Can anyone describe these sensor settings?

bkeevil, 2019-08-20:


bkeevil, 2019-08-22:
bpc and wpc could be black pixel correction and white pixel correction

ithinkido, 2019-08-22:
Aec would be automatic exposure correction I guess, taking values from the web page menu. Aec2... Not sure.

mirronelli, 2019-08-28:
try looking up the register definition of the ov2640 camera module, that is where these values go..
https://www.uctronics.com/download/cam_module/OV2640DS.pdf

bkeevil, 2019-08-29:
Yeah these ones are not in the preliminary datasheet. Omnivision has not been forthcoming with datasheets. If someone comes across a copy of the final datasheet for this module please share it.

closed 2020-01-24 by bkeevil

bakeromso, 2024-08-08:
> Yeah these ones are not in the preliminary datasheet. Omnivision has not been forthcoming with datasheets. If someone comes across a copy of the final datasheet for this module please share it.

The latest I found is final v2.21, see [here](https://www.plexishop.it/pdf/OV2640.pdf)