espressif/esp32-camera #60: Does esp32 camera slow down WiFi TCP transmission?

RoyHsu67, 2019-08-22:
Hi~ I meet a situation. I can smoothly send data and image over TCP by esp32 WiFi module, but when I use esp32 camera to get a photo, once I call esp_camera_init(), TCP transmission becomes so much slow. A way to get better is to initialize camera driver, get a camera image, then immediately deinitialize camera driver, so I can recover TCP transmission speed and transfer image out. Is this normal?

closed 2021-09-19 by github-actions[bot]

kxbin, 2022-07-25:
I had the same problem, my wifi upload speed was only up to 30KB/s, when the camera started working

AxelLin, 2023-12-08:
Hi @kxbin @RoyHsu67 

Do you have solution for this issue?

I hit the same issue, the wifi transmission is very slow after esp_camera_init().
After calling esp_camera_deinit(), the wifi transmission looks fine.