espressif/esp32-camera #62: Failed to flash onto ESP32-Cam

jiapei100, 2019-08-23:
Failed to flash after having successfully compiled **CameraWebServer** .

```console
Sketch uses 2241942 bytes (71%) of program storage space. Maximum is 3145728 bytes.
Global variables use 52696 bytes (16%) of dynamic memory, leaving 274984 bytes for local variables. Maximum is 327680 bytes.
esptool.py v2.6
Serial port /dev/ttyUSB0
Connecting........_____....._____....._____....._____....._____....._____.....____An error occurred while uploading the sketch
_

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
```

Any suggestions?

mirronelli, 2019-08-29:
Try lowering the baudrate for flashing. 115200 is the safest bet but it works almost everytime. If it works try to go higher at 230400 or 460800.

DeboisDeontronseantray, 2019-09-06:
Make sure you’re using a cable that can transfer data at high speeds.

closed 2021-09-19 by github-actions[bot]