espressif/esp32-camera #64: OV2640 poor image quality if resoultion <= CIF

moononournation, 2019-08-26:
It is very different on the image quality, especially on the smoothness and color saturation when I set between the resolution of CIF and VGA. I think it may caused by the setting of ov2640_settings_cif or CLKRC_2X_CIF.

me-no-dev, 2019-08-26:
2640 supports 3 main resolutions: 1600x1200, 800x600 and 400x300. All other resolutions are divided from that. The more pixels you sample, the better image you get. When you run at CIF or lower, you get 4 times less light data than 800x600 and brightness depends directly on the number lines (you can see a control for it on the web interface).
Clock also plays a role, since lower clock will give more scan time, therefore more light into the sensor. It does get easily blurred if something is moving though.

moononournation, 2019-08-27:
Not exactly depends on number of pixels. Below images are capture by the esp32-camera web interface. As you can see the color saturation is near the same from qqvga to cif, but color become much saturated on vga image:
[image]
[image]
[image]
[image]
[image]

closed 2021-09-19 by github-actions[bot]