espressif/esp32-camera #65: Motion Detection

billyrudi, 2019-08-26:
Hi, is it possible to detect motion only using video not pir or external sensor?
How i can detect if the frame video  is changing ?
regards Paolo

ithinkido, 2019-08-26:
https://github.com/raphaelbs/esp32-cam-ai-thinker/blob/master/examples/change_detection/README.md

billyrudi, 2019-08-26:
Tank you. Is it compatible with arduino ide?

bluguru4567, 2019-08-29:
While my ESP32-CAM is waiting for a client to connect to stream video, I'd like it to be a motion detector and email me when it detects motion.  I wonder what would work better:  1) comparing snapshots for change (and possibly missing something while processing or doing other actions), or 
2) using a PIR.   I love the idea of not using the PIR, but I guess I'd like to test both.

bkeevil, 2019-08-29:
I was actually intending to do something similar with [my webcam code](https://github.com/bkeevil/esp32-cam) if you want to compare notes. I will be starting the software motion detect feature soon.

bkeevil, 2019-08-29:
I was thinking of creating a FreeRTOS task that retrieved a frame at a low resolution using YUV422 pixel encoding and comparing the individual pixels. The frame rate does not need to be very high.

alanesq, 2020-02-17:
I have done something along these lines recently which may be of interest:
https://github.com/alanesq/CameraWifiMotion

It is based on the motion detecting code created by https://eloquentarduino.github.io/2020/01/motion-detection-with-esp32-cam-only-arduino-version/
although I have modified it a lot.

bkeevil, 2020-02-18:
Thank you @alanesq that is perfect. I have saved those links in my projects README.md file.

alanesq, 2020-02-19:
Thanks

btw - I find these cameras are very poor in the dark but I have just removed the infra red filter from one and this improves it considerably.
I warmed the camera with hot air to soften the glue and then unscrew the lens and the filter is a small (I think plastic) disc

closed 2021-09-20 by github-actions[bot]