espressif/esp32-camera #66: LEDc configuration error and camera_probe failed

masterd89, 2019-08-27:
Hello there,

i have an issue with ESP32-CAM module with OV2640:

E (1582) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=3
E (1592) camera_xclk: ledc_timer_config failed, rc=ffffffff
I (1602) sccb: pin_sda 26 pin_scl 27

I (1602) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (1662) camera: Detected camera not supported.
E (1662) camera: Camera probe failed with error 0x20004
E (1662) app_camera: Camera init failed with error 0x20004

I've tried everything: enable/disable all camera module support and hw I2C from menuconfig, tried with 3 different unit and different cameras. Tried custom firmware and examples firmware, examined the code and the pin configuration, different USB to serial boards, powering the boards with external PSU.....and yet i still get that error.
I also updated esp-idf framework by deleting and re cloning the repo. No luck with that also.

Any idea?
Thanks,
D

mirronelli, 2019-08-28:
there was a change in ledc driver that is used to drive the clock 

to work with the current version a one line of code needs to be added into xclk.c
    ```	ledc_timer_config_t timer_conf;
	timer_conf.duty_resolution = 2;
	timer_conf.freq_hz = config->xclk_freq_hz;
	timer_conf.speed_mode = LEDC_HIGH_SPEED_MODE;
	timer_conf.timer_num = config->ledc_timer;
	**timer_conf.clk_cfg = LEDC_USE_APB_CLK;**
	esp_err_t err = ledc_timer_config(&timer_conf);
	if (err != ESP_OK) {
		ESP_LOGE(TAG, "ledc_timer_config failed, rc=%x", err);
		return err;
	}
```

masterd89, 2019-08-29:
Thanks @mirronelli, i already figured it out though (while i used LEDC_AUTO_CLK).

Strange thing is, if i ran the ledc initialization snippet (whitout setting the clock source) from the app-main it didn't give any error at all.

Moreover, while this solution solves the duty cycle - frequency error, in my case it still hangs in "skip_frame()" function, thowing a timeout error. I found out this occurred because the VSYNC pin wasn't actually clocking, even though it was correctly initialized. 

I solved it by moving the pheripheral enable command at the end of the function, after the parameters assignment:

    ledc_timer_config_t timer_conf;
    timer_conf.duty_resolution = 2;
    timer_conf.freq_hz = config->xclk_freq_hz;
    timer_conf.speed_mode = LEDC_HIGH_SPEED_MODE;
    timer_conf.timer_num = config->ledc_timer;
    ESP_LOGI(TAG,"dr = %d, fr = %d", timer_conf.duty_resolution, timer_conf.freq_hz);
    timer_conf.clk_cfg = LEDC_AUTO_CLK;
    esp_err_t err = ledc_timer_config(&timer_conf);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "ledc_timer_config failed, rc=%x", err);
        return err;
    }

    ledc_channel_config_t ch_conf;
    ch_conf.gpio_num = config->pin_xclk;
    ch_conf.speed_mode = LEDC_HIGH_SPEED_MODE;
    ch_conf.channel = config->ledc_channel;
    ch_conf.intr_type = LEDC_INTR_DISABLE;
    ch_conf.timer_sel = config->ledc_timer;
    ch_conf.duty = 2;
    ch_conf.hpoint = 0;
    err = ledc_channel_config(&ch_conf);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "ledc_channel_config failed, rc=%x", err);
        return err;
    }
    periph_module_enable(PERIPH_LEDC_MODULE);
    return ESP_OK;

The other issue i found (the one that caused error 0x20004), which is however related to the circuit, is that camera wasn't responding to i2c commands, thus preventing the software to recognize the correct model.

ESP32-CAM board has camera PWDN pin tied to GND (so camera is always powered) while camera RST is passively deasserted a little time afte 3.3V supply is started. However, GPIO32 controls the power supply enable of the camera by means of a transistor. Thus, if GPIO32 is not set low before calling camera_init(), the camera is powered down and cannot reply to i2c commands of course. This solved the issue:

    //initialize the camera
	gpio_config_t gpio_pwr_config;
	gpio_pwr_config.pin_bit_mask = (1ULL << 32);
	gpio_pwr_config.mode = GPIO_MODE_OUTPUT;
	gpio_pwr_config.pull_down_en = GPIO_PULLDOWN_DISABLE;
	gpio_pwr_config.pull_up_en = GPIO_PULLUP_DISABLE;
	gpio_pwr_config.intr_type = GPIO_INTR_DISABLE;
	gpio_config(&gpio_pwr_config);
	gpio_set_level(32,0);
	vTaskDelay(10/ portTICK_PERIOD_MS);

    esp_err_t err = esp_camera_init(&camera_config);

jxyxg, 2020-08-13:
Hello, sorry to bother you guys, now i am writing a driver code of the ov2640 with esp32cam, but actually i have no idea and donnot know how to start and i have found that there is few material about writing a driver code from scratch, I wondering whether you have  some experience or have some useful tutorial material, can you share them with me ? Many  thanks ! @masterd89 @masterd89

jxyxg, 2020-08-13:
Hello, sorry to bother you guys, now i am writing a driver code of the ov2640 with esp32cam, but actually i have no idea and donnot know how to start and i have found that there is few material about writing a driver code from scratch, I wondering whether you have some experience or have some useful tutorial material, can you share them with me ? Many thanks ! @masterd89 @mirronelli

Patsjemoe, 2020-08-16:
Hi,
I have the same error, I checked the xclk file, and the line referred to is in the file xclk.c :     timer_conf.clk_cfg = LEDC_USE_APB_CLK; 

but still get following error : 
Python requirements from C:/msys32/path/esp/esp-idf/requirements.txt are satisfied.
CC build/esp32-camera/driver/xclk.o
C:/msys32/path/esp/nabto-esp32cam/components/esp32-camera/driver/xclk.c: In function 'camera_enable_out_clock':
C:/msys32/path/esp/nabto-esp32cam/components/esp32-camera/driver/xclk.c:23:15: error: 'ledc_timer_config_t {aka struct <anonymous>}' has no member named 'clk_cfg'
     timer_conf.clk_cfg = LEDC_USE_APB_CLK;
               ^
C:/msys32/path/esp/nabto-esp32cam/components/esp32-camera/driver/xclk.c:23:26: error: 'LEDC_USE_APB_CLK' undeclared (first use in this function)
     timer_conf.clk_cfg = LEDC_USE_APB_CLK;
                          ^
C:/msys32/path/esp/nabto-esp32cam/components/esp32-camera/driver/xclk.c:23:26: note: each undeclared identifier is reported only once for each function it appears in
make[1]: *** [/c/msys32/path/esp/esp-idf/make/component_wrapper.mk:286: driver/xclk.o] Fout 1
make: *** [/c/msys32/path/esp/esp-idf/make/project.mk:530: component-esp32-camera-build] Fout 2
I have checked the esp-idf path, and is OK.
any suggestions ?
regards
ludo

gdamoreira, 2021-04-29:
Hi everyone,

I'm facing the same problem with my ESP32-CAM, here are the details:

**Board:** ESP32-CAM (ESP32S)
**Camera module:** TY-OV2640-V2.0
**USB adapter:** ESP32-CAM-MB

I'm using CameraWebServer example with AI_THINKER model defined:
```
// Select camera model
//#define CAMERA_MODEL_WROVER_KIT // Has PSRAM
//#define CAMERA_MODEL_ESP_EYE // Has PSRAM
//#define CAMERA_MODEL_M5STACK_PSRAM // Has PSRAM
//#define CAMERA_MODEL_M5STACK_V2_PSRAM // M5Camera version B Has PSRAM
//#define CAMERA_MODEL_M5STACK_WIDE // Has PSRAM
//#define CAMERA_MODEL_M5STACK_ESP32CAM // No PSRAM
#define CAMERA_MODEL_AI_THINKER // Has PSRAM
//#define CAMERA_MODEL_TTGO_T_JOURNAL // No PSRAM
```

Some comments suggested to change the USB cable, but the problem persists.
[image]
[image]

_Serial output:_
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4

Initializing camera config
[E][camera.c:1113] camera_probe(): Detected camera not supported.
[E][camera.c:1379] esp_camera_init(): Camera probe failed with error 0x20004
```

won0-kim, 2021-06-23:
// Select camera model
// #define CAMERA_MODEL_WROVER_KIT // Has PSRAM
//#define CAMERA_MODEL_ESP_EYE // Has PSRAM
//#define CAMERA_MODEL_M5STACK_PSRAM // Has PSRAM
//#define CAMERA_MODEL_M5STACK_V2_PSRAM // M5Camera version B Has PSRAM
//#define CAMERA_MODEL_M5STACK_WIDE // Has PSRAM
//#define CAMERA_MODEL_M5STACK_ESP32CAM // No PSRAM
**#define CAMERA_MODEL_AI_THINKER // Has PSRAM**
//#define CAMERA_MODEL_TTGO_T_JOURNAL // No PSRAM

@jxyxg I think.. this problem could be solved easily by only changing these comments like that.

closed 2021-09-19 by github-actions[bot]

yugandharc9, 2021-12-24:
In my case I had to comment 
`#define CAMERA_MODEL_WROVER_KIT // Has PSRAM
`
and uncomment 

`#define CAMERA_MODEL_AI_THINKER // Has PSRAM
`

Morchack, 2022-01-04:
> Obrigado @mirronelli , eu já descobri (enquanto usava LEDC_AUTO_CLK).
> 
> O estranho é que, se eu executei o snippet de inicialização do ledc (sem definir a fonte do relógio) do app-main, não deu nenhum erro.
> 
> Além disso, embora esta solução resolva o erro de ciclo de trabalho - frequência, no meu caso ela ainda trava na função "skip_frame ()", resultando em um erro de tempo limite. Descobri que isso ocorreu porque o pino VSYNC não estava realmente sincronizando, embora tenha sido inicializado corretamente.
> 
> Resolvi movendo o comando de habilitação de pheripheral no final da função, após a atribuição dos parâmetros:
> 
> ```
> ledc_timer_config_t timer_conf;
> timer_conf.duty_resolution = 2;
> timer_conf.freq_hz = config->xclk_freq_hz;
> timer_conf.speed_mode = LEDC_HIGH_SPEED_MODE;
> timer_conf.timer_num = config->ledc_timer;
> ESP_LOGI(TAG,"dr = %d, fr = %d", timer_conf.duty_resolution, timer_conf.freq_hz);
> timer_conf.clk_cfg = LEDC_AUTO_CLK;
> esp_err_t err = ledc_timer_config(&timer_conf);
> if (err != ESP_OK) {
>     ESP_LOGE(TAG, "ledc_timer_config failed, rc=%x", err);
>     return err;
> }
> 
> ledc_channel_config_t ch_conf;
> ch_conf.gpio_num = config->pin_xclk;
> ch_conf.speed_mode = LEDC_HIGH_SPEED_MODE;
> ch_conf.channel = config->ledc_channel;
> ch_conf.intr_type = LEDC_INTR_DISABLE;
> ch_conf.timer_sel = config->ledc_timer;
> ch_conf.duty = 2;
> ch_conf.hpoint = 0;
> err = ledc_channel_config(&ch_conf);
> if (err != ESP_OK) {
>     ESP_LOGE(TAG, "ledc_channel_config failed, rc=%x", err);
>     return err;
> }
> periph_module_enable(PERIPH_LEDC_MODULE);
> return ESP_OK;
> ```
> 
> O outro problema que encontrei (o que causou o erro 0x20004), porém relacionado ao circuito, é que a câmera não respondia aos comandos do i2c, impedindo que o software reconhecesse o modelo correto.
> 
> A placa ESP32-CAM tem o pino PWDN da câmera ligado ao GND (para que a câmera esteja sempre ligada), enquanto a câmera RST é desativada passivamente um pouco depois que o fornecimento de 3,3 V é iniciado. No entanto, o GPIO32 controla a ativação da fonte de alimentação da câmera por meio de um transistor. Portanto, se o GPIO32 não estiver definido como baixo antes de chamar camera_init (), a câmera será desligada e não poderá responder aos comandos i2c, é claro. Isso resolveu o problema:
> 
> ```
> //initialize the camera
> gpio_config_t gpio_pwr_config;
> gpio_pwr_config.pin_bit_mask = (1ULL << 32);
> gpio_pwr_config.mode = GPIO_MODE_OUTPUT;
> gpio_pwr_config.pull_down_en = GPIO_PULLDOWN_DISABLE;
> gpio_pwr_config.pull_up_en = GPIO_PULLUP_DISABLE;
> gpio_pwr_config.intr_type = GPIO_INTR_DISABLE;
> gpio_config(&gpio_pwr_config);
> gpio_set_level(32,0);
> vTaskDelay(10/ portTICK_PERIOD_MS);
> 
> esp_err_t err = esp_camera_init(&camera_config);
> ```

Morchack, 2022-01-04:
ola eu também tenho este problema , em que parte do código é que eu devo colocar este novo código que colocou aqui em cima.