espressif/esp32-camera #74: Detected camera not supported

waldorffsmama, 2019-09-17:
I have an ESP32-S Cam module with the OV2640 cam module ([link](https://www.aliexpress.com/item/32992663411.html?spm=a2g0s.9042311.0.0.1bc363c0tjdkke)), flashed it with the Arduino IDE as an AI-Thinker ESP-32 Cam module and also as an ESP32-Wrover module. Unfortunately, I always get the same errors:

[E][camera.c:1049] camera_probe(): Detected camera not supported.
[E][camera.c:1249] esp_camera_init(): Camera probe failed with error 0x20004

At some threads with a similiar issue it was recommended to run the ESP32 with 5V. Others recommended pulling out the sensorcable from the slot, re-inserting and restarting. None of that helped. Any ideas what else can be tried?

waldorffsmama, 2019-09-17:
Ahh.... I did not know that I had to select the cam in the sketch, too. Once I had selected "AI Thinker Module" also in the sketch, it worked like a charme. Sorry for the inconvenience.

closed 2019-09-17 by waldorffsmama

mbt45, 2020-01-23:
Hi, what did you change in the code? I have the same issue but I don't understand your solution

waldorffsmama, 2020-01-23:
In the CameraWebserver file at the very top you will see

`// Select camera model
//,0#define CAMERA_MODEL_WROVER_KIT
//#define CAMERA_MODEL_ESP_EYE
//#define CAMERA_MODEL_M5STACK_PSRAM
//#define CAMERA_MODEL_M5STACK_WIDE
#define CAMERA_MODEL_AI_THINKER
`
Depending on which model you are using, you need to comment or uncomment by adding or deleting  // in the corresponding lines.

lhadvab, 2020-02-06:
Hi. I have same problem. I have already tried every module, but It´s always same error. Can you help me?

refat37, 2020-02-25:
[image]
I have facing same problen

DoddyD, 2020-02-27:
I've 2 pcs of this ESP32-Cam with different brand. 

what I did 
1. try using all board selection
2. try change USB cable
3. try different supply 3.3V and 5 v
4. try different converter Arduino UNO, USB to TTL converter  

But still facing the same problem : 

[E][camera.c:1049] camera_probe(): Detected camera not supported.
[E][camera.c:1249] esp_camera_init(): Camera probe failed with error 0x20004

Please let me know if anybody fixed this issue.

refat37, 2020-02-27:
How can we overcome this problem?

me-no-dev, 2020-02-28:
the error usually means that SDA and SCL for the camera are not working correctly

refat37, 2020-02-28:
Exactly😍But i can't find scl gpio pin😭They are connected sck

me-no-dev, 2020-02-28:
which camera board? did you select the proper board in the pin config?

refat37, 2020-02-29:
[image]

MegaManny, 2020-03-05:
I am facing similar issue. The module was not giving error at first and also started the camera server. Then I uploaded the code again with just changing the ssid, after that this error started to show.
I think one of the two or both(the esp32 or the camera) has been damage, as the GPIO_0 pin that i used to program is connected to the XCLK of the camera.
Is there a way to test my camera and esp32 separately? or should i buy a new pair.
I am using AI Thinker ESP32-CAM

cjanner, 2020-03-06:
Got the same error with 3.3V. After switch to 5V it worked.

hacwa, 2020-03-29:
Thanks, cjanner.

switched to 5V and worked without a problem

MildaGenius, 2020-04-10:
I got, this error after 20 iteration of deep sleep and timer wakeup, with time delay 10s.
Iam operating on 5V.

Bighoneypot, 2020-04-29:
I have solved this problem using the PC usb port without hub.
Form good performance use a PowerBank

mazjed, 2020-05-06:
Hi i came across this error with the same board . the cause was in the sketch i was using
#define CAMERA_MODEL_WROVER_KIT was enabled by default i had to add // like this
//#define CAMERA_MODEL_WROVER_KIT to disable it

kassi, 2020-05-27:
I have a similar error, but I still can't get it running. CAMERA_MODEL_WROVER_KIT results in the 0x20004 error. CAMERA_MODEL_AI_THINKER doesn't give any response, just showing up dots and dots.

    19:16:19.197 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
    19:16:19.197 -> configsip: 0, SPIWP:0xee
    19:16:19.231 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
    19:16:19.231 -> mode:DIO, clock div:1
    19:16:19.231 -> load:0x3fff0018,len:4
    19:16:19.231 -> load:0x3fff001c,len:1216
    19:16:19.231 -> ho 0 tail 12 room 4
    19:16:19.231 -> load:0x40078000,len:9720
    19:16:19.231 -> ho 0 tail 12 room 4
    19:16:19.231 -> load:0x40080400,len:6352
    19:16:19.231 -> entry 0x400806b8
    19:16:21.258 -> 
    19:16:22.541 -> ....................................................................................... <- never finishes

Every other module in the example code gives a compilation error

    esp32-cam:44:25: error: 'SIO4C_GPIO_NUM' was not declared in this scope

tomluken, 2020-06-05:
im new but i made it work using 5 v on the fdti and 5 v on the esp32 cam
hope this helps someone

Andy1968T, 2020-06-16:
Solved ........
Basically add a define the for the pins correctly for T-Camera PLUS the demo and create.
You can also do this for any other camera esp32 module that you know the pin GPIOS for with a OV2640  camera module thus adding new cameras to the demo.
 For the TTGO T-CAMERA PLUS Add this to the camera_pins.h file between two other module type defines 
#elif defined(CAMERA_MODEL_TTGO_TCAMERA_PLUS)
#define PWDN_GPIO_NUM    -1
#define RESET_GPIO_NUM   -1
#define XCLK_GPIO_NUM    4
#define SIOD_GPIO_NUM    18
#define SIOC_GPIO_NUM    23

#define Y9_GPIO_NUM      36
#define Y8_GPIO_NUM      37
#define Y7_GPIO_NUM      38
#define Y6_GPIO_NUM      39
#define Y5_GPIO_NUM      35
#define Y4_GPIO_NUM      26
#define Y3_GPIO_NUM      13
#define Y2_GPIO_NUM      34
#define VSYNC_GPIO_NUM   5
#define HREF_GPIO_NUM    27
#define PCLK_GPIO_NUM    25
 then  comment out all the other board #defines and add your  new define in the camerawebserver demo program if thats the code you are using so that it looks like this below
// Select camera model
//#define CAMERA_MODEL_WROVER_KIT
//#define CAMERA_MODEL_ESP_EYE
//#define CAMERA_MODEL_M5STACK_PSRAM
//#define CAMERA_MODEL_M5STACK_WIDE
//#define CAMERA_MODEL_AI_THINKER
#define CAMERA_MODEL_TTGO_TCAMERA_PLUS

P.S esp32 devices are 3.3v do not run at any other voltage

Andy1968T, 2020-06-16:
Now that PINs are defined correctly and the demo is working . I am finding the ESP32, camera , camerawebserver demo. does crash randomly . And with previous experience over other esp32 D1 having power supply issues. this might be the cause of the random crashes. So will be testing further with a PI 3A adaptor to test the module is getting enough juice.

Andy1968T, 2020-06-16:
kassi . if you are getting ..............................................
this would suggest camera is ok and found.  but the routine is trying to connect to your wifi and failing. Maybe check your SID and password?

papipascal, 2020-06-19:
dear All
Same issue...
i try power slection, add the tempo before WIFI (but  program stop before!), change for TTGO cam, same issue with my 3 boards !! : They doesn't work at all... I've tried to remove camera... 
Je suis perdu !

papipascal, 2020-06-20:
[E][camera.c:1049] camera_probe(): Detected camera not supported.
[E][camera.c:1249] esp_camera_init(): Camera probe failed with error 0x20004

I don't progress after all tested solutions I found in web...

Schaggo, 2020-06-20:
2004 solutions:

Most times there’s a problem with chip recognition during startup. It seems to be a problem connected to the 5V regulator. Solution: hard reboot.

Occasionally you have to erase the flash and reprogram. I don’t know why, but I’ve had situations where it actually resolved a 2004. 

Rarely (4/70) the camera is broken.

abhibhatia98, 2020-07-31:
> [E][camera.c:1049] camera_probe(): Detected camera not supported.
> [E][camera.c:1249] esp_camera_init(): Camera probe failed with error 0x20004
> 
> I don't progress after all tested solutions I found in web...

@papipascal  have u found solution ? if yes please help me

kholia, 2020-07-31:
> Rarely (4/70) the camera is broken.

https://github.com/espressif/esp32-camera/issues/43#issuecomment-656110289 can help in a few such cases.

ncastil6, 2020-08-12:
> kassi . if you are getting ..............................................
> this would suggest camera is ok and found. but the routine is trying to connect to your wifi and failing. Maybe check your SID and password?

This was the fix for me, I had the wrong WiFi information, once corrected the device worked properly and I could stream with the cam! 

Thx Andy1968T!

kassi, 2020-08-13:
@Andy1968T and @ncastil6 yes, I made it finally. I was confused with the indicators because beforehand I got an error regarding camera.

Thyssenb, 2020-09-09:
I was experiencing the same problems, what solved the problem is switching the voltage pin of FDTI programmer and the voltage pin on ESP32 to 5v. Solved quickly enough. Really nice standard lib included as well. Nice little device.

abubakas, 2021-01-26:
Change Vin to 5V and by rearrange the code as below, mine works like charms.

[image]

Cocacola2L, 2021-02-05:
Just in case there are also goofy people like me. It turned out that my problem was that the OV2640 camera was not placed deep enough into the ESP32-CAM module. I just pushed it further in and it worked. :D

tennessine, 2021-05-08:
in my case
```
// Select camera model
// comment the line 
//#define CAMERA_MODEL_WROVER_KIT
//#define CAMERA_MODEL_ESP_EYE
//#define CAMERA_MODEL_M5STACK_PSRAM
//#define CAMERA_MODEL_M5STACK_WIDE
// uncomment the line
#define CAMERA_MODEL_AI_THINKER
```

j3ster2019, 2022-01-02:
1.  #define CAMERA_MODEL_AI_THINKER
2. camera_index.h , comment out index_ov3660.html.gz, Size: 4408
3. ammend this
 static esp_err_t index_handler(httpd_req_t *req){
    httpd_resp_set_type(req, "text/html");
    httpd_resp_set_hdr(req, "Content-Encoding", "gzip");
    sensor_t * s = esp_camera_sensor_get();
    **if (s->id.PID == OV3660_PID) {
        return httpd_resp_send(req, (const char *)index_ov2640_html_gz, index_ov2640_html_gz_len);
    }
    return httpd_resp_send(req, (const char *)index_ov2640_html_gz, index_ov2640_html_gz_len);**

jomjol, 2022-01-02:
?

acuarioasaa, 2023-07-13:
> Tengo un módulo de cámara ESP32-S con el módulo de cámara OV2640 ( [enlace](https://www.aliexpress.com/item/32992663411.html?spm=a2g0s.9042311.0.0.1bc363c0tjdkke) ), lo actualicé con Arduino IDE como un módulo de cámara AI-Thinker ESP-32 y también como un módulo ESP32-Wrover. Desafortunadamente, siempre obtengo los mismos errores:
> 
> [E][camera.c:1049] camera_probe(): la cámara detectada no es compatible. [E][camera.c:1249] esp_camera_init(): la sonda de la cámara falló con el error 0x20004
> 
> En algunos hilos con un problema similar, se recomendó ejecutar el ESP32 con 5V. Otros recomendaron sacar el cable del sensor de la ranura, volver a insertarlo y reiniciarlo. Nada de eso ayudó. ¿Alguna idea de qué más se puede probar?

SOLUCION QUE ME FUNCIONO:

PRUEBA DE CONEXION a WIFFI ELEGIR PLACA: (DOIT ESP32 DEVKIT V1)
	A. Herramientas (Tools)
	B. Placa (Board)
	C. esp32
	D. DOIT ESP32 DEVKIT V1
	E. Cargar el codigo que se encuentra: Ejemplos - Wifi - WiFiScan
Asi sabemos que red admite y elegimos a la que deseamos conectarnos.

mubeen1020, 2023-11-06:
I am facing this error
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6364
entry 0x400806b8
WELLCOME
SPIFFS initialization working.
[E][camera.c:1049] camera_probe(): Detected camera not supported.
[E][camera.c:1249] esp_camera_init(): Camera probe failed with error 0x20004
Raw Door State: 0
The door is closed
Raw Door State: 0
The door is closed
Raw Door State: 0
The door is closed
Raw Door State: 0
The door is closed
Raw Door State: 0
The door is closed
Raw Door State: 0
The door is closed
Raw Door State: 0
The door is closed
Raw Door State: 0
The door is closed
Raw Door State: 1
The door is open
Camera capture failed
Raw Door State: 1
The door is open
Camera capture failed

d3k4-k3rb3r0s, 2024-04-05:
Arduino ide doesnteven recognize or show rover module or ai thinker anymore.