espressif/esp32-camera #75: Guru Meditation Error when calling frame2bmp()

terrorzombie666, 2019-09-20:
I am trying to convert the taken image to BMP using the BMP HTTP Capture example from [README.md](https://github.com/espressif/esp32-camera/blob/master/README.md).

My device: ESP32-CAM

This is my Code:
```
#include "esp_camera.h"

#define CAMERA_MODEL_AI_THINKER
#include "camera_pins.h"

void setupCamera() {
  camera_config_t config;
  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer = LEDC_TIMER_0;
  config.pin_d0 = Y2_GPIO_NUM;
  config.pin_d1 = Y3_GPIO_NUM;
  config.pin_d2 = Y4_GPIO_NUM;
  config.pin_d3 = Y5_GPIO_NUM;
  config.pin_d4 = Y6_GPIO_NUM;
  config.pin_d5 = Y7_GPIO_NUM;
  config.pin_d6 = Y8_GPIO_NUM;
  config.pin_d7 = Y9_GPIO_NUM;
  config.pin_xclk = XCLK_GPIO_NUM;
  config.pin_pclk = PCLK_GPIO_NUM;
  config.pin_vsync = VSYNC_GPIO_NUM;
  config.pin_href = HREF_GPIO_NUM;
  config.pin_sscb_sda = SIOD_GPIO_NUM;
  config.pin_sscb_scl = SIOC_GPIO_NUM;
  config.pin_pwdn = PWDN_GPIO_NUM;
  config.pin_reset = RESET_GPIO_NUM;
  config.xclk_freq_hz = 20000000;
  config.pixel_format = PIXFORMAT_JPEG;

  //init with high specs to pre-allocate larger buffers
  if (psramFound()) {
    config.frame_size = FRAMESIZE_UXGA;
    config.jpeg_quality = 10;
    config.fb_count = 2;
  } else {
    config.frame_size = FRAMESIZE_SVGA;
    config.jpeg_quality = 12;
    config.fb_count = 1;
  }

  // camera init
  esp_err_t err = esp_camera_init(&config);
  if (err != ESP_OK) {
    Serial.printf("Camera init failed with error 0x%x", err);
    return;
  }

  sensor_t * s = esp_camera_sensor_get();
  //initial sensors are flipped vertically and colors are a bit saturated
  if (s->id.PID == OV3660_PID) {
    Serial.println("Do we get here at all?");
    s->set_vflip(s, 1);//flip it back
    s->set_brightness(s, 1);//up the blightness just a bit
    s->set_saturation(s, -2);//lower the saturation
  }
  s->set_framesize(s, FRAMESIZE_UXGA); //FRAMESIZE_SXGA
}

uint8_t* takeImageBmp(){
    camera_fb_t * fb = NULL;

    Serial.println("Calling esp_camera_fb_get");
    fb = esp_camera_fb_get();
    if (!fb) {
      Serial.println("Camera capture failed");
      client.publish(MQTT_SERIAL_PUBLISH_CH, "Camera capture failed");

      return NULL;
    }

    uint8_t * buf = NULL;
    size_t buf_len = 0;

    Serial.print("Calling frame2bmp image size is: ");
    Serial.println(fb->len);

    bool converted = frame2bmp(fb, &buf, &buf_len);    
    esp_camera_fb_return(fb);

    if(!converted){
      Serial.println("BMP conversion failed");
      client.publish(MQTT_SERIAL_PUBLISH_CH, "BMP conversion failed");

      return NULL;
    }

    return buf;
}
```

Where the serial monitor returns:
```
[Info] Opened the serial port - COM3
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6400
entry 0x400806a8

Connecting to FaceScanner
...........
WiFi connected
IP address: 
192.168.1.103
Converting IPAddress to string
192.168.1.103
In the mqtt reconnect loop
Attempting MQTT connection...connected
Publishing message to let them now I am here o/
Subscribing to private channel: /facescanner/ESP32/30aea4d45880
-------new message from broker-----
channel: /facescanner/ESP32/public/rx
data: Take:bmp
Received message: Take:bmp
Received 'take' command, taking picture...
LED: 1
Calling esp_camera_fb_get
Calling frame2bmp image size is: 50304
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400df272  PS      : 0x00060130  A0      : 0x8015732d  A1      : 0x3ffb1c30  
A2      : 0x00000036  A3      : 0x00000036  A4      : 0x000012c0  A5      : 0x00000010  
A6      : 0x00000008  A7      : 0x3ffc2580  A8      : 0x00000000  A9      : 0x00000002  
A10     : 0x000000a5  A11     : 0x00000036  A12     : 0x00000030  A13     : 0x00000000  
A14     : 0x00009600  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000036  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x400df272:0x3ffb1c30 0x4015732a:0x3ffb1c50 0x40061b1b:0x3ffb1c70 0x400e13a7:0x3ffb1cf0 0x400df2f9:0x3ffb1db0 0x400df3b1:0x3ffb1de0 0x400df5b5:0x3ffb1e20 0x400d2670:0x3ffb1e50 0x400d3297:0x3ffb1e80 0x400d3487:0x3ffb1ed0 0x400d517a:0x3ffb1f30 0x400d3376:0x3ffb1f90 0x400da831:0x3ffb1fb0 0x4008d8e5:0x3ffb1fd0

Rebooting...
```

First, excuse the other serial outputs but i did not figure out a way to debug on the esp32 yet.
Also, i am not an experienced c++ developer, so any inputs are welcome :)

The idea is to post the image to an API later in the process, doing this with the camera_fb_t* frame buffer in jpeg format works well.

Another question i am curious about: does this really increase image quality when i convert to bmp this way? When does the compression happen?

Thanks a lot and cheers
Peter

me-no-dev, 2019-09-20:
your frame size is too large :) for BMP you should not go over 800x600. Resulting required memory is 800*600*3+header, so UXGA is a bit much :)

closed 2019-09-20 by me-no-dev

terrorzombie666, 2019-09-22:
oh wow, i see :)
thanks for the feedback!

XerXeX, 2020-03-15:
I'm having the same problem. As soon as a call the conversion to BMP it reboots.
I am using the lowest resolution. What could be the problem?

My code
`#include "esp_camera.h"

//WROVER-KIT PIN Map
#define CAM_PIN_PWDN    32 //power down is not used
#define CAM_PIN_RESET   -1 //software reset will be performed
#define CAM_PIN_XCLK    0
#define CAM_PIN_SIOD    26
#define CAM_PIN_SIOC    27

#define CAM_PIN_D7      35
#define CAM_PIN_D6      34
#define CAM_PIN_D5      39
#define CAM_PIN_D4      36
#define CAM_PIN_D3      21
#define CAM_PIN_D2      19
#define CAM_PIN_D1      18
#define CAM_PIN_D0       5
#define CAM_PIN_VSYNC   25
#define CAM_PIN_HREF    23
#define CAM_PIN_PCLK    22

static camera_config_t camera_config = {
    .pin_pwdn  = CAM_PIN_PWDN,
    .pin_reset = CAM_PIN_RESET,
    .pin_xclk = CAM_PIN_XCLK,
    .pin_sscb_sda = CAM_PIN_SIOD,
    .pin_sscb_scl = CAM_PIN_SIOC,

    .pin_d7 = CAM_PIN_D7,
    .pin_d6 = CAM_PIN_D6,
    .pin_d5 = CAM_PIN_D5,
    .pin_d4 = CAM_PIN_D4,
    .pin_d3 = CAM_PIN_D3,
    .pin_d2 = CAM_PIN_D2,
    .pin_d1 = CAM_PIN_D1,
    .pin_d0 = CAM_PIN_D0,
    .pin_vsync = CAM_PIN_VSYNC,
    .pin_href = CAM_PIN_HREF,
    .pin_pclk = CAM_PIN_PCLK,

    //XCLK 20MHz or 10MHz for OV2640 double FPS (Experimental)
    .xclk_freq_hz = 20000000,
    .ledc_timer = LEDC_TIMER_0,
    .ledc_channel = LEDC_CHANNEL_0,

    .pixel_format = PIXFORMAT_JPEG,//YUV422,GRAYSCALE,RGB565,JPEG
    .frame_size = FRAMESIZE_QQVGA,//QQVGA-QXGA Do not use sizes above QVGA when not JPEG

    .jpeg_quality = 12, //0-63 lower number means higher quality
    .fb_count = 1 //if more than one, i2s runs in continuous mode. Use only with JPEG
};

esp_err_t camera_init(){
    //power up the camera if PWDN pin is defined
    if(CAM_PIN_PWDN != -1){
        pinMode(CAM_PIN_PWDN, OUTPUT);
        digitalWrite(CAM_PIN_PWDN, LOW);
    }

    //initialize the camera
    esp_err_t err = esp_camera_init(&camera_config);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "Camera Init Failed");
        return err;
    }

    return ESP_OK;
}

void getColor(){
  camera_fb_t * fb = esp_camera_fb_get();
  if (!fb) {
      Serial.println("Camera capture failed");
  }

  uint8_t * buf = NULL;
  size_t buf_len = 0;
  bool converted = frame2bmp(fb, &buf, &buf_len);
  esp_camera_fb_return(fb);
  if(!converted){
      Serial.println("Convert to BMP failed");
  }
}

void setup() {
  Serial.begin(115200);
  camera_init();
}

void loop() {
  getColor();
  delay(10000);
}`

And the error on serial

`15:53:44.912 -> ets Jun  8 2016 00:22:57
15:53:44.912 -> 
15:53:44.912 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
15:53:44.912 -> configsip: 0, SPIWP:0xee
15:53:44.912 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
15:53:44.912 -> mode:DIO, clock div:1
15:53:44.912 -> load:0x3fff0018,len:4
15:53:44.912 -> load:0x3fff001c,len:1216
15:53:44.912 -> ho 0 tail 12 room 4
15:53:44.946 -> load:0x40078000,len:9720
15:53:44.946 -> ho 0 tail 12 room 4
15:53:44.946 -> load:0x40080400,len:6352
15:53:44.946 -> entry 0x400806b8
15:53:45.362 -> Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
15:53:45.362 -> Core 1 register dump:
15:53:45.362 -> PC      : 0x400ea92a  PS      : 0x00060430  A0      : 0x800f02dd  A1      : 0x3ffb1d40  
15:53:45.399 -> A2      : 0x00000036  A3      : 0x00000036  A4      : 0x000001e0  A5      : 0x00000010  
15:53:45.399 -> A6      : 0x00000008  A7      : 0x3ffc2228  A8      : 0x00000000  A9      : 0x00000002  
15:53:45.399 -> A10     : 0x00000026  A11     : 0x00000036  A12     : 0x00000030  A13     : 0x00000000  
15:53:45.399 -> A14     : 0x00000f00  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x0000001d  
15:53:45.433 -> EXCVADDR: 0x00000036  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
15:53:45.433 -> 
15:53:45.433 -> Backtrace: 0x400ea92a:0x3ffb1d40 0x400f02da:0x3ffb1d60 0x40061b1b:0x3ffb1d80 0x400ecdbf:0x3ffb1e00 0x400ea9b1:0x3ffb1ec0 0x400eaa69:0x3ffb1ef0 0x400eac6d:0x3ffb1f30 0x400d12b7:0x3ffb1f60 0x400d12f7:0x3ffb1f90 0x400d1a7d:0x3ffb1fb0 0x4008ab09:0x3ffb1fd0
15:53:45.468 -> 
15:53:45.468 -> Rebooting...
`