espressif/esp32-camera #77: OV7670

jjsch-dev, 2019-09-23:
Hi, I writed the driver for the OV7670, if some need it please contact me.

jjsch-dev, 2019-10-23:
Hi, sorry for the late answer, but I don't know how to download. Can you suggest a way?

Obtener Outlook para Android<https://aka.ms/ghei36>

Hi thanks for telling me.
Where is it.
i want.

jjsch-dev, 2019-10-24:
> Download what?
> I don't understand ?
> Do you mean Download from
> Microsoft outlook?
> Your code is in outlook?

Attached I send you the source code.
Sorry but I don't know how to made the commit in github. Can you give instructions?

amirjak, 2019-10-25:
You have to Made new Repositories , like made new project And then can upload your code.

jjsch-dev, 2019-10-25:
thanks for the information, I made a fork of the esp32-camera and the pull request for adding the ov7670 driver. Please tell me if you can get the changes.

amirjak, 2019-10-25:
i saw changes
thanks.

amirjak, 2019-10-25:
I have a problem with ESP-IDF. 
This project has no (main) folder. 
When I want to use it. 
Show this line.

[image]
 If you can help me.

jjsch-dev, 2019-10-25:
The esp32-camera-master is part of the idf component, you need to copy it to the folder idf/componnets. I use a fork of the igrr/esp32-cam-demo, if you need it please let me know.

amirjak, 2019-10-25:
Yes please 😃

jjsch-dev, 2019-10-25:
ok, I will made the fork tomorrow.

amirjak, 2019-10-26:
**I** download and extract the repository to the components folder of **my** ESP-IDF project.
**But** when I want to **make menuconfig** this happened.

Annotation 2019-10-25 204656

amirjak, 2019-10-26:
I use **python 2.7 8** ( windows 10 ) 64 bit

jjsch-dev, 2019-10-26:
I made the pull request to the igrr/esp32-cam-demo. Please use this project and compile.

jjsch-dev, 2019-10-26:
Checking the rep for the es32-camera, by mistake I forgot to add the drivers files: 
sensors/ov7670.c
sensors/private_include/ov7670.h
sensors/private_include/ov7670_regs.h
please check now.

jjsch-dev, 2019-10-26:
Now I merge the igrr/esp32-cam-demo in my repository. Please use this project and compile.

amirjak, 2019-10-27:
I compile **igrr/esp32-cam-demo** project in **your** repository **But** this happened.
**Unfortunately**, I'm not good at working with **ESP-IDF**.

[image]

jjsch-dev, 2019-10-27:
The esp32_camera driver need to be inside the components folder of esp-idf, please check if you have a similar list.
[image]

amirjak, 2019-10-27:
I **don't** have **esp32 - camera** folder inside the **components** folder

[image]

jjsch-dev, 2019-10-27:
ok, please try to move the folder esp32-camera to the folder esp-idf/components.

amirjak, 2019-10-27:
Do you mean esp32-cam-demo-master folder I downloaded?
Like this 
esp\esp-idf\components\esp32-cam-demo-master

amirjak, 2019-10-27:
[image]

jjsch-dev, 2019-10-27:
you need to obtain two source code, one is the esp32-cam-demo-master (the application), please leave it in the example folder, but the driver code esp32-camera need to be in the folder esp-idf/components.

amirjak, 2019-10-27:
Where is the **driver code camera**  esp32-camera?

jjsch-dev, 2019-10-27:
the camera driver: https://github.com/jjsch-dev/esp32-camera
the camera demo: https://github.com/jjsch-dev/esp32-cam-demo
[image]
[image]

amirjak, 2019-10-28:
**esp32-cam-demo**

amirjak, 2019-10-28:
When I **extract** the zip file (**esp32-cam-demo. zip**)
 In the (**esp32-cam-demo\components\http_server**) is **empty**?
Nothing on it. 
And then I want to compile gets error **http_server. h**

amirjak, 2019-10-28:
When I copy two file in folder http_server and compile 

[image]

this happened

LD build/esp32-cam-demo.elf
c:/users/ali/downloads/compressed/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/ali/Downloads/Compressed/msys32/home/ali/esp/esp-idf/examples/esp32-cam-demo/build/main\libmain.a(app_main.o):(.literal.write_frame+0x8): **undefined reference to `http_response_write'**
c:/users/ali/downloads/compressed/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/ali/Downloads/Compressed/msys32/home/ali/esp/esp-idf/examples/esp32-cam-demo/build/main\libmain.a(app_main.o):(.literal.handle_jpg_stream+0xc): **undefined reference to `http_response_begin'**
c:/users/ali/downloads/compressed/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/ali/Downloads/Compressed/msys32/home/ali/esp/esp-idf/examples/esp32-cam-demo/build/main\libmain.a(app_main.o):(.literal.handle_jpg_stream+0x10): **undefined reference to `http_response_begin_multipart'**
c:/users/ali/downloads/compressed/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/ali/Downloads/Compressed/msys32/home/ali/esp/esp-idf/examples/esp32-cam-demo/build/main\libmain.a(app_main.o):(.literal.handle_jpg_stream+0x14): undefined reference to `http_response_end_multipart'
c:/users/ali/downloads/compressed/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/ali/Downloads/Compressed/msys32/home/ali/esp/esp-idf/examples/esp32-cam-demo/build/main\libmain.a(app_main.o):(.literal.handle_jpg_stream+0x18): undefined reference to `http_response_end'
c:/users/ali/downloads/compressed/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/ali/Downloads/Compressed/msys32/home/ali/esp/esp-idf/examples/esp32-cam-demo/build/main\libmain.a(app_main.o):(.literal.handle_jpg+0xc): undefined reference to `http_response_set_header'
c:/users/ali/downloads/compressed/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/ali/Downloads/Compressed/msys32/home/ali/esp/esp-idf/examples/esp32-cam-demo/build/main\libmain.a(app_main.o):(.literal.app_main+0x88): undefined reference to `http_server_start'
c:/users/ali/downloads/compressed/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/ali/Downloads/Compressed/msys32/home/ali/esp/esp-idf/examples/esp32-cam-demo/build/main\libmain.a(app_main.o):(.literal.app_main+0x8c): undefined reference to `http_register_handler'
c:/users/ali/downloads/compressed/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/ali/Downloads/Compressed/msys32/home/ali/esp/esp-idf/examples/esp32-cam-demo/build/main\libmain.a(app_main.o): in function `write_frame':
C:/Users/ali/Downloads/Compressed/msys32/home/ali/esp/esp-idf/examples/esp32-cam-demo/main/app_main.c:151: undefined reference to `http_response_write'
c:/users/ali/downloads/compressed/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/ali/Downloads/Compressed/msys32/home/ali/esp/esp-idf/examples/esp32-cam-demo/build/main\libmain.a(app_main.o): in function `handle_jpg_stream':
C:/Users/ali/Downloads/Compressed/msys32/home/ali/esp/esp-idf/examples/esp32-cam-demo/main/app_main.c:359: undefined reference to `http_response_begin'
c:/users/ali/downloads/compressed/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/ali/Downloads/Compressed/msys32/home/ali/esp/esp-idf/examples/esp32-cam-demo/main/app_main.c:367: undefined reference to `http_response_begin_multipart'
c:/users/ali/downloads/compressed/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/ali/Downloads/Compressed/msys32/home/ali/esp/esp-idf/examples/esp32-cam-demo/main/app_main.c:375: undefined reference to `http_response_end_multipart'
c:/users/ali/downloads/compressed/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/ali/Downloads/Compressed/msys32/home/ali/esp/esp-idf/examples/esp32-cam-demo/main/app_main.c:379: undefined reference to `http_response_end'
c:/users/ali/downloads/compressed/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/ali/Downloads/Compressed/msys32/home/ali/esp/esp-idf/examples/esp32-cam-demo/build/main\libmain.a(app_main.o): in function `handle_jpg':
C:/Users/ali/Downloads/Compressed/msys32/home/ali/esp/esp-idf/examples/esp32-cam-demo/main/app_main.c:284: undefined reference to `http_response_begin'
c:/users/ali/downloads/compressed/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/ali/Downloads/Compressed/msys32/home/ali/esp/esp-idf/examples/esp32-cam-demo/main/app_main.c:284: undefined reference to `http_response_set_header'
[... cut ...]
c:/users/ali/downloads/compressed/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/ali/Downloads/Compressed/msys32/home/ali/esp/esp-idf/examples/esp32-cam-demo/build/main\libmain.a(app_main.o):C:/Users/ali/Downloads/Compressed/msys32/home/ali/esp/esp-idf/examples/esp32-cam-demo/main/app_main.c:124: more undefined references to `http_register_handler' follow
collect2.exe: error: ld returned 1 exit status
make: *** [C:/Users/ali/Downloads/Compressed/msys32/home/ali/esp/esp-idf/make/project.mk:544: /home/ali/esp/esp-idf/examples/esp32-cam-demo/build/esp32-cam-demo.elf] Error 1

jjsch-dev, 2019-10-28:
please copy all files.
[image]

amirjak, 2019-10-29:
**Thank you for your work .**
**It works .**
 I have a **problem** with the **color** of the photo ?

http://192.168.1.132/bmp

**capture.bmp**

[image]

amirjak, 2019-10-29:
**Pin Configuration**

  **PWDN**    Not connected
  **RESET**   2
  **XCLK**    21
  **SIOD**    26
  **SIOC**    27

  **D7**      35
  **D6**      34
  **D5**      39
  **D4**      36
  **D3**      19
  **D2**      18
  **D1**       5
  **D0**       4
  **VSYNC**   25
  **HREF**    23
  **PCLK**    22

amirjak, 2019-10-30:
sdkconfig.defaults
**CONFIG_ENABLE_TEST_PATTERN=y**
CONFIG_OV2640_SUPPORT=n
CONFIG_OV7725_SUPPORT=n
CONFIG_OV7670_SUPPORT=y
CONFIG_ESP32_DEFAULT_CPU_FREQ_240=y
CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ=240
CONFIG_MEMMAP_SMP=y
CONFIG_TASK_WDT=n
CONFIG_FREERTOS_UNICORE=n
CONFIG_FREERTOS_HZ=100
CONFIG_LOG_DEFAULT_LEVEL_INFO=y
CONFIG_LOG_DEFAULT_LEVEL=3

sdkconfig.File 

CONFIG_BT_RESERVE_DRAM=0
# CONFIG_BLE_MESH is not set
**CONFIG_ENABLE_TEST_PATTERN=y**
# CONFIG_OV2640_SUPPORT is not set
# CONFIG_OV7725_SUPPORT is not set
CONFIG_OV7670_SUPPORT=y

jjsch-dev, 2019-10-30:
The file is the sdkconfig, but you can change the configuration using the utility menuconfig. Some times you will need to clean project to update the configuration, use: make clean.

[image]

**Launch the menuconfig**
[image]

**Change the clock frequency**
[image]
[image]

**Enable test pattern**
[image]
[image]

amirjak, 2019-10-31:
I think I **missed something**

So I **deleted everything** and **started** from the **beginning**

I have **downloaded these files again**
[image]
[image]

And put them in **these folders**

**jjsch-dev/esp32-camera**
# esp\esp-idf\components\esp32-camera

**jjsch-dev/esp32-cam-demo**
# esp\esp-idf\examples\esp32-cam-demo
**igrr/esp32-http-server**
# esp\esp-idf\examples\esp32-cam-demo\components\http_server

And then at **mingw32.exe**
[image]
[image]

[image]

[image]
[image]
[image]

# And then it happened

[image]

jjsch-dev, 2019-10-31:
I reproduce the problem by increasing the frequency of the camera clock by 120 Mhz, please use 12 Mhz.

[image]

amirjak, 2019-11-01:
frequency 12000000 not working

jjsch-dev, 2019-11-01:
I do some test with my board, and found the camera clock limit is 24 Mhz, but work fine from 12 to 20 Mhz, and present the problems that you first reported with 10 Mhz. All test was made it with 240 Mhz for the cpu clock.

**Camera clock = 10 Mhz**
[image]

**Camera clock = 15 Mhz**
[image]

**Camera clock = 20 Mhz**
[image]

**Camera clock = 24 Mhz**
[image]

**CPU clock = 240 Mhz**
[image]

amirjak, 2019-11-02:
After the **camera color** had a problem
[image]

 I deleted the msys32 program with its folder

After that, **when I reinstalled** the program, I had a **this problem**

**I made these changes**

[image]

**And then it happened**

[image]

[image]

If you can **send** your **cpu_start.c** file to me

jjsch-dev, 2019-11-02:
ok
[cpu_start.zip](https://github.com/espressif/esp32-camera/files/3800690/cpu_start.zip)

jjsch-dev, 2019-11-03:
It is possible that you use are using a newer version of module ledc than me. I found in the esp32_camera a similar report:
**LEDc configuration error and camera_probe failed #66** of https://github.com/masterd89 

https://github.com/mirronelli suggest add the following sentence to fix the problem.

> there was a change in ledc driver that is used to drive the clock
> 
> to work with the current version a one line of code needs to be added into xclk.c
> the line is:
> 
> timer_conf.clk_cfg = LEDC_USE_APB_CLK;
> put in in the block bellow
> 
>         ledc_timer_config_t timer_conf;
>         timer_conf.duty_resolution = 2;
>         timer_conf.freq_hz = config->xclk_freq_hz;
> 	timer_conf.speed_mode = LEDC_HIGH_SPEED_MODE;
> 	timer_conf.timer_num = config->ledc_timer;
> 	timer_conf.clk_cfg = LEDC_USE_APB_CLK;
> 	esp_err_t err = ledc_timer_config(&timer_conf);
> 	if (err != ESP_OK) {
> 		ESP_LOGE(TAG, "ledc_timer_config failed, rc=%x", err);
> 		return err;
> 	}
> you will find the the file in yourproject/components/esp32-camera/driver/xclk.c
> 
> use of the APB clock was assumed in previous commit to ledc.c and now it needs to be explicitly set

amirjak, 2019-11-03:
# Thank you for your work .
# 
[image]
It works .

jjsch-dev, 2019-11-03:
Glad to hear that!

billyrudi, 2019-11-05:
Hi, how i can use camera with arduino ide?
i have esp32-cam board and ov7670 camera module.
regards Paolo

amirjak, 2019-11-05:
https://github.com/amirjak/how-made-night-vision-camera-with-esp32

amirjak, 2019-11-05:
https://www.google.com/url?sa=t&source=web&rct=j&url=https://medium.com/%40mudassar.tamboli/esp32-ov7670-websocket-video-camera-26c35aedcc64&ved=2ahUKEwiukfa3wdPlAhUSDOwKHfDTAHoQFjAEegQIAxAB&usg=AOvVaw2MoYvB0PihTokaT2NyFkqX

billyrudi, 2019-11-06:
tank you!!!

jjsch-dev, 2019-11-06:
I update the code of **esp32_camera** and **esp_camera_demo** to the idf version 4.1 v4.1-dev-815 (master branch).

The web server of camera demo was rewritten to use the **hhtp_server** of esp_idf.

Now the WIFI_SSID and WIFI_PASSWORD are configured in the example connection menu.

The /esp32-camera/blob/master/driver/xclk.c was patched with **timer_conf.clk_cfg = LEDC_USE_APB_CLK;**

[image]

amirjak, 2019-11-07:
Thank you for your work .
 idf version 4.1 v4.1-dev-815 (master branch).
[image]

Robinroy-peter, 2020-03-21:
hi, could u help me how can i connect ov7670 fifo camera using this codes

jjsch-dev, 2020-03-21:
Hi @Robinroy-peter for compile the driver you need the [idf](https://github.com/espressif/esp-idf) and then copy or clone this [driver](https://github.com/espressif/esp32-camer) in the components folder.

To test you can use many app, for example [esp-who](https://github.com/espressif/esp-who) or [esp32-demo-cam](https://github.com/jjsch-dev/esp32-cam-demo).

Robinroy-peter, 2020-03-21:
thank you bro @jjsch-dev  will it work for ov7670 fifo camera bcz that camera  does not have xclk and pclk pins??
and now i m getting below error
CMake Error at CMakeLists.txt:85 (idf_build_set_property):
  Unknown CMake command "idf_build_set_property".

i m getting this error any suggt

jjsch-dev, 2020-03-21:
The OV7670 has clk and pclk, please check these hardware pictures and menuconfig.
[image]
[image]

the error is a result of compilation of esp32_cam_demo?
 **CMake Error at CMakeLists.txt:85 (idf_build_set_property):**

jjsch-dev, 2020-03-21:
I searched the web and discovered a similar [problem](https://github.com/espressif/esp-idf/issues/3959) which is related to running idf.py from IDF_PATH and not from the application path.

Robinroy-peter, 2020-03-21:
[](https://ae01.alicdn.com/kf/H04d5a3e445374fcc80ed0d1fd4fbde11J.jpg_q50.jpg)  this model i m using check it once

jjsch-dev, 2020-03-21:
please send me a screenshot of the console build result. The following photo is my result.
[image]

Robinroy-peter, 2020-03-21:
[image]

Robinroy-peter, 2020-03-21:
[image]

Robinroy-peter, 2020-03-21:
[image]

jjsch-dev, 2020-03-21:
@Robinroy-peter good news, you need to enable the rtc with the menuconfig, go to the:
 `Component config` -> `Driver configurations`  -> `RTCIO Configuration`  and enable `support array rtc_gpio_desc for ESP32`

Robinroy-peter, 2020-03-21:
> [image]

thatas fine but i m facing this error so i can't run menuconfig

Robinroy-peter, 2020-03-21:
> @Robinroy-peter good news, you need to enable the rtc with the menuconfig, go to the:
> `Component config` -> `Driver configurations` -> `RTCIO Configuration` and enable `support array rtc_gpio_desc for ESP32`

thank you bro please checck out below error
Executing action: all (aliases: build)
Running cmake in directory /home/robinrobotics/esp/esp32-cam-demo/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /home/robinrobotics/esp/esp32-cam-demo"...
Warn about uninitialized values.
-- Found Git: /usr/bin/git (found version "2.7.4") 
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- The ASM compiler identification is unknown
-- Found assembler: xtensa-esp32-elf-gcc
CMake Error at /home/robinrobotics/esp/esp-idf/tools/cmake/project.cmake:307 (__project):
  The CMAKE_C_COMPILER:

    xtensa-esp32-elf-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:12 (project)

CMake Error at /home/robinrobotics/esp/esp-idf/tools/cmake/project.cmake:307 (__project):
  The CMAKE_CXX_COMPILER:

    xtensa-esp32-elf-g++

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:12 (project)

CMake Error at /home/robinrobotics/esp/esp-idf/tools/cmake/project.cmake:307 (__project):
  The CMAKE_ASM_COMPILER:

    xtensa-esp32-elf-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:12 (project)

-- Warning: Did not find file Compiler/-ASM
-- Configuring incomplete, errors occurred!
See also "/home/robinrobotics/esp/esp32-cam-demo/build/CMakeFiles/CMakeOutput.log".
See also "/home/robinrobotics/esp/esp32-cam-demo/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1

jjsch-dev, 2020-03-21:
I don't think that will solve the problem, but the canonical way is to put the examples in the esp/esp-id/examples folder. Move esp32-demo-camera to this folder and then run idf.py build.

[image]

[37 more replies]

closed 2021-09-17 by github-actions[bot]