espressif/esp32-camera #78: get frame bytes as soon as available

MrMabulous, 2019-10-01:
Currently it seems only possible to access a frame once it has been completed.
For low-latency applications it would be great if frame data could be accessed as it comes in, even before the full frame has completed

me-no-dev, 2019-10-01:
the problem is that if you do that, there will not be enough processing time for the data to be pushed to PSRAM and you to work with it in between. It might be doable with some more hardcore task/cpu tricks, but not something that will work for everyone.

closed 2019-10-01 by me-no-dev

MrMabulous, 2019-10-01:
The only thing I am doing with the data is forwarding it over wifi to a computer. Would that be impossible too?

Can I somehow select only a single framebuffer and have the camera write to it continuously without blocking? I don't care about frame tearing for the moment

me-no-dev, 2019-10-01:
what frame rate and resolution are you looking for? frame format? you can not tear JPEG :)

MrMabulous, 2019-10-01:
CIF at 60 fps. I already manage streaming that using the asyncTCP client, but i‘d like to reduce latency as far as possible