espressif/esp32-camera #79: Is framebuffer limited to 23200 bytes?

MrMabulous, 2019-10-01:
I'm running the esp32 with ov2640 camera in CIF with JPG format.

It works fine and fast so long as I capture natural images. When I point the camera however at an artificial, high detail pattern, the jpg size increases (which makes sense) but caps at 23200 bytes (esp_camera_fb_get()->len) and at that point the images that are returned are 'cut off' to the bottom (see attachment).

this happens independently of whether I run the clock at 20Mhz or 10Mhz.

I didn't find 23200 in the code, so I wonder what else is imposing this limit and how to resolve it?

[https://ibb.co/Dr2NPw6](url)

MrMabulous, 2019-10-01:
I assume the problem is related to this line:

https://github.com/espressif/esp32-camera/blob/1b8fdfe335b3c8c8b9e091e3615b0aee16946ef0/driver/camera.c#L1136

because if I change the compression quality to 11, the cap goes to 14400 (and the frame is still cut off).

Are there wrong assumptions being made here?

nutfrey, 2020-01-17:
I have the same problem, but my files are capped at 370,000 bytes when using 1600x1200 resolution at quality of 10. The image is truncated exactly as described above. I have tried increasing the buffer count to 2 (and more) as suggested in other posts, but to no effect. I’m guessing the line above is indeed the culprit. Perhaps we could decrease the compression_ratio_bound which would result in higher allocation and a successful image capture?

S0-L0, 2020-05-28:
```
  ...
  config.pin_reset = RESET_GPIO_NUM;
  config.xclk_freq_hz = 20000000;
  config.pixel_format = PIXFORMAT_JPEG;
  config.frame_size = FRAMESIZE_CIF;
  config.jpeg_quality = 10;
  config.fb_count = 2;
  Serial.println(psramFound());
  esp_err_t err = esp_camera_init(&config);
  if (err != ESP_OK) {
    return;
  }

  sensor_t * s = esp_camera_sensor_get();
  if (s->id.PID == OV2640_PID) {
    s->set_brightness(s, 2);
    s->set_saturation(s, 0);
    s->set_exposure_ctrl(s, 1);
  }
  s->set_framesize(s, FRAMESIZE_XGA);
  ...
```
I had the same issue with this code, the problem was changing the framesize to a larger one after initializing the camera with
```
esp_err_t err = esp_camera_init(&config);
```
because this is when the framebuffer size gets calculated and it won't be recalculated if not reinitialized.

closed 2021-09-17 by github-actions[bot]