espressif/esp32-camera #81: Why running at 20Mhz rather than 24Mhz?

MrMabulous, 2019-10-03:
Why is the ov2640 internal doubler activated at 10'000'000 XCLK rather than 12'000'000 XCLK?

https://github.com/espressif/esp32-camera/blob/1b8fdfe335b3c8c8b9e091e3615b0aee16946ef0/sensors/ov2640.c#L244

In order to reach 60 fps @ CIF, it's necessary to run at 24Mhz PCLK (at 20Mhz it's only possible to get to 50)

Changing that to 12'000'000 allows me to run at 60fps

me-no-dev, 2019-10-03:
it will not be stable because it does not divide to what the ESP can actually process reliably. Do as you wish, but I can not add this to the driver itself.

closed 2019-10-03 by me-no-dev

MrMabulous, 2019-10-03:
Why not stable? I'm streaming the frames out to my computer and it works great at 60 fps with no framedrops

me-no-dev, 2019-10-04:
I'm really happy for you :) but I did have issues on different boards. The ESP uses 80MHz APB clock to run the peripherals. That same clock powers the I2S bus, which in turn can sample at max of 20MHz in parallel mode. I can accept a PR that will change that if to `if (sensor->pixformat == PIXFORMAT_JPEG && sensor->xclk_freq_hz < 20000000) {`

me-no-dev, 2019-10-04:
will that do for you?

MrMabulous, 2019-10-04:
That sounds great :)

so does that mean sensor->xclk_freq_hz also is the clock frequency of the I2S bus (ignoring internal frequency doubler of the sensor)?

How does that relate to

https://github.com/espressif/esp32-camera/blob/1b8fdfe335b3c8c8b9e091e3615b0aee16946ef0/sensors/private_include/ov2640_settings.h#L351

?
If i understand correctly this causes PCLK = sensor->xclk_freq_hz * (freqdoubler?2:1) / 4

do i understand this line correctly? why does pixelclock have to be divided by 4 if format is jpg?

MrMabulous, 2019-10-09:
Just as info:

with ESP32-CAM (AI-Thinker ESP32-S & OV2640), I'm able to stream very stable CIF @ 80fps and SVGA @40fps if I set xclk_freq_hz to 16'000'000 with frequency doubling enabled and jpg quality at 10.

I'm also able to stream (stable too) CIF @ 120fps even and SVGA @ 60 fps, if I set xclk_freq_hz to 24'000'000 with frequency doubling enabled and  {R_DVP_SP, 0x08}
However, at 120 fps CIF (or 60 fps SVGA) there are some jpg artifacts.

Celppu, 2020-03-11:
What kind of artifacts? Have you tried to capture frames without wifi, because i have noticed that momentary current draw from wifi might be the reason why some cant use higher unstable frequencies.

me-no-dev, 2020-03-12:
I have revised the clock configs for OV2640 and the current code will run at almost the max frame rate if the doubler is enabled (disabled on purpose) but CIF and below will do 110FPS if the doubler is on (can be on for SVGA too, but not for freqs above). Anyway... it's quite pointless to run at htat speed because the time that the sensor has to expose the frame is too short and that is the reason for the artefacts. Just not enough image data for good image.