espressif/esp32-camera #84: saving Mjpeg stream to SD card

cheeta1, 2019-10-14:
Hi there can I save the Mjpeg stream to SD card as well as streaming it to the web-server?
if yes can you guide me.

Thanks

me-no-dev, 2019-10-14:
MJPEG is just JPEG images one after another. If you save them as such in SD, you can recreate the stream later. Writing to SD at the same time as sending to the web will take resources and slow down the speed.

billyrudi, 2019-10-14:
Hi,
look here
[https://github.com/jameszah/ESP32-CAM-Video-Recorder]

cheeta1, 2019-10-14:
@me-no-dev  is it possible to encode the video and then store it. Is esp32 capable of encoding 720p stream and saving it in sd card. I do not need the web streaming function. Just recording video on the sd card. 

Thanks

me-no-dev, 2019-10-14:
Just saving the stream should be possible. The format that we use for web streaming might not work once downloaded though. The thing is that real compression keeps a few frames and sends the differences, but we do not have enough memory and CPU to handle all of that and have a reasonable frame rate. The MJPEG that we do support just sends the images one after another and the farme rate is really controlled by the throughput of the network and the settings of the camera. If you save the file and then play it again, there will not be anything to help time the frames properly. Surely there is a way to produce proper MJPEG but we have not had the time to research it enough. Not much info online :)

cheeta1, 2019-10-14:
@me-no-dev  thanks for the quick answer. I can guess from your answer that esp32 is not capable of doing any encoding like H,263 etc.. Is there any other way that you can suggest which will help me out in making cam recorder?

Thanks

nadeem268, 2023-10-18:
Mjpeg stream to SD card 
 Can  anyone help me
i just simply store jpeg in sd card , now  i wnat to store Mjpeg stream in sd card

mgi24, 2024-04-19:
https://github.com/s60sc/ESP32-CAM_MJPEG2SD check this out, it have all you need+more