espressif/esp32-camera #92: How to use a not yet supported camera?

riccardobrue, 2019-11-12:
Hi, I would like to use [this](https://www.arducam.com/product/1-6-cmos-ov7670-standalone-camera-uc7670-a/) camera module, can someone explain to me if it is possible to use it on the ESP-EYE by modifying the drivers for Linux (ie [this one](https://github.com/allwinner-zh/linux-3.4-sunxi/blob/master/drivers/media/video/sunxi-vfe/device/ar0330.c))?

Is this a overkilling process to do? Do you suggest me to avoid this kind of work and to use the supported ones (OV2640 and OV3660)?

I need 3MP images, I am in Europe and I cannot find the OV3660-based cameras, what do you suggest me to do? 

Thank you.

me-no-dev, 2019-11-12:
seems that this sensor does not support compression, which means that the frame size that you can acquire will be quite limited. the OV sensors that we support can output JPEGs and that is how we can grab the frame.

closed 2019-11-12 by me-no-dev

riccardobrue, 2019-11-12:
> OV sensors that we suppo

Oh! perfect thank you for the reply, appreciated. So the maximum that I can obtain is 3MP from the OV3660 right?

me-no-dev, 2019-11-12:
yup :)

riccardobrue, 2019-11-12:
A last question, how can i ship the OV3660 in Europe? I cannot find it easily here. Do you know any module that includes such sensor? Thank you.