espressif/esp32-camera #93: esp_camera_fb_get(): Failed to get the frame on time!

grandiloquent, 2019-11-13:
```
#include <esp_camera.h>

#define PIN_PWDN 32
#define PIN_RESET -1
#define PIN_XCLK 0
#define PIN_SIOD 26
#define PIN_SIOC 27
#define PIN_D7 35
#define PIN_D6 34
#define PIN_D5 39
#define PIN_D4 36
#define PIN_D3 21
#define PIN_D2 19
#define PIN_D1 18
#define PIN_D0 5
#define PIN_VSYNC 25
#define PIN_HREF 23
#define PIN_PCLK 22

//
//#define PIN_PWDN -1
//#define PIN_RESET 15
//#define PIN_XCLK 27
//#define PIN_SIOD 25
//#define PIN_SIOC 23
//#define PIN_D7 19
//#define PIN_D6 36
//#define PIN_D5 18
//#define PIN_D4 39
//#define PIN_D3 5
//#define PIN_D2 34
//#define PIN_D1 35
//#define PIN_D0 17
//#define PIN_VSYNC 22
//#define PIN_HREF 26
//#define PIN_PCLK 21

static camera_config_t camera_example_config = {
  .pin_pwdn       = PIN_PWDN,
  .pin_reset      = PIN_RESET,
  .pin_xclk       = PIN_XCLK,
  .pin_sscb_sda   = PIN_SIOD,
  .pin_sscb_scl   = PIN_SIOC,
  .pin_d7         = PIN_D7,
  .pin_d6         = PIN_D6,
  .pin_d5         = PIN_D5,
  .pin_d4         = PIN_D4,
  .pin_d3         = PIN_D3,
  .pin_d2         = PIN_D2,
  .pin_d1         = PIN_D1,
  .pin_d0         = PIN_D0,
  .pin_vsync      = PIN_VSYNC,
  .pin_href       = PIN_HREF,
  .pin_pclk       = PIN_PCLK,
  .xclk_freq_hz   = 20000000,
  .ledc_timer     = LEDC_TIMER_0,
  .ledc_channel   = LEDC_CHANNEL_0,
  .pixel_format   = PIXFORMAT_JPEG,
  .frame_size     = FRAMESIZE_SVGA,
  .jpeg_quality   = 10,
  .fb_count       = 2
};
esp_err_t camera_example_init() {
  return esp_camera_init(&camera_example_config);
}
esp_err_t camera_example_capture() {
  //capture a frame
  camera_fb_t * fb = esp_camera_fb_get();
  if (!fb) {
    ESP_LOGE(TAG, "Frame buffer could not be acquired");
    return ESP_FAIL;
  }
  Serial.printf("len: %d, width: %d, height: %d\n",fb->len,fb->width,fb->height);
  //replace this with your own function
  //display_image(fb->width, fb->height, fb->pixformat, fb->buf, fb->len);
  //return the frame buffer back to be reused
  esp_camera_fb_return(fb);
  return ESP_OK;
}
void setup() {
  Serial.begin(115200);
  Serial.printf("camera_example_init: %d\n", camera_example_init());
  camera_example_capture();
}
void loop() {
  delay(1000);
}

```
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
camera_example_init: 0
[E][camera.c:1344] esp_camera_fb_get(): Failed to get the frame on time!

```

Schaggo, 2019-12-30:
could be your partition size

SumRandomGuy, 2020-01-08:
For anyone else coming along, I found the solution to this problem here:
https://randomnerdtutorials.com/esp32-cam-troubleshooting-guide/

ccrockatt, 2020-03-31:
Thanks for the link!

fazaio, 2020-05-29:
> For anyone else coming along, I found the solution to this problem here:
> https://randomnerdtutorials.com/esp32-cam-troubleshooting-guide/

thanks for your reference, i try to solve "esp_camera_fb_get(): Failed to get the frame on time!" using that link but its not work for me.

i used esp32 lib version 1.0.2 and i got this **esp_camera_fb_get(): Failed to get the frame on time!**

but when i downgrade esp32 from 1.0.2 to version 1.0.1 the message **esp_camera_fb_get():** is gone, but when i connect to server stream its keep loading very long time and just black screen like this issue #56

RobbesU, 2020-07-05:
THe combination of frame size and jpeg quality takes too long to process.  I found reducing `xclk_freq_hz` solves this.

ccrockatt, 2020-08-06:
I am using an ESP32-CAM and see this error frequently.  The module I have has PSRAM and when I see the error there is often plenty of PSRAM available.  
Interestingly though I have not run into this issue using the [ESP-EYE](https://www.espressif.com/en/products/devkits/esp-eye/overview) which is more powerful with 8 Mbyte PSRAM vs 4MB PSRAM in the ESP32-CAM.
@RobbesU I'll take a look at the {{xclk_freq_hz}} to see if it helps on the less powerful system.

RobbesU, 2020-08-07:
Thanks @ccrockatt that's exactly what I ended up doing as already mentioned above. Since reducing the `xclk_freq_hz` I haven't seen the error (dropping it as low as 5000000 I can process largest `PIX_FORMAT`.  Hike it back up and error returns, so I suspect the unit has a certain number of clock cycles to process the image. Too big an image and the image cannot be processed in the clock cycles available.

ccrockatt, 2020-08-07:
Good to know @RobbesU , I'd looked in esp_camera.h and saw the comment `    int xclk_freq_hz;               /*!< Frequency of XCLK signal, in Hz. Either 20KHz or 10KHz for OV2640 double FPS (Experimental) */` and hadn't thought to try anything other than what is recommended (10 or 20).

jofrev, 2020-11-23:
@RobbesU @ccrockatt Reducing XCLK probably works for you because it reduces the frame rate and thus the rate at which pixel data will come out of the cam (at least for OV2640). The ESP will now have more time (obviously enough) to write them to its framebuffer in PSRAM.  
The maximum rate at which data can be dumped to PSRAM seems to be around 7MB/s according to a few tests someone in another forum ran. So you should make sure the image stream coming from the camera does not exceed this limit.

FYI, in theory, there's another way of reducing the frame rate.
OV2640 (probably the other cams as well) provides a register called **CLKRC** which configures a clock divider effectively reducing the framerate.  
If you look at [**set_framerate**](https://github.com/espressif/esp32-camera/blob/010709376a131c12c14bb074b6c5be82d2241338/sensors/ov2640.c#L196) in [_sensors/ov2640.c_](https://github.com/espressif/esp32-camera/blob/010709376a131c12c14bb074b6c5be82d2241338/sensors/ov2640.c) you'll see it calls [**set_window**](https://github.com/espressif/esp32-camera/blob/010709376a131c12c14bb074b6c5be82d2241338/sensors/ov2640.c#L136), which then applies some default settings to the pixel clock and the divider:

```c
ov2640_clk_t c;
// [...]
c.pclk_auto = 0;
c.pclk_div = 8;
c.clk_2x = 0;
c.clk_div = 0;

if(sensor->pixformat != PIXFORMAT_JPEG){
    c.pclk_auto = 1;
    c.clk_div = 7;
}

if (mode == OV2640_MODE_CIF) {
    regs = ov2640_settings_to_cif;
     if(sensor->pixformat != PIXFORMAT_JPEG){
        c.clk_div = 3;
    }
} else if (mode == OV2640_MODE_SVGA) {
    regs = ov2640_settings_to_svga;
} else {
    regs = ov2640_settings_to_uxga;
    c.pclk_div = 12;
}
// [...]
WRITE_REG_OR_RETURN(BANK_SENSOR, CLKRC, c.clk);
WRITE_REG_OR_RETURN(BANK_DSP, R_DVP_SP, c.pclk);
```

Unfortunately, the cam driver doesn't seem to provide direct access to this register so you can't set it to a value suiting your application's needs without modifying the driver yourself.

If you want to learn more about how to reduce the frame rate take a look at the cam's datasheet, there's a few other things to be aware of apart from **CLKRC**.

damartinezru, 2020-12-05:
Look i've just had that problem for a month and  if you're using esp32- ov2640 look the camera reference because the one i've had was TY 0V2640-V2.0 and what i've saw it's like a cheap copy of the normal camera so be careful, i had to buy a new one because everything looked like a broken image although all the config worked fine.  Even changing the PIX_FORMAT and xclk_freq_hz didn't work, the camera was the problem.

umartechboy, 2021-01-11:
For me, reducing xclk_freq_hz improved the situation but what really solved it was an HW power fix. I was powering up the module through an old Arduino mega connected to the computer's USB and using it's onboard CH340 for serial. It looks like my USB wasn't capable enough to deliver enough power. I don't have an oscilloscope yet to profile the power demand with time, but as soon as I powered the module with an ~8V supply of two serially connected Li-Ion laptop battery cells (Battery> 7805 > Onboard AMS1117 3.3v > ESP32) the problem was resolved.

techturb, 2021-04-29:
Hi I stucked on this same problem for almost two months,and I searched for various website for the solution and I can't find a proper solution.atlast I looked up on the partition scheme and searched for changing partition scheme instead of 'huge app' and I found new partition settings from "https://youtu.be/mu3-Sff0B9w" His youtube video,Just change the partion scheme and download and paste the CSV file as same as he do.
Then re-open the IDE and change partion scheme. it will work😊

ligantx, 2021-05-09:
in my case, it was used to work fine, but out of nowhere, it started giving this error.
i just restart circuit, and everything was back to normal. I guess this is not the usual case, but it deserves a restart..

etothex23, 2025-04-11:
If you change the values for: Frame Rate, clock freq, and memory size it works. I did SVGA, quality 8, clock freq 10000000. I finallaly got video and then I could play with the values a bit.