espressif/esp32-camera #94: get frame buffer and use it later?

Adrianotiger, 2019-11-23:
I would like to shoot a picture as JPEG on an interrupt and don't use it immediately, but just when I ask this picture from a webpage.
Once the picture was sent, I would like to enable again the interrupt to take the next frame buffer.

My question is:
- is there a way to lock the framebuffer and use it maybe 5 minutes later?
- Or is it better to allocate 500kb in PSRAM and copy the frame buffer in this RAM (does not know how do it in Arduino)
- Or can I set 3 frame buffers for example?

I have 4MB or PSRAM, but does not know if all is used by the cam.

There are a lot of examples, but every example is always use the buffer just after getting the picture (saving it to SD-card, to SPIFFS, to webpage,...) But how can I store it just in PSRAM or RAM?

wzli, 2019-12-08:
you can call esp_camera_fb_return(fb) 5 minutes later?

Adrianotiger, 2019-12-08:
Thanks, I tryed, but I got only 1/2 image some times... I found ho to save on spiram now (I have to copy 16kb many times). So I can store 4 images before using them (1-5 minutes later).

closed 2019-12-08 by Adrianotiger