espressif/esp32-camera #97: Rotating the output image clockwise or anticlock wise by 90 deg

sriranjanr, 2019-12-06:
Hello,

I have an issue. I need to rotate the output image by 90 deg clockwise or anti clock wise (that is viewed in the browser in cameraserver example). Please let me know how to do it?

sriranjanr, 2019-12-13:
Can somebody help me with this please..?

me-no-dev, 2019-12-13:
huh... I thought I answered this one already :)
You can not rotate it in the camera itself. To do it in software you need to have another buffer that you copy the data to rotated, then you turn that into JPEG

rbretmounet, 2023-04-01:
> huh... I thought I answered this one already :) You can not rotate it in the camera itself. To do it in software you need to have another buffer that you copy the data to rotated, then you turn that into JPEG

Could explain how this is done in software please?