espressif/esp32-camera #99: Why is VSYNC for JPEG only?

wzli, 2019-12-10:
I was running running into issues capturing a continuous stream of grayscale output, where sometimes the rows of the image would "roll over"
[image]
After digging a little, it turns out that in that the VSYNC interrupt was not used expect when output format is JPEG, and the fix was simply to bypass this logic.
Just curious why this was in place in the first place.

jjbubka, 2019-12-11:
I have faced this issue too. Using Vsync will prevent of this rolling, but you have to discard frames with less lines than expected. It seems that some lines are missing when esp32 is busy.

me-no-dev, 2019-12-11:
vsync is used for JPEG, will add this to the rest to discard half frames as well. Will see how this will go ;)

me-no-dev, 2019-12-11:
BTW if you use 1 frame buffer (and you should when not using JPEG), this should not happen a the code is waiting for the start of the frame and should overlap the next one only if it has missing data (as @jjbubka noted)

wzli, 2019-12-11:
I am actually using 3 frame buffers to pipeline from camera > vision processing > web streaming (for optional real-time debug), pumping low resolution grayscale at 50Hz (thanks to hints from @jjbubka )

It's quite the stress test, and I realized that anything except JPEG support seems to be uncharted territory currently. However, I'm building a highly reactive system, so frame-rate and latency takes priority. The alternative of JPEG > RGB > GRAYSCALE is too inefficient.

In any case, for most applications, I don't think there's any reason to keep incomplete frames (except for debugging the camera driver). And it's definitely a bug when the half frames persist to subsequent captures.

jjbubka, 2019-12-11:
If you use multithreading, it helps to minimize bad frames if you only have camera task in one core (core 1 e.g.), or light ones. Tasks using internet must be on the other core.

wzli, 2019-12-11:
> If you use multithreading, it helps to minimize bad frames if you only have camera task in one core (core 1 e.g.), or light ones. Tasks using internet must be on the other core.

Yeah, thanks, basically I'm running the image capture and vision processing pipeline in one core. and the non-critical wifi, TCP/IP, and streaming server with on the fly JPEG compression on the other.

wzli, 2019-12-11:
Actually, the bad frames happen very rarely in my setup, even when one core is fully loaded with image processing, and the other serving multiple simultaneous streams of different filtered outputs (eg I ran it over night, didn't happen once).
After tracking down the problem, I was able to reproducible it when different parts of the RTOS enters critical section eg: calling taskENTER_CRITICAL() and taskEXIT_CRITICAL().
Mostly I encountered it because, I had URI request handler to display tasks/core/heap usage in the WebUI, which in response calls uxTaskGetSystemState() has a critical section.

closed 2020-03-31 by wzli