espressif/esp32-camera #101: OV2640 standby support

uname2219, 2019-12-12:
Hello,

Has anyone been able to use the software power-down available on the OV2640? When I send the command, I see a drop in power, but the sensor also locks the I2C SDA pin low until it is power-cycled.

The OV2640 datasheet claims "Executing a software power-down through the SCCB interface suspends internal circuit activity but does not halt the device. All register content is maintained in standby mode." That's exactly what I would like to do - configure the camera, put it in standby, then bring it out of standby and be able to use it immediately.

I added a set_sleep function to the ov2640.c driver:
```
static int set_sleep(sensor_t * sensor, int enable)
{
    sensor->status.sleep = enable;
    int ret = set_reg_bits(sensor, BANK_SENSOR, COM2, 4, 1, enable?1:0);
    return ret;<br>
}
```

When I call this function, I see my board current drop ~30mA (@5V input) which seems reasonable. However, watching the I2C SDA pin, I see it is being held low by the camera. This means I can't send a command to come out of standby. I also can't use any other I2C devices on the bus until a power cycle of the camera. Based on several I2C troubleshooting documents, I tried sending a series of SCL pulses to see if that would reset the camera with no luck.

At this point, entering software standby appears to be a one-way operation that has no advantage over the hardware PWDN signal since it in not reversible. I'd love to hear from anyone who has been able to use this feature and learn what I should do differently.

VooDust, 2020-03-26:
I, too, would like to know. I can't even put into sleep:

sccb: SCCB_Write Failed addr:0x30, reg:0x09, data:0x12, ret:263

Aren't there any decent cameras on the market for makers instead of this OV crap?

VooDust, 2020-03-26:
After some guesswork, comparing with other OV models (OV3640, I tried to completely suspend XCLK and resume it before waking up, to no avail. It remains a mistery.

me-no-dev, 2020-03-27:
I did some research and will implement this very soon. It must be based on pins+registers, depending on what is available. The function of the PWDN pin is to suspend the camera, while keeping all registers unchanged. If your module has the PWDN broken, you can give it a shot.

uname2219, 2020-03-27:
Thank you for investigating. My goal is to reduce power usage without needing an ESP32 pin.

Right now I cut power to the chip externally using an ESP32 pin to control a FET. This works but I would prefer to put the OV2640 into this software standby mode and then bring it out of standby. I can see the chip going into standby - I just don't know how to recover.

ankit3600, 2024-03-27:
@me-no-dev I am also facing this issue.
Do you have a solution in mind that you can implement.

Appreciate your response.
Thanks in advance :)

jksemple, 2024-07-17:
I would like a solution to this as well.