espressif/esp32-camera #102: esp-32  Camera probe failed with error 0x20004

PriontoAbdullah, 2019-12-13:
E][camera.c:1049] camera_probe(): Detected camera not supported.
[E][camera.c:1249] esp_camera_init(): Camera probe failed with error 0x20004

me-no-dev, 2019-12-13:
sure... but what camera address did it find? what is the board you are using, the example... just by those two lines you can not be helped ;)

guatemartin, 2019-12-27:
Hi,
   I get the same error.  Board is an ESP32 -Cam with AI Thinker ESP32-S chip.  The camera on board says it is an ov-2640 v2.0, or TY-OV2640-V2.0 to be exact.  Below is the spit out from  Serial Monitor.

  Have ESP32 1.0.4 installed, and using the  ESP32 WRover Module Board.
  Have #define CAMERA_MODEL_AI_THINKER , uncommented.
  Using Arduino 1.8.10.  Trying to run the CameraWebServer example.

  I also get the "Brownout detector was triggered" error a couple of times before the below info, but I know how to turn that off if needed.

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

[E][camera.c:1049] camera_probe(): Detected camera not supported.
[E][camera.c:1249] esp_camera_init(): Camera probe failed with error 0x20004

guatemartin, 2019-12-28:
Hi,

  Thought I would follow up.  I have changed so many things, that I can't say for certain what fixed it.  The last thing I changed was the usb cable, as I am powering up an arduino UNO instead of an ftdi adapter to program the esp-cam board.  Once I changed the cable, and made sure it was getting enough power, it seems to of have resolved itself.  I still got SCCB errors etc for a while, but that all seemed to have fixed itself after reloading up the CameraWebServer sketch.  Cheers.

e-labInnovations, 2020-01-29:
### Same error
[E][camera.c:1049] camera_probe(): Detected camera not supported.
[E][camera.c:1249] esp_camera_init(): Camera probe failed with error 0x20004

### Camera label:
HW-297
OV2640

### In program:
#define CAMERA_MODEL_AI_THINKER

### Board selection:
ESP32 Wrover Module

bkeevil, 2020-01-29:
@e-labInnovations 

I'm not in a position to check atm but I recall there being on option for the WRover module in `make menuconfig`. 

Also isn't it CONFIG_CAMERA_MODEL_XXX

Carlos1Eduardo1Colpo, 2020-02-01:
esse erro pode ocorrer pela alimentação inapropriada tente a alimentação de 5V em vez de 3.3V

v12cat, 2020-02-24:
The brownout will be happening when the ESP starts WiFi. Most usb leads and ports struggle with the 200ma+ peak load when it connects. Place a serial print and  a delay in the code just before wifi.begin. Then another after. I bet it hangs before the second print.

Power the board from a 5v PSU and disconnect the vcc from the FTDI programmer if you like.

zakkie, 2020-04-21:
I got the same error.
Power off and on ESP32-CAM, the error is no longer there.
hummm....

Bighoneypot, 2020-04-29:
I solved this problem using the USB port of the PC without hub.
For performance use a PowerBank

mazjed, 2020-05-06:
Hi i came  across this error with the same board . the cause was in the sketch i was using 
#define CAMERA_MODEL_WROVER_KIT was enabled by default  i had to add // like this
//#define CAMERA_MODEL_WROVER_KIT to disable it

supperka, 2020-05-15:
yes, thank you! I had same problem.

mfricci1, 2020-06-19:
> Power the board from a 5v PSU and disconnect the vcc from the FTDI programmer if you like.

How do I get the localhost address from COM if I power it separately?

Bamzz027, 2020-06-29:
Hi,, I have same problem with ESP32-S CAM
My error like this ;

Brownout detector was triggered

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

[E][camera.c:1049] camera_probe(): Detected camera not supported.
[E][camera.c:1249] esp_camera_init(): Camera probe failed with error 0x20004

Please help me to solved this problem..

Constantin17, 2020-06-30:
Hy, I had same problem but now it is work. Try this:
On board select: ESP32 Wrover Module
and
in program select:   #define CAMERA_MODEL_AI_THINKER
I changed also: Partition Scheme: Huge APP (3MB no OTA/1MB SPIFFS).
Good luck.

aseem179, 2020-07-05:
hey there iam having problem ---
[E][camera.c:1049] camera_probe(): Detected camera not supported.
[E][camera.c:1249] esp_camera_init(): Camera probe failed with error 0x20004

Bamzz027, 2020-07-06:
> Hy, I had same problem but now it is work. Try this:
> On board select: ESP32 Wrover Module
> and
> in program select: #define CAMERA_MODEL_AI_THINKER
> I changed also: Partition Scheme: Huge APP (3MB no OTA/1MB SPIFFS).
> Good luck.

Already use this, but got the same problem like before

aseem179, 2020-07-07:
thanks for your suggestion but i have uploaded code from this specification
and it is not working and sorry for late response i have solved thisproblem

just-jason, 2020-07-07:
@Bamzz027  

> 
> Brownout detector was triggered
> 
You have power supply issues. Use a seperate 5v supply, do not try to power the board from the serial adapter.

Sojourneer, 2020-07-15:
I powered the ESP32CAM from a wallwart, and the FTDI via USB, connecting the two only by Rx/Tx. That seems to work fine, except possibly for the gibberish that comes out on the serial monitor after it starts working.  Camera data I presume, in which case no problem.

stanislav-brabec, 2020-07-22:
I have the same problem here. But in my case, I am sure that my software setup worked before. Cable is OK, power supply is OK as well.

I am starting to suspect a broken hardware. If I add support for software PWM LED, I can succeed sometimes, but only if PWM is set to 84% off, 16% on. It sounds very strange. I measured all camera power voltages, and everything seems to be OK.

To confirm hardware issue, I am waiting for a new module.

liv-in-sky, 2020-07-25:
I had the same errors 
solution for me:
On board select: ESP32 Wrover Module
and
in program select: #define CAMERA_MODEL_AI_THINKER
I changed also: Partition Scheme: Huge APP (3MB no OTA/1MB SPIFFS).

AND I HAVE TO USE AN OLDER BOARD DRIVER (1.0.2) -  only this worked
[image]

Jakemmd, 2020-08-06:
I had same issues as noted above, tried everything, nothing worked, then I tried a new fdti adaptor and worked 1st time, pretty sure it was a voltage issue

stanislav-brabec, 2020-08-06:
> I had same issues as noted above, tried everything, nothing worked, then I tried a new fdti adaptor and worked 1st time, pretty sure it was a voltage issue

I will try a different power source. But I measured voltage and even voltace fluctuation (with 1 ms resolution), and everything was OK. (And the board worked with the same convertor few weeks ago.)

Yes, there is some type of voltage problem, as the camera makes a snapshot in the moment of turning LED on, and I am even able to make camera temporarily working with the big LED in a PWM mode somewere between 16 to 20% on.

My ESP32-CAM is not made by AI Thinker. It is a clone with a module with the Espressif logo.

Jimmy76gr, 2020-08-11:
after check all found that is power issue, check and/or replace the usb serial adapter, this base on computer that you use and the USB power supply, many case two usb nearby into a pc cannot supply approriate power for camera.

aseem179, 2020-08-11:
thanks to all of your suggestions but I solved this problem earlier and
sorry for the late response

vult-r, 2020-08-23:
Same issue (VSYNC failures)   figured using this call in setup() was causing the issue:
randomSeed(analogRead(0));

// randomSeed(analogRead(0)); fixed that

stanislav-brabec, 2020-08-31:
> I will try a different power source. But I measured voltage and even voltace fluctuation (with 1 ms resolution), and everything was OK. (And the board worked with the same convertor few weeks ago.) [...]

In my case nothing helped except board replacement. The new board with the old camera works.

NikunjChiluvuri, 2020-09-08:
ok.... i used this esp32 cam a long time ago. it worked then with a 5 v power supply. now, with my laptop, it is not working. try connecting it to a 5 v power supply.

NikunjChiluvuri, 2020-09-08:
> In my case nothing helped except board replacement. The new board with the old camera works.

well... then you have a worng code in your old one.

stanislav-brabec, 2020-09-08:
> > In my case nothing helped except board replacement. The new board with the old camera works.
> 
> well... then you have a worng code in your old one.

Both boards have exactly the same code uploaded.

NikunjChiluvuri, 2020-09-09:
> > > In my case nothing helped except board replacement. The new board with the old camera works.
> > 
> > 
> > well... then you have a worng code in your old one.
> 
> Both boards have exactly the same code uploaded.

well..................... i meant bootloader. try uploading a new bootloader.

stanislav-brabec, 2020-09-09:
> well..................... i meant bootloader. try uploading a new bootloader.

I just copied the new device bootloader to the old one:
```
esptool.py --port /dev/ttyUSB5 --chip esp32 --after hard_reset read_flash 0x1000 0x4000 bootloader_dio_40m.bin
```
on the working device, and then
```
esptool.py --port /dev/ttyUSB5 --chip esp32 --before default_reset --after hard_reset write_flash 0x1000 bootloader_dio_40m.bin
```
on the broken device.

There is no change. Camera works only on the new device.

Both devices are equal with exception of MAC:
```
Chip is ESP32D0WDQ6 (revision 1)
Crystal is 40MHz
Warning: ESP32 has no Chip ID.
Manufacturer: 20
Device: 4016
Detected flash size: 4MB
Status value: 0x0200
```

NikunjChiluvuri, 2020-09-11:
> I just copied the new device bootloader to the old one:
>
> ```
> esptool.py --port /dev/ttyUSB5 --chip esp32 --after hard_reset read_flash 0x1000 0x4000 bootloader_dio_40m.bin
> ```
>
> on the working [...]

IF YOU READ IT, YOU CAN SEE THAT IT SAYS THAT THE CHIP HAS NO ID.

NikunjChiluvuri, 2020-09-11:
> IF YOU READ IT, YOU CAN SEE THAT IT SAYS THAT THE CHIP HAS NO ID.

AND THAT MEANS THAT THE FLASH IS NOT WORKING OR THE CHIP IS BROKEN.

NikunjChiluvuri, 2020-09-11:
TRY ERASING THE FLASH SETTINGS IN ARDUINO IDE AND TRY AGAIN. IF IT STILL DOESEN'T WORK, TRY FINDING A GOOD BOOTLOADER FOR THE ESP32 AND FLASH IT ON YOUR BROKEN ESP32. IF IT STILL DOESENT WORK, I WILL TELL WHAT TO DO.

NikunjChiluvuri, 2020-09-11:
> AND THAT MEANS THAT THE FLASH IS NOT WORKING OR THE CHIP IS BROKEN.

TRY ERASING THE FLASH SETTINGS IN ARDUINO IDE AND TRY AGAIN. IF IT STILL DOESEN'T WORK, TRY FINDING A GOOD BOOTLOADER FOR THE ESP32 AND FLASH IT ON YOUR BROKEN ESP32. IF IT STILL DOESENT WORK, I WILL TELL WHAT TO DO.

stanislav-brabec, 2020-09-11:
> IF YOU READ IT, YOU CAN SEE THAT IT SAYS THAT THE CHIP HAS NO ID.

I don't know what is the meaning of this message. But the same message appears in most ESP32-CAM dumps on the internet.

> TRY ERASING THE FLASH SETTINGS IN ARDUINO IDE AND TRY AGAIN. IF IT STILL DOESEN'T WORK, TRY FINDING A GOOD BOOTLOADER FOR THE ESP32 AND FLASH IT ON YOUR BROKEN ESP32. IF IT STILL DOESENT WORK, I WILL TELL WHAT TO DO.

What exactly means "erasing the flash settings" in Arduino IDE? I see no such option. I see only option to select board. I picked AI Thinker ESP32-CAM.

I found a good bootloader on the working board. I copied the bootloader from the working board to the broken board, so both have the same bootloader. One works, one does not work:

```
[E][camera.c:1049] camera_probe(): Detected camera not supported.
[E][camera.c:1249] esp_camera_init(): Camera probe failed with error 0x20004
```

NikunjChiluvuri, 2020-09-11:
there is an option. see on internet on how. i can't explane. google will tell you how to do it.

NikunjChiluvuri, 2020-09-11:
and ok.

ronitsrma16, 2020-10-04:
@NikunjChiluvuri, I unable to find any option on google. Can you please put some light on it how yo get rid of this error?

NeilStewart19, 2020-11-02:
I solved this error by making sure the white line on the camera connector is still just visible when the retaining tab is closed. Do not push the camera connector all the way in.

NikunjChiluvuri, 2020-11-02:
> I solved this error by making sure the white line on the camera connector is still just visible when the retaining tab is closed. Do not push the camera connector all the way in.

i tried this but it doesn't work

ianiliff, 2020-11-02:
I am looking to implement this board in a production device in the near future, so these comments are very helpful to my testing. I have found some variation between boards which is a bit off putting. I have a few running stable by doing this:
Sending power straight from a 5v 2A adapter to the 5v and adjacent ground pin.
Using the ESP32 Dev Module board using the Default 4MB with spiffs partition (My application requires OTA).

I only have tested with a hand full of boards and based on their low cost I will be trying more, But i have found at least 25% of them having issues with this camera module. I will try some of the specified fixes you guys mentioned.

NeilStewart19, 2020-11-02:
> I am looking to implement this board in a production device in the near future, so these comments are very helpful to my testing. I have found some variation between boards which is a bit off [...]

I’ve only been using one - but it definitely has some tolerance issues with the insulation mask on the cable. Moving the position of the cable back and forth in the connector has solved it for me.

NikunjChiluvuri, 2020-11-03:
> > I am looking to implement this board in a production device in the near future, so these comments are very helpful to my testing. I have found some variation between boards which is a bit off-putting. I have a few running stables by doing this:
> > Sending power straight from a 5v 2A adapter to the 5v and adjacent ground pin.
> > Using the ESP32 Dev Module board using the Default 4MB with spiffs partition (My application requires OTA).
> > I only have tested with a hand full of boards and based on their low cost I will be trying more, But I have found at least 25% of them having issues with this camera module. I will try some of the specified fixes you guys mentioned.
> 
> I’ve only been using one - but it definitely has some tolerance issues with the insulation mask on the cable. Moving the position of the cable back and forth in the connector has solved it for me.

for me, that is not a problem. it still works

NikunjChiluvuri, 2020-11-03:
> I am looking to implement this board in a production device soon, so these comments are very helpful to my testing. I have found some variation between boards which is a bit off-putting. I have a few running stables by doing this:
> Sending power straight from a 5v 2A adapter to the 5v and adjacent ground pin.
> Using the ESP32 Dev Module board using the Default 4MB with spiffs partition (My application requires OTA).
> 
> I only have tested with a hand full of boards and based on their low cost I will be trying more, But I have found at least 25% of them having issues with this camera module. I will try some of the specified fixes you guys mentioned.

for me, the second one is working.

stanislav-brabec, 2020-11-24:
It seems that I identified a cause of a (permanent?) damage of two my boards: **Attempt to use ledc driver  (software PWM) to the GPIO4 (LED) damages the board** in ESPHome.

I had a new board. I used if for some time with CameraWebServer. Everything was OK.

Then I flashed ESPHome. All was still OK.

But when I added these lines, board got damaged:

```
output:
  - platform: ledc
    pin: GPIO4
    id: gpio_4_1

light:
  - platform: monochromatic
    gamma_correct: 5
    output: gpio_4_1
    name: "Camera 1 LED"
    id: esp32cam1_led
```

Now, trying any firmware, camera does not work any more.

I am nearly sure that my previous board was damaged in the same way!

Note that plain LED switch does not damage the board.

stanislav-brabec, 2020-11-24:
Confirming. https://github.com/esphome/issues/issues/619

NikunjChiluvuri, 2020-11-24:
> It seems that I identified a cause of a (permanent?) damage of two my boards: **Attempt to use ledc driver (software PWM) to the GPIO4 (LED) damages the board** in ESPHome.
>
> I had a new board. I [...]

i think that it is shorting the board or the led it taking a lot of power from the chip and it is getting burned.

bNjtHub, 2020-11-28:
> I solved this error by making sure the white line on the camera connector is still just visible when the retaining tab is closed. Do not push the camera connector all the way in.

Thank you very much !!! Finally i found a solution of my issue !!!
I had the 
"[E][camera.c:1049] camera_probe(): Detected camera not supported. 
[E][camera.c:1249] esp_camera_init(): Camera probe failed with error 0x20004****"

stanislav-brabec, 2020-11-28:
> i think that it is shorting the board or the led it taking a lot of power from the chip and it is getting burned.

After several days, I can give a better answer: There is a complex fix. My board is works again.

I  can confirm the second problem mentioned here: a very limited number of lifetime insertions of the ZIF/FPC joint. Each insertion makes small holes in the cable terminal (visible with a magnifying glass). Next time, you have to find a different position that is not affected by these holes. If you do this several times, you can have a big trouble.

I suspect that the manufacturer of the camera module used a soft glue between the polyimide flexible printed circuit (FPC) and the reinforcing pad. Each click of the zero insertion force connector (ZIF) lock deforms the glue and the polyimide with the copper+gold contact.

In my case, all problems mentioned above came together:
- Firmware with an camera PWM channel reprogrammed to LED.
- Problems with camera ZIF connector on the board.
- Small holes in the camera FPC cable connector.

It caused that some combinations of board + camera worked, some did not.

A full fix consists of:
- Upload a correct firmware that reprograms ledc PWM to use channel 2. Bootloader does not do that. Nor the minimal ESPHome firmware. You have to upload a firmware that explicitly reprograms ledc PWM and execute it. Then you can upload any firmware again.
- Clean the ZIF connector on the board with a contact cleaner.
- Clean the camera FPC cable and then insert it in a position that is not affected by holes created by a previous insertion.

gin66, 2020-12-08:
Have the same 0x20004 issue. What seems to help:
- set PWDN_GPIO_NUM to HIGH
- wait a while
- set PWDN_GPIO_NUM to LOW
- wait a while
- restart or redo camera initialization

This sequence powers off and on the camera chip. In my logs, it may take up to three tries until it finally works.

Next I will check, what the driver is doing with this line. Eventually I will redefine PWDN_GPIO_NUM to -1, so that the camera driver does not fiddle around with the camera’s power line and control it from my app.

gin66, 2020-12-08:
camera driver does this:
        gpio_set_level(config->pin_pwdn, 1);
        vTaskDelay(10 / portTICK_PERIOD_MS);
        gpio_set_level(config->pin_pwdn, 0);
        vTaskDelay(10 / portTICK_PERIOD_MS);

So I will turn that off and use seconds in between in order to let the power circuit and camera to settle down.

gin66, 2020-12-08:
works much better now. five times wake up out of deep sleep and no camera init failure

Kabron287, 2020-12-19:
I get these errors in case of MLX90640 is connected via the same SDA/SCL pins.
It is strange because no addresses conflicts are possible.
On the scope I see no differences in SDA/SCL signals levels and forms.

me-no-dev, 2020-12-29:
driver is not set to share the I2C bus. Probably the address of your other device is found and the driver is trying to use that to connect.

vlast239, 2021-01-04:
Hi !
Several times I caught in the logs "Brownout detector was triggered" so disabled the detector. Because first I fed the backlight display from the output of the 3.3V ESP32-CAM

Code
#include "soc/rtc_cntl_reg.h"
...
WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0); //disable brownout detector

Source
https://temofeev.ru/info/articles/esp32-cam-video-streaming-server-podklyuchenie-i2c-i-spi-displeev/

Kabron287, 2021-01-04:
I solved it by supplying MLX90640 ground via TFT backlight and switching backlight after camera init.

vlast239, 2021-01-04:
It's good

dietermontanez, 2021-02-05:
[image]

I had similar problems. The WIFI "brownout" issues and permanent rebooting I solved by connecting an external 
3.3v supply to the ESP32 (lower left in picture above). Still connecting simultaneously with the IDE via USB8 on the laptop..

The code i used is a bastradization of "https://www.electronics-lab.com/project/video-streaming-server-esp32-cam/"
to which i added some SSD1306 display features to do debugging while not connected to the USD serial console.

The issue with error 0x20004 was solved by defining the correct I/O. I modified the "defines" as follows:

#if defined(CAMERA_MODEL_WROVER_KIT)
#define PWDN_GPIO_NUM    26
#define RESET_GPIO_NUM   -1
#define XCLK_GPIO_NUM    32
#define SIOD_GPIO_NUM    13
#define SIOC_GPIO_NUM    12

#define Y9_GPIO_NUM      39
#define Y8_GPIO_NUM      36
#define Y7_GPIO_NUM      23
#define Y6_GPIO_NUM      18
#define Y5_GPIO_NUM      15
#define Y4_GPIO_NUM       4
#define Y3_GPIO_NUM      14
#define Y2_GPIO_NUM       5
#define VSYNC_GPIO_NUM   27
#define HREF_GPIO_NUM    25
#define PCLK_GPIO_NUM    19

that matches the "datasheet" from the manufacturer:

[image]

[23 more replies]

closed 2022-06-21 by PriontoAbdullah