espressif/esp32-camera #105: framesize seem not working for me.

kotran88, 2019-12-22:
https://storage.googleapis.com/kaka-218303.appspot.com/wwwpro%2FThumbnail3.jpg?GoogleAccessId=firebase-adminsdk-d6rqe%40kaka-218303.iam.gserviceaccount.com&Expires=16730290800&Signature=M5qR3iHB524v9iWvYkfXO547VeJNm20lvvc%2F07UqaELRYb4IZbm5vI%2B9%2FwkeegyoLmwBhksHE7uWeLrP1x3DaoxUbfEXtJ75f6YsnZcQFPPg8oyfYcvGNlP2fbK4vHeJnMEMjFTCH%2BD4OPaY8zfiRUettC8t9baTu%2FUAA2EAlpiYwk9eQJYIbjwLiumoKcvFkd%2Fx4njBz5Zpbay8820TfzbDma1oRFs8t05aKP50xFH1vara2Y7A9OGWXG2dGlucio0o8qXrX0x9Hf1teNAcfDAN7rsaZjNXPVWLgdlE9DhUQPDyeQHeeyDB0pXtClTsCr1SXAERjxLPHf7cz9XXqw%3D%3D

above link is the one that I capture image with esp-32 cam. 
but framesize is too small and quality is not good...
I tried to change framesize in code, but all not working. 

how can I have bigger framesize and better quality?
framesize UXGA is the mos bigger one? but it seems same as above...

  camera_config_t config;
  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer = LEDC_TIMER_0;
  config.pin_d0 = Y2_GPIO_NUM;
  config.pin_d1 = Y3_GPIO_NUM;
  config.pin_d2 = Y4_GPIO_NUM;
  config.pin_d3 = Y5_GPIO_NUM;
  config.pin_d4 = Y6_GPIO_NUM;
  config.pin_d5 = Y7_GPIO_NUM;
  config.pin_d6 = Y8_GPIO_NUM;
  config.pin_d7 = Y9_GPIO_NUM;
  config.pin_xclk = XCLK_GPIO_NUM;
  config.pin_pclk = PCLK_GPIO_NUM;
  config.pin_vsync = VSYNC_GPIO_NUM;
  config.pin_href = HREF_GPIO_NUM;
  config.pin_sscb_sda = SIOD_GPIO_NUM;
  config.pin_sscb_scl = SIOC_GPIO_NUM;
  config.pin_pwdn = PWDN_GPIO_NUM;
  config.pin_reset = RESET_GPIO_NUM;
  config.xclk_freq_hz = 20000000;
  config.pixel_format = PIXFORMAT_JPEG;
  //init with high specs to pre-allocate larger buffers
  if(psramFound()){
    Serial.println("gotouxga");
    config.frame_size = FRAMESIZE_UXGA;
    config.jpeg_quality = 1;  //0-63 lower number means higher quality
    config.fb_count = 2;
  } else {
    config.frame_size = FRAMESIZE_UXGA;
    config.jpeg_quality = 12;  //0-63 lower number means higher quality
    config.fb_count = 1;
  }
  
  // camera init
  esp_err_t err = esp_camera_init(&config);
  if (err != ESP_OK) {
    Serial.printf("Camera init failed with error 0x%x", err);
    delay(1000);
    ESP.restart();
  }

  //drop down frame size for higher initial frame rate
  sensor_t * s = esp_camera_sensor_get();
  s->set_framesize(s, FRAMESIZE_QQVGA);  // VGA|CIF|QVGA|HQVGA|QQVGA   ( UXGA? SXGA? XGA? SVGA? )
 s->set_quality(s,20);

me-no-dev, 2019-12-22:
do not use .jpeg_quality = 1

kotran88, 2019-12-22:
> do not use .jpeg_quality = 1

ok I used 10 or 12 but it's same.  FRAMESIZE_UXGA is the best framesize?

me-no-dev, 2019-12-23:
the best framesize depends on the sensor that you are using. UXGA is the largest for OV2640

me-no-dev, 2019-12-23:
```
config.xclk_freq_hz = 20000000;
config.pixel_format = PIXFORMAT_JPEG;
config.frame_size = FRAMESIZE_UXGA;
config.jpeg_quality = 12;
config.fb_count = 2;
```

kotran88, 2019-12-23:
> ```
> config.xclk_freq_hz = 20000000;
> config.pixel_format = PIXFORMAT_JPEG;
> config.frame_size = FRAMESIZE_UXGA;
> config.jpeg_quality = 12;
> config.fb_count = 2;
> ```

it seems same as my code....I don't get it because photo is too small for UXGA....is it 1600X1200? 
as you can see link above...

me-no-dev, 2019-12-25:
yup, it's 1600x1200... now why do you get that small image.... I am at a loss.... have you tried the web server demo as-is?

Schaggo, 2019-12-30:
I just came across this post. 

> 

do not use .jpeg_quality = 1

> 

Why should I not? is there a way to use lower compression / get better image quality?

me-no-dev, 2019-12-30:
you should use RGB or YUV in that case. JPEG quality set to 1 might produce images too large to handle in a one frame time.

Schaggo, 2019-12-31:
Thanks for the reply! 

I get stable results with quality set to 3. 
What is the bottle neck? PSRAM speed? 

I'd really like to get max quality. 
What would it take?

Thanks and a happy new year!

fritsjan, 2020-08-04:
For future reference: 
In the camera example, note that first the framesize is set to FRAMESIZE_UXGA (when psram is found) or FRAMESIZE_SVGA. The last one is when no external PSRAM is found because then we would not have enough memory for the buffer.

**AND NOTE THEN**, further down the line in the example sketch the initial framesize is dropped down!
s->set_framesize(s, FRAMESIZE_QVGA);

This is then the actual framesize being used when taking a picture (while the buffer size will be the size of an UXGA image).

So if you wish to use a higher framesize, then use:
s->set_framesize(s, FRAMESIZE_UXGA);

These are the valid sizes for the OV2640 sensor:
// FRAMESIZE_ + QVGA|CIF|VGA|SVGA|XGA|SXGA|UXGA

DallaThaun, 2020-12-16:
> For future reference:
> In the camera example, note that first the framesize is set to FRAMESIZE_UXGA (when psram is found) or FRAMESIZE_SVGA. The last one is when no external PSRAM is found because [...]

Why are we not able to use 720p exactly? I'm curious since this camera is advertised as 720p. The options in this file are better or worse than 720p, but not exactly that. Is there a reason?

me-no-dev, 2020-12-16:
you can.

DallaThaun, 2020-12-16:
How? The only way I'd think of is adding to the header file, but I can't find the .src to do so. I thought if they didn't include that as an option, there might be a good reason?

me-no-dev, 2020-12-17:
https://github.com/espressif/esp32-camera/blob/master/driver/include/sensor.h#L45

chupocro, 2021-06-14:
> For future reference:
> In the camera example, note that first the framesize is set to FRAMESIZE_UXGA (when psram is found) or FRAMESIZE_SVGA. The last one is when no external PSRAM is found because [...]

There is an error in the camera example user interface for selecting the resolution - the resulting resolution is always **one step lower than the one selected**. For example when you select **SVGA (800x600)** the resolution is **VGA (640x480)**, when you select **VGA (640x480)** the resolution is **CIF (400x296)** etc. The easyest way of checking the picture resolution is to just drag'n'drop the image into some folder and to check the image dimensions.

When setting:

`s->set_framesize(s, FRAMESIZE_UXGA);`

the selected resolution is **"nothing" (the drop down menu is blank)** and the output image really is 1600x1200 but if you then change the resolution using the user interface there is no way of enabling 1600x1200 again without restarting the microcontroller.