espressif/esp32-camera #110: ov3660 FRAMESIZE_QXGA not working

Schaggo, 2020-01-07:
I am trying to get full resolution with ov3660 modules. 
Framesite is set to QXGA but image output is still 1600x1200 pixels.

Any tip?

Cheers

me-no-dev, 2020-01-08:
it probably has to do with the fact that some new resolutions were added and indexes changed. How are you setting the resolution?

Schaggo, 2020-01-08:
The project is made with vscode /platformio using the Arduino framework.
I simply tried to use

`config.frame_size = FRAMESIZE_QXGA; `

in my main.cpp

me-no-dev, 2020-01-08:
and you run the default example? frame size is not changed by other code?

Schaggo, 2020-01-08:
The start point for the project is the default example. 
I added minor things like print commands in camera.c to verify that the model is indeed a ov3660.
Apart from that the cam driver is vanilla.

I will try to rebuild the project from the ground up to make sure.

Schaggo, 2020-01-08:
ok. that did it. Jesus I'm a moron...
Thanks for the help and sorry for the rubber ducking!

me-no-dev, 2020-01-08:
it happens :)

closed 2020-01-08 by me-no-dev