espressif/esp32-camera #111: OV2640 - with h.264 output

WTHfloyd, 2020-01-16:
I working on getting a OV2640 with h.264 (I think I found one...). 
I requested specs for the camera - but sounds like it is possible...

How hard would it be to update the library to use h.264 rather than the current JPEG output?

And from there... I am thinking if gstreamer with a webRTC module can work on the esp32 ) - then you can do some live streaming to anywhere...... but not sure if that is possible. Or something of the sort...

Thanks.

Phil

me-no-dev, 2020-01-16:
not really doable on esp32. not enough memory and processing power to encode into h264 :) mJPEG is prolly the best you can get.

closed 2020-01-16 by me-no-dev

WTHfloyd, 2020-01-16:
Hmmm

No no the camera would do the encoding...not the esp32.
So output from camera is h.264.

Then run it through pre-encoded h.264 signal through esp32....

Is that possible?

me-no-dev, 2020-01-17:
I do not know of sensor that supports h264. I guess that is because it requires memory and processing that the sensors can not afford :)

WTHfloyd, 2020-01-17:
I am talking to a factory to get costs, etc... for such a camera - waiting for details. They said they can make it. However Chinese New Year coming up so will take a bit of time to get info.

But if I could find a low cost camera that outputted h.264 video as the current OV2640 with similar speeds and specs - can you adapt the library to work?
And what data specs are important?
What info would you need from the camera to make to make it work?

Thx.

me-no-dev, 2020-01-17:
The camera needs to support 8 bit camera interface for the image data + I2C or SPI for communication. MIPI will not work.

gitcnd, 2023-01-18:
There's lots around ( https://www.aliexpress.com/i/32963953305.html)  but the all seem to use USB - so you'd need to write code to read a USB Video Class (UVC) stream to get it to work...