espressif/esp32-camera #118: ESP32 CAM - Detected camera not supported

lhadvab, 2020-02-06:
Hello, 
I have an ESP32-Cam modul from Aliexpress: https://www.aliexpress.com/item/32971094258.html?spm=a2g0s.9042311.0.0.27424c4dreFgYS .  I followed this video: https://youtu.be/MicAM_A0_lU .

 I always get the same errors:
[E][camera.c:1049] camera_probe(): Detected camera not supported.
[E][camera.c:1249] esp_camera_init(): Camera probe failed with error 0x20004

I´ve also tried this: https://github.com/espressif/esp32-camera/issues/74 , but even this wasn´t helpful.

Can you help me with this?

YashwantPanwar, 2020-02-11:
Hello lhadvab ,
1.Please  Change your connecting USB Cable. may be it has some voltage problem so your USB Cable not working properly .
2.Change your connecting port. Use laptop port .

YashwantPanwar, 2020-02-11:
Please  tried this.....

https://randomnerdtutorials.com/esp32-cam-video-streaming-face-recognition-arduino-ide/

lhadvab, 2020-02-12:
I´ve skipped this important step:
1) Go to Tools > Board and select AI-Thinker ESP32-CAM.
Before that, I had selected Wrover module.

This was very important thing because I have AI-Thinker module. 
But I have some problem: 
[E][camera.c:1049] camera_probe(): Detected camera not supported.
[E][camera.c:1249] esp_camera_init(): Camera probe failed with error 0x20004

lhadvab, 2020-02-12:
I also tried different USB cable, but with no result.

liuandong, 2020-02-18:
`#define CAM_PIN_PWDN    32
#define CAM_PIN_RESET   -1`

reset is -1 not 32
32 is pwdn this is alway LOW

DoddyD, 2020-02-20:
> `#define CAM_PIN_PWDN 32 #define CAM_PIN_RESET -1`
> 
> reset is -1 not 32
> 32 is pwdn this is alway LOW

@liuandong   Is it in the camera_pins.h file ?

lhadvab, 2020-02-29:
@liuandong I´ve tried it, but I still have same problem.  I´ve also tried 5V instead of 3,3V. Any of these are not working. I´am not using original blue arduino usb cable, maybe this is reason why it isn´t working.

while0l1, 2020-03-01:
Please try this. I solved the problem. 😁😁
https://github.com/espressif/esp32-camera/issues/66#issuecomment-525960018

jojos115, 2020-03-24:
My problem same as you "Detected camera not support". Many way I've try still can get it working untill today 😁. I do some google and find out at the website maybe esp pin not solder properly and some say camera pins code not same as on board. So it is true about the camera pin and i make a test. first find esp32 cam circuit diagram, take your multimeter, check camera pin to esp32 pin, check the camera_pins.h, and make the correction on gpio num.. Goodluck buddy 👍

xenpac, 2020-03-24:
Yes these little esp32-cam boards are ..delicate :-)
There is always the chance to have hardware issues.
Mine had bad i2c SDA signal being clamped to 1.2V. so it could not probe the camera.

in general, a connected serial cable will cause wifi issues and slooow network transmissions.
(probably because the radio signal gets weekened by the cable)
Also a connected program-enable-cable will decrease the cam clock signal.
The high power LED will cause power surges which causes unexpected malfunction.

The OV2640 to be probed, 
- needs an input clock signal XCLK
- needs a powercycle (PWDN pin controls the 2 voltage regulators).
- Then it needs some init being written to function (ie.VSYNC clocks visible)

rekomerio, 2020-06-20:
I had this issue and after taking power from external power supply instead of the ftdi programmer, the issue got fixed

aamacaya, 2020-06-29:
@lhadvab  I had the same issue, but it was a pin configuration problem. Here is my code and configuration. I hope it helps.

Software:
- using esp-idf, not arduino.
- Remember to clone this repository on a folder called "components" in your project folder.

Hardware:
- ESP32-CAM
- Programmer: FTDI232 
    - in 3.3V config
    - Only using GND, VCC, TX and RX

Configuration (menuconfig):
- Component config > driver configurations > RTCI0 configuration > "Support array 'rtc_gpio_desc'
- Component config > Esp32-Specific > Support for external, SPI supported RAM

**CODE**
This code initializes the camera, takes a picture, and saves it on "fb" variable. Then it restarts. It does not do anything with the picture. Is just for testing purpose.
```
#include <stdio.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_spi_flash.h"
#include "driver/gpio.h"

#include "esp_camera.h"

//ESP32-CAM
#define CAM_PIN_PWDN    32 
#define CAM_PIN_RESET   -1 //software reset will be performed
#define CAM_PIN_XCLK    0//GPIO
#define CAM_PIN_SIOD    26//GPIO
#define CAM_PIN_SIOC    27//GPIO

#define CAM_PIN_D7      35//GPIO
#define CAM_PIN_D6      34//GPIO
#define CAM_PIN_D5      39//GPIO (sensor VN)
#define CAM_PIN_D4      36//GPIO (sensor VP)
#define CAM_PIN_D3      21//GPIO
#define CAM_PIN_D2      19//GPIO
#define CAM_PIN_D1      18//GPIO
#define CAM_PIN_D0       5//GPIO
#define CAM_PIN_VSYNC   25//GPIO
#define CAM_PIN_HREF    23//GPIO
#define CAM_PIN_PCLK    22//GPIO

static camera_config_t camera_config = {
    .pin_pwdn  = CAM_PIN_PWDN,
    .pin_reset = CAM_PIN_RESET,
    .pin_xclk = CAM_PIN_XCLK,
    .pin_sscb_sda = CAM_PIN_SIOD,
    .pin_sscb_scl = CAM_PIN_SIOC,

    .pin_d7 = CAM_PIN_D7,
    .pin_d6 = CAM_PIN_D6,
    .pin_d5 = CAM_PIN_D5,
    .pin_d4 = CAM_PIN_D4,
    .pin_d3 = CAM_PIN_D3,
    .pin_d2 = CAM_PIN_D2,
    .pin_d1 = CAM_PIN_D1,
    .pin_d0 = CAM_PIN_D0,
    .pin_vsync = CAM_PIN_VSYNC,
    .pin_href = CAM_PIN_HREF,
    .pin_pclk = CAM_PIN_PCLK,

    //XCLK 20MHz or 10MHz for OV2640 double FPS (Experimental)
    .xclk_freq_hz = 10000000,
    .ledc_timer = LEDC_TIMER_0,
    .ledc_channel = LEDC_CHANNEL_0,

    .pixel_format = PIXFORMAT_JPEG,//YUV422,GRAYSCALE,RGB565,JPEG
    .frame_size = FRAMESIZE_UXGA,//QQVGA-QXGA Do not use sizes above QVGA when not JPEG

    .jpeg_quality = 12, //0-63 lower number means higher quality
    .fb_count = 1 //if more than one, i2s runs in continuous mode. Use only with JPEG
};

esp_err_t camera_init1(){
    //power up the camera if PWDN pin is defined
    if(CAM_PIN_PWDN != -1){
        //pinMode(CAM_PIN_PWDN, OUTPUT);
        //digitalWrite(CAM_PIN_PWDN, LOW);
        gpio_pad_select_gpio(CAM_PIN_PWDN);
        gpio_set_direction(CAM_PIN_PWDN, GPIO_MODE_OUTPUT);
        gpio_set_level(CAM_PIN_PWDN, 0);
    }

    //initialize the camera
    esp_err_t err = esp_camera_init(&camera_config);
    if (err != ESP_OK) {
        return err;
    }

    return ESP_OK;
}

void app_main(void)
{
    printf("Begining of code\n");

    printf("Camera init...\n");
    esp_err_t res = camera_init1();
    if(res == ESP_OK)
    {
        printf("Camera init success!!\n");
    }
    else
    {
        printf("Camera init fail!!\n");
    }
    

    for (int i = 10; i >= 0; i--) {
        printf("Taking Picture in %d seconds...\n", i);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
    printf("Taking Picture Now!...\n");

    camera_fb_t * fb = esp_camera_fb_get();
    if (!fb) 
    {
        printf("Camera capture fail!!\n");
    }
    else
    {
        printf("Camera capture success!!\n");
    }

    fflush(stdout);
    esp_restart();
}
```

Aikidet, 2020-07-04:
Gehe auf die camara Pins und kopiere die  #elif defined(CAMERA_MODEL_AI_THINKER)  in die  #if defined(CAMERA_MODEL_WROVER_KIT) dann klappt es

Aikidet, 2020-07-05:
> Hallo,
> ich habe ein ESP32-Cam Modul von Aliexpress: https://www.aliexpress.com/item/32971094258.html?spm=a2g0s.9042311.0.0.27424c4dreFgYS . Ich habe dieses Video verfolgt: https://youtu.be/MicAM_A0_lU .
> 
> Ich bekomme immer die
> gleichen Fehler: [E][camera.c:1049] camera_probe(): Erkannte Kamera wird nicht unterstützt.
> [E][camera.c:1249] esp_camera_init(): Kamerasonde mit Fehler 0x20004 fehlgeschlagen
> 
> Ich habe auch versucht: #74 , aber auch das war nicht hilfreich.
> 
> Können Sie mir dabei helfen?
Gehe  in der Arduino IDE bei den Programm CameraWebserver auf die camera Pins und kopiere die #elif definiert(CAMERA_MODEL_AI_THINKER) Pins in die #if definiert(CAMERA_MODEL_WROVER_KIT) dann klappt es

lksmj, 2020-08-12:
I've worked this out for you know, uh... rx,tx This scalp was used for usb to ttl and powered by usb 5v succeeded.Thank you.
[image]

kerauzen, 2020-11-01:
i bougth ESP32-CAM
On the other hand, it bought a power supply realy bad 
said to deliver 2A with 6V 
When used with ESP32-CAM i measure the VCC to 3.5V 
So i change the power supply and problem 2004 disappears

Conclusion : check power supply...:(

raijadas, 2020-11-24:
In my case i **burn up** the ground pin or area with **reverse connection** during pin connection time. Just upload the code and use another ground pin for _ground_ terminal of power supply.
_Solution is always beneath the problem._

gdamoreira, 2021-04-29:
Hi everyone,

I'm facing the same problem with my ESP32-CAM, here are the details:

**Board:** ESP32-CAM (ESP32S)
**Camera module:** TY-OV2640-V2.0
**USB adapter:** ESP32-CAM-MB

I'm using CameraWebServer example with AI_THINKER model defined:
```
// Select camera model
//#define CAMERA_MODEL_WROVER_KIT // Has PSRAM
//#define CAMERA_MODEL_ESP_EYE // Has PSRAM
//#define CAMERA_MODEL_M5STACK_PSRAM // Has PSRAM
//#define CAMERA_MODEL_M5STACK_V2_PSRAM // M5Camera version B Has PSRAM
//#define CAMERA_MODEL_M5STACK_WIDE // Has PSRAM
//#define CAMERA_MODEL_M5STACK_ESP32CAM // No PSRAM
#define CAMERA_MODEL_AI_THINKER // Has PSRAM
//#define CAMERA_MODEL_TTGO_T_JOURNAL // No PSRAM
```

Some comments suggested to change the USB cable, but the problem persists.
[image]
[image]

_Serial output:_
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4

Initializing camera config
[E][camera.c:1113] camera_probe(): Detected camera not supported.
[E][camera.c:1379] esp_camera_init(): Camera probe failed with error 0x20004
```

Schaggo, 2021-04-30:
Here are my usual steps:
- Reset the cam cable. 
- check power voltage 
- Try different cam

It does happen that the cam is DOA. In my case ~ 7 of 100 cams.

bauhaus1977, 2021-05-07:
> Hi everyone,
>
> I'm facing the same problem with my ESP32-CAM, here are the details:
>
> **Board:** ESP32-CAM (ESP32S)
> **Camera module:** TY-OV2640-V2.0
> **USB adapter:** ESP32-CAM-MB
>
> I'm using [...]

Hi, I just bought the same shield from Wish and found the same issue. After many hours thinking on software side I took the last chance that it maybe was a hardware problem...and that's it!!!! I just pushed down with my fingers the connector clip in order to keep down as much as possible the cable flat electric pins onto the related socket electric pins. Then I plug the uUSB connector to the Extended board to supply the module and magically the Serial Monitor of my Arduino IDE shows me the booting process and the CAM recognition and furthermore the DHCP address assigned by my router! So, in the end it's because of the very bad connector and pins...very cheap material, like a toy! I must now think how to fix it in a proper way so the electric contacts can be blocked and have a good electric connection

lksmj, 2021-05-08:
Esp32-cam-mb를 최근에 사용해보다 실패를했습니다.
Esp32-cam-mb 보드의 IO0버튼을 토글을 한상태에서 컴파일시도에 성공했습니다.
[image]

gdamoreira, 2021-05-27:
Hi everyone, found the issue, the problem was the camera module itself, bought a new one and working fine now.

Thanks.

lawong888, 2021-07-11:
I had this issue of exactly the same error for the longest time, and I solved it quite simply:

1.  Check the SSID and password for the wifi connectivity, I made a simple mistake of wrong password and it just fails with the error of Detected Camera not supported.

2.  Also make sure #define CAMERA_MODEL_AI_THINKER is uncommented (remove the // prefix)

closed 2021-09-17 by github-actions[bot]

zaynkhan421, 2022-04-25:
finally i have fixed this issue connect the GND of esp32 module with the GND pin of ESPCAM that is nearby with 5V and and Give the 3V FROM ESP32 MODULE to 3V of ESP32CAM without Press Rest Code Upload after disconecting the GPIO PINS IP address will be shown.

amiteshsingh-cpi, 2023-01-26:
> 

I've the same board and the board does not work for the pixel format JPEG. it works for RGB555 pixel format but the quality of image/video is extremely bad. 
I guess it's a hardware issue.
:/
[gdamoreira](https://github.com/gdamoreira)

https://www.reddit.com/r/esp32/comments/10lvlm1/broken_esp32cam_module/

gdamoreira, 2023-02-01:
Hi @amiteshsingh-cpi , I don't even use this module anymore, sorry I can't help you. Have you tried with another module? In my case it was a broken one, so I just changed and it started to work.