espressif/esp32-camera #119: How to get all captured  images from esp32 sdcard module .

YashwantPanwar, 2020-02-10:
i want save images on ftp server when motion is not detect. how can mange both functionality work at same time like
1. Many images capture one by one from esp32cam and save into esp32cam sd card at a same time .
2. And after some time when motion is not detected by pir sensor then images transfer form esp32 sdcard to ftp server .
can u plz help me …i am new in IOT.

here is my code ..

#include "esp_camera.h"
#include "Arduino.h"
#include <WiFi.h>
#include<WiFiClient.h>
#include<ESP32_FTPClient.h>
#include"FS.h"                // SD Card ESP32
#include "SD_MMC.h"
#include"soc/soc.h"           // Disable brownour problems
#include"soc/rtc_cntl_reg.h"  // Disable brownour problems
#include"driver/rtc_io.h"
#include<EEPROM.h>
#include <HTTPClient.h>
#include<ArduinoJson.h>
#include<Ethernet.h>
#include<string.h>
#include<SimpleTimer.h>

// define the number of bytes you want to access
#define EEPROM_SIZE 1

RTC_DATA_ATTR int bootCount = 0;

// Pin definition for CAMERA_MODEL_AI_THINKER
#define PWDN_GPIO_NUM 32
#define RESET_GPIO_NUM-1
#define XCLK_GPIO_NUM 0
#define SIOD_GPIO_NUM 26
#define SIOC_GPIO_NUM 27
#define Y9_GPIO_NUM 35
#define Y8_GPIO_NUM 34
#define Y7_GPIO_NUM 39
#define Y6_GPIO_NUM 36
#define Y5_GPIO_NUM 21
#define Y4_GPIO_NUM 19
#define Y3_GPIO_NUM 18
#define Y2_GPIO_NUM 5
#define VSYNC_GPIO_NUM 25
#define HREF_GPIO_NUM 23
#define PCLK_GPIO_NUM 22

int pictureNumber = 0;
int count = 0;

#define uS_TO_S_FACTOR 1000000

const char* ssid = “xxxxxx”;
const char* password = “xxxxx”;

char ftp_server[] = “xxxx xxx xx”;
char ftp_user[] = “user”;
char ftp_pass[] = “12345”;

char outBuf[128];
char outCount;

byte clientBuf[64];
int clientCount = 0;
void setup() {
Serial.begin(115200);

WiFi.begin(ssid, password);
// ESP32_FTPClient ftp(ftp_server, ftp_user, ftp_pass, 5000, 2);

while (WiFi.status() != WL_CONNECTED) {
delay(1000);
Serial.println(“Connecting to WiFi..”);
}

Serial.println(“Connected to the WiFi network”);

WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0); //disable brownout detector
//Serial.begin(115200);
Serial.setDebugOutput(true);

camera_config_t config;
config.ledc_channel = LEDC_CHANNEL_0;
config.ledc_timer = LEDC_TIMER_0;

config.pin_d0 = Y2_GPIO_NUM;
config.pin_d1 = Y3_GPIO_NUM;
config.pin_d2 = Y4_GPIO_NUM;
config.pin_d3 = Y5_GPIO_NUM;
config.pin_d4 = Y6_GPIO_NUM;
config.pin_d5 = Y7_GPIO_NUM;
config.pin_d6 = Y8_GPIO_NUM;
config.pin_d7 = Y9_GPIO_NUM;
config.pin_xclk = XCLK_GPIO_NUM;
config.pin_pclk = PCLK_GPIO_NUM;
config.pin_vsync = VSYNC_GPIO_NUM;
config.pin_href = HREF_GPIO_NUM;
config.pin_sscb_sda = SIOD_GPIO_NUM;
config.pin_sscb_scl = SIOC_GPIO_NUM;
config.pin_pwdn = PWDN_GPIO_NUM;
config.pin_reset = RESET_GPIO_NUM;
config.xclk_freq_hz = 20000000;
config.pixel_format = PIXFORMAT_JPEG;

pinMode(4, INPUT);
digitalWrite(4, LOW);
rtc_gpio_hold_dis(GPIO_NUM_4);

if (psramFound()) {
config.frame_size = FRAMESIZE_UXGA; // FRAMESIZE_ + QVGA|CIF|VGA|SVGA|XGA|SXGA|UXGA
config.jpeg_quality = 10;
config.fb_count = 2;
} else {
config.frame_size = FRAMESIZE_SXGA;
config.jpeg_quality = 12;
config.fb_count = 1;
}

// Init Camera
esp_err_t err = esp_camera_init(&config);
if (err != ESP_OK) {
Serial.printf(“Camera init failed with error 0x%x”, err);
return;
}

Serial.println(“Starting SD Card”);

delay(500);
if (!SD_MMC.begin()) {
Serial.println(“SD Card Mount Failed”);
//return;
}

uint8_t cardType = SD_MMC.cardType();
if (cardType == CARD_NONE) {
Serial.println(“No SD Card attached”);
return;
}

camera_fb_t*fb = NULL;

// Take Picture with Camera
fb = esp_camera_fb_get();
if (!fb) {
Serial.println(“Camera capture failed”);
return;
}
// initialize EEPROM with predefined size
EEPROM.begin(EEPROM_SIZE);
pictureNumber = EEPROM.read(0) + 1;
// Path where new picture will be saved in SD Card
String path = “/CapturePicture/” + String(pictureNumber) + “.jpg”;
fs::FS&fs = SD_MMC;
Serial.printf(“Picture file name: %s\n”, path.c_str());
// listDir(SD_MMC, “/CapturePicture”, 0);
File file = fs.open(path.c_str(), FILE_WRITE);
if (!file) {
Serial.println(“Failed to open file in writing mode”);
} else {
file.write(fb->buf, fb->len); // payload (image), payload length
Serial.printf(“Saved file to path: %s\n”, path.c_str());
EEPROM.write(0, pictureNumber);
EEPROM.commit();
}
file.close();
esp_camera_fb_return(fb);
delay(1000);
pinMode(4, OUTPUT);
digitalWrite(4, LOW);
//rtc_gpio_hold_en(GPIO_NUM_4);
// esp_sleep_enable_ext0_wakeup(GPIO_NUM_13, 0);
// Serial.println(“Going to sleep now”);
// delay(1000);
// esp_deep_sleep_start();
//Serial.println(“This will never be printed”);
/*
delay(500);
if (!SD_MMC.begin()) {
Serial.println(“SD Card Mount Failed”);
//return;
}

uint8_t cardType = SD_MMC . cardType ();
if (cardType == CARD_NONE) {
Serial.println(“No SD Card attached”);
return;
}
fs::FS &fs = SD_MMC;
// listDir(SD_MMC, “/”, 0);
createDir(SD_MMC, “/CapturePicture”);
delay(3000);
*/
}

void loop() {
delay(5000);

if ((WiFi.status() == WL_CONNECTED)) { //Check the current connection status

HTTPClient http;

http.begin(“http://worldtimeapi.org/api/timezone/Asia/Kolkata”); //Specify the URL
int httpCode = http.GET(); //Make the request

if (httpCode > 0) { //Check for the returning code
const size_t capacity = JSON_OBJECT_SIZE(15) + 60;
//DynamicJsonBuffer jsonBuffer(capacity);
DynamicJsonDocument doc(1024);
String payload = http.getString(); //Get the request response payload
Serial.println(payload); //Print the response payload
// JsonObject root = jsonBuffer.parseObject(payload);
deserializeJson(doc, payload);
String currentTime = doc[“datetime”].as();
Serial.println(“Current Date-Time = ” + currentTime);
String har = (currentTime.substring(11, 13));
Serial.println(“Current hour = ” + har);
Serial.println(“Starting SD Card”);

delay(500);
if (!SD_MMC.begin()) {
Serial.println(“SD Card Mount Failed”);
//return;
}

uint8_t cardType = SD_MMC . cardType ();
if (cardType == CARD_NONE) {
Serial.println(“No SD Card attached”);
return;
}

fs::FS &fs = SD_MMC;
listDir(SD_MMC, “/CapturePicture”, 0);
delay(3000);
auto error = deserializeJson(doc, payload);
if (error) {
Serial.print(F(“deserializeJson() failed with code “));
Serial.println(error.c_str());
return;
}

}

else {
Serial.println(“Error on HTTP request”);
}

http.end(); //Free the resources
}

delay(10000);

}

void listDir(fs::FS&fs, const char*dirname, uint8_t levels) {
Serial.printf(“Listing directory: %s\n”, dirname);

File root = fs.open(dirname);
if (!root) {
Serial.println(“Failed to open directory”);
return;
}
if (!root.isDirectory()) {
Serial.println(“Not a directory”);
return;
}

File file = root.openNextFile();
while (file) {
if (file.isDirectory()) {
Serial.print(” DIR : “);
Serial.println(file.name());
if (levels) {
listDir(fs, file.name(), levels – 1);
}
} else {
const char * fileName = file.name();
size_t fileSize = file.size();

// Print file size
Serial.println(“\nFile size is: ” + String(fileSize));

//Dynammically alocate buffer
unsigned char * downloaded_file = (unsigned char *) malloc(fileSize);

Serial.print(” FILE: “);
Serial.print(file.name());
Serial.print(” SIZE: “);
Serial.println(file.size());
ESP32_FTPClient ftp(ftp_server, ftp_user, ftp_pass, 5000, 2);
http://ftp.OpenConnection();
//Create a file and write the image data to it;
http://ftp.InitFile(“Type I”);
http://ftp.ChangeWorkDir(“/FaceRecogPython/”); // change it to reflect your directory
// const char * f_name = pic_name . c_str ();
http://ftp.NewFile(fileName );
// http://ftp.WriteData((const uint8_t *)clientBuf, clientCount);
http://ftp.WriteData(downloaded_file, fileSize);
free(downloaded_file);
delay(1000);
http://ftp.CloseFile();
}
file = root.openNextFile();
}
}

Thanks
Yashwant Singh

me-no-dev, 2020-02-12:
i see no reason why you want to delay the FTP connection, but generally you will set the time of the last PIR event and if certain time has passed, you start the upload.

BTW this is an issue tracker, not a help center :) since you ar an Arduino user, you should use the Arduino channels for communication.

closed 2020-02-12 by me-no-dev

Vicentetomas, 2020-03-23:
Hi, Thank you for the code. It's exactly what I was looking for.
The problem is that I get the files in the ftp with the name and bytes correctly but I cannot open them: File error. Please, can you help me?