espressif/esp32-camera #121: Increase frame rate with windowing？

linfan68, 2020-02-16:
I’m trying to use the OV2640 camera to track one small glint on an image with >20 FPS.
 The problem I’m getting is that if I use 800x600 resolution, just loading the image to RGB takes 300ms or so...I’ve tried using GRAY format , or JPEG format and then convert. The performance are about the same. And my guess the bottle neck maybe the PSRAM as mentioned on the document.

So I wonder if I can predict the rough location of the glint, can I use the windowing function of the sensor to set the ROI and reduce data been processed?

me-no-dev, 2020-02-18:
I'm actually working on getting that functionality in the driver. We have support for a few different sensors now and it's much clearer how to implement that functionality. On a side note, there is some commented code in the 2460 driver that exactly deals with ROI. Maybe you can uncomment and give it a shot?

linfan68, 2020-02-24:
> I'm actually working on getting that functionality in the driver. We have support for a few different sensors now and it's much clearer how to implement that functionality. On a side note, there is [...]

I noticed the commented out code and tried to uncomment that. After calling the set window size function, I'm getting error like "failed to get image" about some Vsync stuff (sorry I don't have the code running right now)

linfan68, 2020-02-24:
Just in case someone else run into similar issue, we found another way to solve this problem, it works like this:

1. It seems like OV2640's windowing function cannot increase the framerate anyway, the bottle neck seems to be the transfer bandwidth and memory bandwidth.

2. Based on #1, to get decent FPS, we should use JPEG format output (as mentioned in readme of this project), I'm getting 12.5 FPS for the 800x600 resolution here.

3. To do any on-chip image processing, most of time spent is decoding JPEG image (about 350ms per frame)

4. So our solution is modify the JPEG decoding library.  The one used by eps32-cam is this:
http://elm-chan.org/fsw/tjpgd/00index.html

The change is actually pretty simple (you'll need some knowledge about how jpeg works. Basically jpeg is divided into 8x8 blocks. By reading the raw data without actually decoding it, you'll get the average brightness of each 8x8 block for free. After this, just decode the blocks you're interested in.
We've tried to decode 64x64 area, with the JPEG read time it only takes about 40-60ms per frame for 800x600 resolution, fast enough to run at sensor speed (12.5fps)

closed 2020-02-24 by linfan68

Softwaretom007, 2020-04-20:
@linfan68 
Hi, 
I'm working on exactly same problem. I need to process 200x600pix image in grayscale (or binary). 
Can you share some code for hi speed jpeg decoding?

linfan68, 2020-05-04:
> @linfan68
> Hi,
> I'm working on exactly same problem. I need to process 200x600pix image in grayscale (or binary).
> Can you share some code for hi speed jpeg decoding?

sorry it’s a company project so I cannot share the code with you. But believe me it’s pretty simple... search some blogs about how jpeg works. Make the tiny jpeg library running on you PC and debug through it, add some printf and understand how it works. Then adding some flags to skip the decoding part for the blocks.

The codebase it tiny, it only took me less than a day to modify it.

Softwaretom007, 2020-05-05:
Hi,
thanks for response. Anyway, in meantime, I was browsing some articles about jpeg compression. 
I added tjpgd source codes to project and started optimizing decompression for grayscale output image. 
Now, decoding 640x160pix image takes around 31ms@160MHz default cpu freq. Still can be better, because I dont skip loading and searching codes for chrominance channels.

linfan68, 2020-05-09:
> Hi,
> thanks for response. Anyway, in meantime, I was browsing some articles about jpeg compression.
> I added tjpgd source codes to project and started optimizing decompression for grayscale output [...]

Glad to hear you got some progress. I don’t think there are much room here to improve. In our case the majority time spent is reading and jpeg block data. Since they are sequential and to locate the block data in the middle of image you need to read all blocks before it, it seems like your ROI is a “column” of image, in this case have you considered turn the camera 90 degrees, and only use the top rows of the image. By doing that you can only decide the first rows of blocks and exit.

Softwaretom007, 2020-05-11:
In my case, ROI is multiple lines 640pix wide and  2-4pix in height (will be averaged). Something like barcode scanner, but each scanned line is phase shifted.
I tried rotate camera and set camera window that way. But there is some strange limit in aspect ratio and output is garbage below some aspect ratio.
Anyway, thanks for hints. For now, 31ms is good time, below VSYNC window I need.