espressif/esp32-camera #122: gpio_install_isr_service(410): GPIO isr service already installed

shownb, 2020-02-18:
I (28566) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (28566) sccb: pin_sda 18 pin_scl 23

I (28586) sccb: SCCB_Probe start
I (29076) sccb: SCCB_Probe start
I (29606) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (29606) gpio: GPIO[37]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (29616) gpio: GPIO[38]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (29626) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (29636) gpio: GPIO[35]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (29646) gpio: GPIO[26]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (29646) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (29656) gpio: GPIO[34]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (29666) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (29676) gpio: GPIO[27]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (29686) gpio: GPIO[25]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (29696) camera: Allocating 1 frame buffers (37 KB total)
I (29706) camera: Allocating 37 KB frame buffer in OnBoard RAM
E (29706) gpio: gpio_install_isr_service(410): GPIO isr service already installed
E (29716) camera: gpio_install_isr_service failed (103)
E (29726) camera: Camera init failed with error 0x103
E (29726) camera: Camera Init Failed

shownb, 2020-02-19:
in camera.c from line 1208
    gpio_install_isr_service(ESP_INTR_FLAG_LEVEL1 | ESP_INTR_FLAG_IRAM);
/*
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "gpio_install_isr_service failed (%x)", err);
        goto fail;
    }
*/
it works. but still have E (25205) gpio: gpio_install_isr_service(410): GPIO isr service already installed
but it's not import any more.

apiesse, 2020-05-14:
submitted a pull request to hopefully fix this #145

closed 2021-09-10 by shownb