espressif/esp32-camera #123: Unable to adjust image settings (brightness, contrast etc.)

alanesq, 2020-02-25:
The camera settings for brightness, contrast etc. just do not seem to work at all for me?
I am switching between greyscale and jpg mode, I don't know if this is why I am experiencing this issue.

i.e. if i use the commands:
  sensor_t *s = esp_camera_sensor_get();
  s->set_brightness(s, 2);
This seems to work ok, but if I then just add this line
  s->set_contrast(s, 0);
Then the settings no longer have any effect on the images. This is the case with most of the other available settings too.

You can view the full code here where I am experiencing this problem: 
  https://github.com/alanesq/CameraWifiMotion
the settings are in motion.h in cameraImageSettings() - 
  https://github.com/alanesq/CameraWifiMotion/blob/master/CameraWifiMotion/motion.h

me-no-dev, 2020-02-25:
which sensor? 2460 is not properly set, I'll try to fix that. It changes more than what you have selected, but the info is too limited. Now with more experience with other sensors, I have some ideas on how to pull it off. Stay tuned :)

alanesq, 2020-02-25:
thanks for your reply
I am using the cheap esp32 camera that ebay is full of, with the OV2640 camera sensor

I am using the Arduino IDE v1.8.12 with the ESP32 board manager v1.0.4 (running on Linux Mint)
I have the esp32 dev board selected with PSRam enabled as OTA doesn't work with "AI Thinker esp32-cam" selected.
BTW - I have several of the boards and I have the same problem with all of them

AdrianEddy, 2020-02-26:
I have the same issue with OV2640 on both ESP-EYE and the cheap ESP32-CAM

alanesq, 2020-02-28:
I have discovered that I can get two settings to work if I only adjust the following:

      s->set_agc_gain(s, x);                      // (1 - 31) - may require gain_ctrl set to 0 to operate?
      s->set_gain_ctrl(s, 0);                       // Auto White Balance gain control? (0 or 1)
      s->set_brightness(s, y);                    // (-2 to 2)

VooDust, 2020-03-22:
Thanks, your forum post helped me a lot.

mosazahany, 2022-03-02:
i solved this problem.

PR #701 (2024-11-15): Add mega ccm driver support

## Description

## Related

## Testi
PR #714 (2025-01-31): Add error handling for esp_jpg_decode

## Description
esp_jpg_decode does not return if writer fails, causing unhandled exceptions. Also logs when rgb_write malloc fails.

## Related
PR #789 (2025-08-05): fix: Fixed GPIO implicit declaration

## Description

- Fixed Implicit declaration of GPIO functions

## Related

Fixes: #786
Closes: #786
PR #803 (2025-09-22): Use renamed sdkconfig option

## Description

As per the changes in ESP-IDF, the config flag `CONFIG_SPIRAM_SUPPORT` is deprecated and replaced by `CONFIG_SPIRAM`

https://github.com/espressif/esp-idf/blob/5c5eb99eabc7d3e7f704656ffe9ebf8f5a77afbd/components/esp_hw_support/sdkconfig.rename.esp32#L20

## Related