espressif/esp32-camera #125: ESP restart on server query.

six8byte, 2020-02-26:
While trying to connect to webserver where video is hosted, following error occurs:

`
I (7499) server:: station:8c:85:90:80:08:de join, AID=1
I (7649) tcpip_adapter: softAP assign IP to station,IP is: 192.168.4.2
E (44419) camera: Failed to get the frame on time!
E (44419) camera.c:: Camera capture failed
`

and ESP restarts.

code I use:

camera.c
```
esp_err_t video_stream_handler(httpd_req_t *req){
    camera_fb_t * fb = NULL;
    esp_err_t res = ESP_OK;
    size_t _jpg_buf_len;
    uint8_t * _jpg_buf;
    char * part_buf[64];
    static int64_t last_frame = 0;
    if(!last_frame) {
        last_frame = esp_timer_get_time();
    }

    res = httpd_resp_set_type(req, _STREAM_CONTENT_TYPE);
    if(res != ESP_OK){
        return res;
    }

    while(true){
        fb = esp_camera_fb_get();
        if (!fb) {
            ESP_LOGE(TAG, "Camera capture failed");
            res = ESP_FAIL;
        } else {
            if(fb->format != PIXFORMAT_JPEG){
                bool jpeg_converted = frame2jpg(fb, 80, &_jpg_buf, &_jpg_buf_len);
                if(!jpeg_converted){
                    ESP_LOGE(TAG, "JPEG compression failed");
                    esp_camera_fb_return(fb);
                    res = ESP_FAIL;
                }
            } else {
                _jpg_buf_len = fb->len;
                _jpg_buf = fb->buf;
            }
        }
        if(res == ESP_OK){
            res = httpd_resp_send_chunk(req, _STREAM_BOUNDARY, strlen(_STREAM_BOUNDARY));
        }
        if(res == ESP_OK){
            size_t hlen = snprintf((char *)part_buf, 64, _STREAM_PART, _jpg_buf_len);

            res = httpd_resp_send_chunk(req, (const char *)part_buf, hlen);
        }
        if(res == ESP_OK){
            res = httpd_resp_send_chunk(req, (const char *)_jpg_buf, _jpg_buf_len);
        }
        if(fb->format != PIXFORMAT_JPEG){
            free(_jpg_buf);
        }
        esp_camera_fb_return(fb);
        if(res != ESP_OK){
            break;
        }
        int64_t fr_end = esp_timer_get_time();
        int64_t frame_time = fr_end - last_frame;
        last_frame = fr_end;
        frame_time /= 1000;
        ESP_LOGI(TAG, "MJPG: %uKB %ums (%.1ffps)",
            (uint32_t)(_jpg_buf_len/1024),
            (uint32_t)frame_time, 1000.0 / (uint32_t)frame_time);
    }

```
server.c

```
httpd_uri_t index_uri = {
    .uri       = "/",
    .method    = HTTP_GET,
    .handler   = video_stream_handler,
    .user_ctx  = NULL
  };

void start_webserver(void)
{
    httpd_handle_t server = NULL;
    httpd_config_t config = HTTPD_DEFAULT_CONFIG();
    config.server_port = 80;

    ESP_LOGI(TAG, "Starting server on port: '%d'", config.server_port);
    vTaskDelay(50 / portTICK_PERIOD_MS);
    if (httpd_start(&server, &config) == ESP_OK) {
        ESP_LOGI(TAG, "Registering URI handler");
        httpd_register_uri_handler(server, &index_uri);
    }

    return NULL;
}

```

six8byte, 2020-02-26:
Same thing happens, when I try to host JPG picture.

me-no-dev, 2020-02-27:
in camera.c where you have the following code
```cpp
        fb = esp_camera_fb_get();
        if (!fb) {
            ESP_LOGE(TAG, "Camera capture failed");
            res = ESP_FAIL;
        } else {
```

After `res = ESP_FAIL;` add a new line with `break;`.
This is fixed in ESP-WHO and will be fixed in Arduino also.

six8byte, 2020-02-27:
I'm using ESP-IDF.
After adding break statement following error occurs:
```
error: break statement not within loop or switch
         break;
         ^

```

Also this only resolves problem with restarting. Using picture capture and saving it to SDcard, everything works. But there is still problem with capturing.

me-no-dev, 2020-02-27:
that is something else :) you need to explain more. saving to sdcard and getting a frame is the same thing in therms of the driver. If you successfully get a frame and then have issues with the browser, check headers and maybe even get a tcpdump/wireshark to check what is going on

six8byte, 2020-02-27:
Actually solved. Changed pin connection. Probably one was burnt or not receiving signal properly. Closing :D

closed 2020-02-27 by six8byte